emsesp/EMS-ESP32 #2323: Boiler starts:  heatstarts vs. dhw_starts - Change of logic requested

tp1de, 2025-01-02:
I am watching the boiler starts per 24 hours.

During Summer (heating off) boiler starts are initiated by ww preparation and counted in dhw_starts whenever boiler is switched on.

During Winter when heating is permanent on, ww prep does not create any new ignition of the boiler.
Therefore dhw_starts has not changed since 2 months anymore.
But ww prep is setting boiler settemp significant higher then heating. 
Afterwards the boiler switches off and then after a while a new start is counted as a new heatstart.

Is there any chance to count those starts after ww preparation as dhw_starts as well?
Or the other way around: Whenever ww prep starts increase the counter.

MichaelDvP, 2025-01-02:
The counter is read from boiler, emsesp does not change it. And can not change the logic. Most boilers have a setting "dhw prirority", if this is selected and dhw needs to be heated the flow pump stops, dhw is prepared and dhw start counter counts up. If dhw priority is not set, the heating runs all the time and there is no extra dhw start, dhw is just parallel to the heating.
I guess with 3-way-valve  dhw-priority is always set (or should be), because there is no way to combine the haeting and dhw.

tp1de, 2025-01-02:
Heating priority is active. 
The heating pumps are stoping but boiler stays on and after some seconds the new settemp is active and the ww buffer load pump is active.

tp1de, 2025-01-02:
> The counter is read from boiler, emsesp does not change it

Yes but the boiler has only one counter. This one should stay the same: burnstarts
Are you reading heatstarts and dhw_starts out of a telegram or are you calculating them on conditions?

MichaelDvP, 2025-01-02:
dhw starts: ems: telegram 0x34, offset13, ems+: telegram 0xE9, offset 17
heatstarts: ems: telegram 0x19, offset, 22, ems+: telegram 0xE5, offset 22
burnstarts: ems: tlegram 0x19, offset 10, ems+: telegram 0xE5, offset 10

No calculation from ems-esp!

tp1de, 2025-01-02:
Good to know, then I need to calculate myself

MichaelDvP, 2025-01-02:
> Could I use the scheduler doing it?

Yes. Something like condition "boiler/tapwateractive == on" But you need to use a ram value as counter, so it will be lost on a ems-esp reset.

tp1de, 2025-01-02:
Non persistent is no option. I need to calculate in HA.

closed 2025-01-02 by tp1de