emsesp/EMS-ESP32 #2325: Multiple EMS Gateway(BBQ Kees) on Home Assist

kevinwolswijk, 2025-01-03:
Hi All,

I own 3x EMS gateway from BBQ Kees.
Due to multiple heat pumps, we also have multiple gateways.

these are linked with their own ID to an MQTT broker running on Synology NAS.
So far this is running well, and I can see all the data neatly in the MQTT Broker.

Unfortunately, homeassist's mqtt discover does not process this information properly.
BBQ Kees has indicated this (only seen later) in his manual.

Does anyone know another way to get this all working in home assist?

Thanks!

closed 2025-01-03 by kevinwolswijk

proddy, 2025-01-03:
you fixed it then using the multi-instance MQTT option?

kevinwolswijk, 2025-01-03:
yeah correct:)
At first it was not working and I found (probably an old document) that say's it's not working on Home assist.

After a clean installation it's working.