emsesp/EMS-ESP32 #2326: powerreduction not available

Scratcher48, 2025-01-03:
Hey,

i have a CS6800.
It seems, that the silent mode can be controlled, but the powerreduction is not available, although the silent mode is activated.

I'm runnig 3.7.2-dev.6

greetings

proddy, 2025-01-04:
See https://docs.emsesp.org/Support/ 

Attach the support info so we can see the product ID and figure out if and why that entity is not being shown

Scratcher48, 2025-01-04:
Attacken the Support information:

{"system":{"version":"3.7.2-dev.6","uptime":"000+20:45:45.374","uptimeSec":74745,"platform":"ESP32","cpuType":"ESP32-D0WD-V3","arduino":"ESP32 Arduino v2.0.17","sdk":"v4.4.7-dirty","freeMem":131,"maxAlloc":83,"freeCaps":4147,"usedApp":2399,"freeApp":3297,"partition":"app0","resetReason":"Software reset CPU / Software reset CPU","psram":true,"psramSize":4093,"freePsram":4052,"model":""},"network":{"network":"WiFi","hostname":"ems-esp","RSSI":-58,"WIFIReconnects":1,"TxPowerSetting":0,"staticIP":false,"lowBandwidth":false,"disableSleep":true,"enableMDNS":true,"enableCORS":false,"APProvisionMode":"disconnected","APSecurity":"wpa2","APSSID":"ems-esp"},"ntp":{"NTPStatus":"connected","enabled":true,"server":"time.google.com","tzLabel":"Europe/Amsterdam"},"mqtt":{"MQTTStatus":"connected","MQTTPublishes":16657,"MQTTQueued":0,"MQTTPublishFails":0,"MQTTReconnects":1,"enabled":true,"clientID":"Heizung","keepAlive":60,"cleanSession":true,"entityFormat":0,"base":"Heizung","discoveryPrefix":"homeassistant","discoveryType":0,"nestedFormat":1,"haEnabled":true,"mqttQos":0,"mqttRetain":true,"publishTimeHeartbeat":60,"publishTimeBoiler":10,"publishTimeThermostat":10,"publishTimeSolar":10,"publishTimeMixer":10,"publishTimeWater":10,"publishTimeOther":10,"publishTimeSensor":10,"publishSingle":false,"publish2command":false,"sendResponse":true},"syslog":{"enabled":false},"sensor":{"temperatureSensors":0,"temperatureSensorReads":0,"temperatureSensorFails":0,"analogSensors":0,"analogSensorReads":0,"analogSensorFails":0},"api":{"APICalls":1,"APIFails":0},"bus":{"busStatus":"connected","busProtocol":"HT3","busTelegramsReceived":427779,"busReads":67521,"busWrites":3,"busIncompleteTelegrams":50,"busReadsFailed":0,"busWritesFailed":0,"busRxLineQuality":100,"busTxLineQuality":100},"settings":{"boardProfile":"E32V2","locale":"de","txMode":1,"emsBusID":11,"showerTimer":false,"showerMinDuration":180,"showerAlert":false,"hideLed":false,"noTokenApi":false,"readonlyMode":false,"fahrenheit":false,"dallasParasite":false,"boolFormat":1,"boolDashboard":1,"enumFormat":1,"analogEnabled":true,"telnetEnabled":true,"maxWebLogBuffer":50,"modbusEnabled":false,"forceHeatingOff":false,"developerMode":false},"devices":[{"type":"boiler","name":"XCU_THH/CS*800i, Logatherm WLW*","deviceID":"0x08","productID":8,"brand":"Bosch","version":"00.00","entities":167,"handlersReceived":"0xBF 0x14 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F","handlersFetched":"0xE6 0xEA 0x048D 0x04A2 0x0485 0x0486 0x0492 0x0488 0x0484 0x048B 0x0491 0x0499 0x049C 0x049D 0x04AE 0x04AF 0x02CC 0x04AA 0x04A7 0x02D6","handlersPending":"0x10 0x11 0xC2 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0x28 0x048A 0x04A5","handlersIgnored":"0x023E 0x0240 0x0291 0x0292 0x0293 0x0294 0x029B 0x02A5 0x02F5 0x04A8 0x04A9 0x04AB 0x04AC 0x04CA 0x04CB 0xE7 0x02E0 0x02EA 0x061E 0x04D8 0x17 0xF7 0x0497 0xF9"},{"type":"thermostat","name":"HMI800.2/Rego 3000, UI800, Logamatic BC400","deviceID":"0x10","productID":253,"brand":"","version":"47.10","entities":74,"handlersReceived":"0x06 0x02BA 0x02BB 0x02BC 0x031D 0x0267","handlersFetched":"0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02CC 0x0467 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240 0xBB 0x023E 0x016E","handlersPending":"0xA3 0xA2 0x12 0x13 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x02F6 0x031B 0x031E 0x0269","handlersIgnored":"0xBF 0x0507 0x0508 0x0509 0x050A 0x059D"},{"type":"gateway","name":"K40RF/MX400","deviceID":"0x48","productID":17,"brand":"Bosch","version":"11.01","entities":0,"handlersIgnored":"0x2040 0x0168 0x0169 0x016A 0x016B 0x016C 0x016E 0x023E 0x0240 0x0291 0x0484 0xF9 0x0495 0x0494 0x04AF 0x04AE 0x0DE8 0x0DE9 0x0DEC 0x0DED 0x0DEF 0x0DF0 0x0292 0x0499 0xFF06 0x0293 0x0294 0x029B 0x029C 0x029D 0x029E 0x02AF 0x02B0 0x02B1 0x02B2 0x02B9 0x02BA 0x02BB 0x02BC 0x02C3 0x02C4 0x02C5 0x02C6 0xFF07 0x02E0 0x02FF 0x0449 0x044A 0x044B 0x044C 0x0471 0x0472 0x0473 0x0474 0x0CBD 0x0165 0x02CC 0x0486 0x048B 0x04A7 0x04AD 0x0593 0x0B0B 0x0AF7 0x0AF8 0xFE06 0x0AF9 0x0AFA 0x0B01 0x0B0C 0x0B0D 0x0B"},{"type":"gateway","name":"K40RF/MX400","deviceID":"0x4B","productID":17,"brand":"","version":"11.01","entities":0,"handlersIgnored":"0x04D6 0x04D7 0xBF"}]}

proddy, 2025-01-05:
ok, it should be there. perhaps it doesn't have a value. You can check if you go to Customizations and search for the entity.

Also, if you can enable Developer Mode (application settings), go to the System Log and click on the blue arrow and enter `8 484` . Copy & paste the 484 telegram that is fetched so we can see if the value is in there.

Scratcher48, 2025-01-05:
Hey,
I des not have a value, thats the problem.

2025-01-05 11:17:01.000 INFO 1: [emsesp] EMS-ESP version 3.7.2-dev.6 (app0 partition)
2025-01-05 11:17:01.000 INFO 2: [emsesp] Last system reset reason Core0: Software reset CPU, Core1: Software reset CPU
2025-01-05 11:17:01.000 INFO 3: [emsesp] Board profile set to E32V2
2025-01-05 11:17:01.000 INFO 4: [emsesp] Loaded EMS device library (122 entries)
2025-01-05 11:17:01.000 INFO 5: [emsesp] Starting Scheduler service
2025-01-05 11:17:01.000 INFO 6: [emsesp] Starting Custom Entity service
2025-01-05 11:17:01.000 INFO 7: [temperaturesensor] Starting Temperature Sensor service
2025-01-05 11:17:01.000 INFO 8: [analogsensor] Starting Analog Sensor service
2025-01-05 11:17:01.000 INFO 9: [emsesp] Starting Modules service
2025-01-05 11:17:01.000 INFO 10: [module] No external library modules found
2025-01-05 11:17:01.000 INFO 11: [emsesp] Starting Web Server
2025-01-05 11:17:02.764 WARNING 12: [emsesp] WiFi disconnected (#1). Reason: auth fail (202)
2025-01-05 11:17:02.845 INFO 13: [emsesp] WiFi connected (Local IP=192.168.178.24, hostname=ems-esp, TxPower=15.0 dBm)
2025-01-05 11:17:02.872 INFO 14: [emsesp] Starting mDNS Responder service
2025-01-05 11:17:02.872 INFO 15: [emsesp] Starting NTP service
2025-01-05 11:17:02.919 INFO 16: [mqtt] MQTT connected
2025-01-05 11:17:04.544 INFO 17: [emsesp] IPv6 (WiFi) local: fe80:0000:0000:0000:ce7b:5cff:fe3b:8090
2025-01-05 11:17:09.969 INFO 18: [emsesp] Detected EMS device: boiler (0x08)
2025-01-05 11:17:26.794 WARNING 19: [telegram] Incomplete Rx: CB 08 FF 00 03 D8 FF FF 77 08
2025-01-05 11:17:38.117 WARNING 20: [telegram] Tx queue overflow, skip one message
2025-01-05 11:17:38.117 WARNING 21: [telegram] Tx queue overflow, skip one message
2025-01-05 11:17:38.117 WARNING 22: [telegram] Tx queue overflow, skip one message
2025-01-05 11:17:38.117 INFO 23: [emsesp] Detected EMS device: thermostat (0x10)
2025-01-05 11:17:38.117 WARNING 24: [telegram] Tx queue overflow, skip one message
2025-01-05 11:17:42.367 WARNING 25: [telegram] Tx queue overflow, skip one message
2025-01-05 11:17:42.367 WARNING 26: [telegram] Tx queue overflow, skip one message
2025-01-05 11:18:46.666 INFO 27: [emsesp] Detected EMS device: gateway (0x4B)
2025-01-05 11:19:31.602 INFO 28: [emsesp] Detected EMS device: gateway (0x48)
2025-01-05 11:48:59.647 INFO 29: [command] Called command boiler/silentmode (Silentmodus) with value aus
2025-01-05 11:52:25.520 WARNING 30: [telegram] Incomplete Rx: CB 00 FF 00 03 D7 62
2025-01-05 12:07:26.687 WARNING 31: [telegram] Incomplete Rx: CB 08 FF 00 03 D8 FF FF 77 08
2025-01-05 12:40:55.894 INFO 32: [command] Called command boiler/ecostop (ECO Stopptemp.) with value 47 and id 9 on device 0x08
2025-01-05 13:06:02.254 INFO 33: [command] Called command boiler/ecostop (ECO Stopptemp.) with value 47 and id 9 on device 0x08
2025-01-05 13:06:23.302 INFO 34: [command] Called command boiler/ecostop (ECO Stopptemp.) with value 47 and id 9 on device 0x08
2025-01-05 13:06:58.877 INFO 35: [command] Called command boiler/ecostop (ECO Stopptemp.) with value 47 and id 9 on device 0x08
2025-01-05 13:07:03.319 INFO 36: [command] Called command boiler/seltemplow (ausgewÃ¤hlte untere Temperatur) with value 41 and id 9 on device 0x08
2025-01-05 13:07:13.890 INFO 37: [command] Called command boiler/seltemplow (ausgewÃ¤hlte untere Temperatur) with value 43 and id 9 on device 0x08
2025-01-05 13:31:13.488 WARNING 38: [telegram] Incomplete Rx: CB 48 FF 96 FE 06 E2 08
2025-01-05 13:32:23.790 WARNING 39: [telegram] Incomplete Rx: CB 00 FF 00 08
2025-01-05 13:33:14.424 INFO 40: [command] Called command boiler/silentmode (Silentmodus) with value an
2025-01-05 13:34:53.235 INFO 41: [emsesp] Board profile set to E32V2
2025-01-05 13:35:50.966 INFO 42: [command] Called command system/read (send read request) with value 8 484
2025-01-05 13:35:51.438 NOTICE 43: [emsesp] boiler(0x08) W me(0x0B), HPSilentMode(0x0484), data: 01 00 1E 28 04 00 00 00 00 02 02 F6 00 00 01
2025-01-05 13:36:27.679 INFO 44: [command] Called command system/read (send read request) with value 8 484
2025-01-05 13:36:27.861 NOTICE 45: [emsesp] boiler(0x08) W me(0x0B), HPSilentMode(0x0484), data: 01 00 1E 28 04 00 00 00 00 02 02 F6 00 00 01

proddy, 2025-01-05:
i can see its missing in the 484 telegram

Scratcher48, 2025-01-05:
Ok, what to Do?

Scratcher48, 2025-01-09:
?

proddy, 2025-01-09:
no idea. Maybe @MichaelDvP has one. Look at https://docs.emsesp.org/Troubleshooting/#im-missing-certain-data-from-an-ems-device and see if you can find it out yourself.