emsesp/EMS-ESP32 #2329: Update via Webpage=>Settings=>Button_Upgrade=>Button_Install does not work

bergteam, 2025-01-05:
### DESCRIPTION
The automatic update via Webpage=>Settings=>Button_Upgrade=>Button_Install does not work. Same behavior appeared for the last few beta versions (was working some versions/months ago).

Currently installed 3.7.2-dev.7
Trying to upgrade to 3.7.2-dev.7
Platform ESP32-16MB
Browser Edge, same behavior on Firefox. Both latest versions.

### TO REPRODUCE
1. Go to the Upgrade Webpage. Press the button "Install"
[image]

2. This message appears and 3 sec later the main EMS Webpage is loaded
[image]

3. Try again. Go to Upgrade page shown in (1). Press again "Install". 
This Webpage appears and hangs forever (at least 5min) where a manual update takes only 1min.
[image]

Press reload on the browser and I'm back to the main menu. 

### EXPECTED BEHAVIOUR
Pressing "Install" will upgrade the EMS automatically and return to main screen in browser.

PS: Manual firmware file download and upload to ESP works fine.

proddy, 2025-01-05:
I'll investigate

proddy, 2025-01-05:
I couldn't reproduce. I'll keep testing.

bergteam, 2025-01-05:
Thanks. I will check and trace on next beta version update. Also will delete browser cache.

proddy, 2025-01-05:
I'll add a re-install option too

proddy, 2025-01-08:
please re-open if it happens again

closed 2025-01-08 by proddy

bergteam, 2025-01-09:
Same issue happens during upgrade 3.7.2-dev.8 => 3.7.2-dev.9 
[image]

Again this happens on re-install on newest version
[image]

Seems it get's stuck on uploading firmware.

Fine with me as there is a workaround and no one else seems to have this issue.