emsesp/EMS-ESP32 #2330: Can`t set selflowtemp via SAT or EMS-ESP

Ed1954, 2025-01-05:
Sendig command
2025-01-05 15:12:20.919 I 66: [command] Called command boiler/selflowtemp (Gewählte Vorlauftemperatur) with value 40.80
are ignored by the boiler and/or EMS-ESP. Unable to run SAT
Boiler Buderus GB 182i2

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x ] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ x] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [ x] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [ x] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
{
  "system": {
    "version": "3.7.2-dev.8",
    "uptime": "000+00:55:50.747",
    "uptimeSec": 3350,
    "platform": "ESP32S3",
    "cpuType": "ESP32-S3",
    "arduino": "ESP32 Arduino v2.0.17",
    "sdk": "v4.4.7-dirty",
    "freeMem": 163,
    "maxAlloc": 135,
    "freeCaps": 162,
    "usedApp": 2294,
    "freeApp": 3402,
    "partition": "app0",
    "resetReason": "Software reset CPU / Software reset CPU",
    "psram": false,
    "model": "",
    "temperature": 43.4
  },
  "network": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -63,
    "WIFIReconnects": 0,
    "BSSID": "set",
    "TxPowerSetting": 0,
    "staticIP": true,
    "lowBandwidth": false,
    "disableSleep": true,
    "enableMDNS": true,
    "enableCORS": false,
    "APProvisionMode": "disconnected",
    "APSecurity": "wpa2",
    "APSSID": "ems-esp"
  },
  "ntp": {
    "NTPStatus": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tzLabel": "Europe/Amsterdam"
  },
  "mqtt": {
    "MQTTStatus": "connected",
    "MQTTPublishes": 928,
    "MQTTQueued": 0,
    "MQTTPublishFails": 0,
    "MQTTReconnects": 0,
    "enabled": true,
    "clientID": "esp32-463bdaec",
    "keepAlive": 60,
    "cleanSession": false,
    "entityFormat": 0,
    "base": "ems-esp",
    "discoveryPrefix": "homeassistant",
    "discoveryType": 0,
    "nestedFormat": 1,
    "haEnabled": true,
    "mqttQos": 0,
    "mqttRetain": false,
    "publishTimeHeartbeat": 60,
    "publishTimeBoiler": 10,
    "publishTimeThermostat": 10,
    "publishTimeSolar": 10,
    "publishTimeMixer": 10,
    "publishTimeWater": 10,
    "publishTimeOther": 10,
    "publishTimeSensor": 10,
    "publishSingle": false,
    "publish2command": false,
    "sendResponse": false
  },
  "syslog": {
    "enabled": false
  },
  "sensor": {
    "temperatureSensors": 0,
    "temperatureSensorReads": 0,
    "temperatureSensorFails": 0
  },
  "api": {
    "APICalls": 0,
    "APIFails": 0
  },
  "bus": {
    "busStatus": "connected",
    "busProtocol": "Buderus",
    "busTelegramsReceived": 6100,
    "busReads": 1728,
    "busWrites": 47,
    "busIncompleteTelegrams": 0,
    "busReadsFailed": 0,
    "busWritesFailed": 0,
    "busRxLineQuality": 100,
    "busTxLineQuality": 100
  },
  "settings": {
    "boardProfile": "S32S3",
    "locale": "de",
    "txMode": 1,
    "emsBusID": 11,
    "showerTimer": false,
    "showerMinDuration": 180,
    "showerAlert": false,
    "hideLed": false,
    "noTokenApi": false,
    "readonlyMode": false,
    "fahrenheit": false,
    "dallasParasite": false,
    "boolFormat": 1,
    "boolDashboard": 1,
    "enumFormat": 1,
    "analogEnabled": false,
    "telnetEnabled": true,
    "maxWebLogBuffer": 25,
    "modbusEnabled": false,
    "forceHeatingOff": false,
    "developerMode": false
  },
  "devices": [
    {
      "type": "boiler",
      "name": "ACU-MH/Condens 5000i, Greenstar 8000/GC9800IW, GB192i*2",
      "deviceID": "0x08",
      "productID": 195,
      "brand": "Buderus",
      "version": "05.06",
      "entities": 86,
      "handlersReceived": "0xBF 0xC2 0x15 0xD1 0xE3 0xE4 0xE5 0xE9 0x2E 0x3B 0x04",
      "handlersFetched": "0x14 0xE6 0xEA 0x28",
      "handlersPending": "0x10 0x11 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A",
      "handlersIgnored": "0x02E0 0x02EA 0xD7 0xD8 0x02D6"
    },
    {
      "type": "thermostat",
      "name": "BC400-FO/UI800, BC400",
      "deviceID": "0x10",
      "productID": 4,
      "brand": "",
      "version": "49.08",
      "entities": 51,
      "handlersReceived": "0x06 0x02BA 0x02BB 0x02BC 0x0267",
      "handlersFetched": "0x02A5 0x02B9 0x02AF 0x029B 0x02CC 0x0291 0x0292 0x0293 0x0294 0x02F5 0x02F6 0x023A 0x0240 0x016E",
      "handlersPending": "0xA3 0xA2 0x12 0x13 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x0467 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031B 0x031D 0x031E 0xBB 0x023E 0x0269",
      "handlersIgnored": "0xBF 0xC4 0x31 0x3B 0xF7 0xF9"
    },
    {
      "type": "gateway",
      "name": "MX300/WiFi module",
      "deviceID": "0x48",
      "productID": 252,
      "brand": "Buderus",
      "version": "08.02",
      "entities": 0,
      "handlersIgnored": "0xF9 0x2040 0x2E 0x3B 0xE5 0xE9"
    }
  ]
}

```

### TO REPRODUCE
Enter selflowtemp manually

proddy, 2025-01-05:
Thanks. Could you
 - set logging log level to ALL
 - run the command again using a whole integer value (easier to trace) which is lower than `heatingtemp`
 - show the output telegrams after the call

SoftVienka, 2025-01-05:
Also so. Bosch Condens 2300

[emsesp_system_info.json](https://github.com/user-attachments/files/18310588/emsesp_system_info.json)

2025-01-05 15:42:24.489 INFO 147: [command] Called command boiler/selflowtemp (selected flow temperature) with value 49
2025-01-05 15:42:24.641 TRACE 148: [emsesp] me(0x0B) W boiler(0x08), UBASetPoints(0x1A), data: 31
2025-01-05 15:42:26.862 TRACE 149: [emsesp] me(0x0B) R boiler(0x08), UBAMonitorFastPlus(0xE4), length: 27
2025-01-05 15:42:26.915 TRACE 150: [emsesp] boiler(0x08) W me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 2D 2D 00 00 C8 30 01 E1 64 0D 03 00 01 E5 00 00 00 00 00 91 FF 00 01 E1 00 00
2025-01-05 15:42:26.915 TRACE 151: [emsesp] me(0x0B) R boiler(0x08), UBAMonitorFastPlus(0xE4), length: 27 (offset 27)
2025-01-05 15:42:26.915 TRACE 152: [emsesp] boiler(0x08) W me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 00 00 00 00 0D 64 (offset 35)
2025-01-05 15:42:26.915 TRACE 153: [emsesp] me(0x0B) R boiler(0x08), UBAMonitorFastPlus(0xE4), length: 27 (offset 42)
2025-01-05 15:42:26.915 TRACE 154: [emsesp] boiler(0x08) W me(0x0B), UBAMonitorFastPlus(0xE4), data: <empty> (offset 42)
2025-01-05 15:42:28.001 INFO 155: [command] Called command custom/casovac_1 (set custom value on ems) with value 4
2025-01-05 15:42:28.001 INFO 156: [command] Called command analogsensor/rele_cerpadlo (digital_out) with value 0
2025-01-05 15:42:28.953 TRACE 157: [emsesp] me(0x0B) R boiler(0x08), UBAParametersPlus(0xE6), length: 27
2025-01-05 15:42:29.005 TRACE 158: [emsesp] boiler(0x08) W me(0x0B), UBAParametersPlus(0xE6), data: 01 4B 00 52 64 0C 00 01 05 FE 07 01 01 64 01 00 00 00 00 2E 01 01 00 00 01 00

Ed1954, 2025-01-05:
2025-01-05 15:53:27.614 TRACE 780: [emsesp] boiler(0x08) B all(0x00), UBAMonitorWWPlus(0xE9), data: 00 80 00 80 00 00 00 00 00 46 3C 00 00 00 00 00 00 00 00 00 00 00 00 05 02 00
2025-01-05 15:53:33.507 INFO 781: [command] Called command boiler/selflowtemp (Gewählte Vorlauftemperatur) with value 40.10
2025-01-05 15:53:33.619 TRACE 782: [emsesp] me(0x0B) W boiler(0x08), UBASetPoints(0x1A), data: 28
2025-01-05 15:53:35.663 TRACE 783: [emsesp] me(0x0B) R boiler(0x08), UBAMonitorFastPlus(0xE4), length: 27
2025-01-05 15:53:35.698 TRACE 784: [emsesp] boiler(0x08) W me(0x0B), UBAMonitorFastPlus(0xE4), data: 10 20 2D 2D 00 C8 3A 02 47 64 24 03 00 02 13 00 00 01 C4 01 FF 10 00 02 48 00 00
2025-01-05 15:53:35.698 TRACE 785: [emsesp] me(0x0B) R boiler(0x08), UBAMonitorFastPlus(0xE4), length: 27 (offset 27)
2025-01-05 15:53:35.698 TRACE 786: [emsesp] boiler(0x08) W me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 01 CE 01 BF 1D 50 (offset 35)
2025-01-05 15:53:35.698 TRACE 787: [emsesp] me(0x0B) R boiler(0x08), UBAMonitorFastPlus(0xE4), length: 27 (offset 42)
2025-01-05 15:53:35.698 TRACE 788: [emsesp] boiler(0x08) W me(0x0B), UBAMonitorFastPlus(0xE4), data: <empty> (offset 42)
2025-01-05 15:53:36.980 TRACE 789: [emsesp] boiler(0x08) B all(0x00), UBAMonitorFastPlus(0xE4), data: 10 20 2D 2D 00 C8 3A 02 48 64 24 03 00 02 13 00 00 01 C4 01 FB 10 00 02 48 00 00
2025-01-05 15:53:37.159 TRACE 790: [emsesp] boiler(0x08) B all(0x00), UBAMonitorFastPlus(0xE4), data: 00 01 CE 01 BF 1D 50 (offset 35)
2025-01-05 15:53:37.348 TRACE 791: [emsesp] boiler(0x08) B all(0x00), UBAMonitorSlowPlus2(0xE3), data: 01 00 01 00 00 00 00 00 00 00 00 02 47 1D 64 46 00 3A 00 00
2025-01-05 15:53:37.622 TRACE 792: [emsesp] boiler(0x08) B all(0x00), UBAMonitorWWPlus(0xE9), data: 00 80 00 80 00 00 00 00 00 46 3C 00 00 00 00 00 00 00 00 00 00 00 00 05 02 00
2025-01-05 15:53:46.953 TRACE 793: [emsesp] boiler(0x08) B all(0x00), UBAMonitorFastPlus(0xE4), data: 10 20 2D 2D 00 C8 3A 02 47 64 24 03 00 02 13 00 00 01 C4 02 00 10 00 02 48 00 00
2025-01-05 15:53:46.997 INFO 794: [command] Called command boiler/selflowtemp (Gewählte Vorlauftemperatur) with value 40
2025-01-05 15:53:47.125 TRACE 795: [emsesp] boiler(0x08) B all(0x00), UBAMonitorFastPlus(0xE4), data: 00 01 CE 01 BF 1D 50 (offset 35)
2025-01-05 15:53:47.319 TRACE 796: [emsesp] boiler(0x08) B all(0x00), UBAMonitorSlowPlus2(0xE3), data: 01 00 01 00 00 00 00 00 00 00 00 02 47 1D 64 46 00 3A 00 00
2025-01-05 15:53:47.525 TRACE 797: [emsesp] boiler(0x08) B all(0x00), UBAMonitorWWPlus(0xE9), data: 00 80 00 80 00 00 00 00 00 46 3C 00 00 00 00 00 00 00 00 00 00 00 00 05 02 00
2025-01-05 15:53:47.766 TRACE 798: [emsesp] me(0x0B) W boiler(0x08), UBASetPoints(0x1A), data: 28
2025-01-05 15:53:49.815 TRACE 799: [emsesp] me(0x0B) R boiler(0x08), UBAMonitorFastPlus(0xE4), length: 27
2025-01-05 15:53:49.851 TRACE 800: [emsesp] boiler(0x08) W me(0x0B), UBAMonitorFastPlus(0xE4), data: 10 20 2D 2D 00 C8 3A 02 48 64 24 03 00 02 13 00 00 01 C3 01 FE 10 00 02 47 00 00
2025-01-05 15:53:49.851 TRACE 801: [emsesp] me(0x0B) R boiler(0x08), UBAMonitorFastPlus(0xE4), length: 27 (offset 27)
2025-01-05 15:53:49.851 TRACE 802: [emsesp] boiler(0x08) W me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 01 CE 01 BF 1D 50 (offset 35)
2025-01-05 15:53:49.851 TRACE 803: [emsesp] me(0x0B) R boiler(0x08), UBAMonitorFastPlus(0xE4), length: 27 (offset 42)
2025-01-05 15:53:49.851 TRACE 804: [emsesp] boiler(0x08) W me(0x0B), UBAMonitorFastPlus(0xE4), data: <empty> (offset 42)
2025-01-05 15:53:56.955 TRACE 805: [emsesp] boiler(0x08) B all(0x00), UBAMonitorFastPlus(0xE4), data: 10 20 2D 2D 00 C8 3A 02 48 64 24 03 00 02 13 00 00 01 C3 01 FE 10 00 02 47 00 00
2025-01-05 15:53:57.131 TRACE 806: [emsesp] boiler(0x08) B all(0x00), UBAMonitorFastPlus(0xE4), data: 00 01 CF 01 BF 1D 50 (offset 35)
2025-01-05 15:53:57.321 TRACE 807: [emsesp] boiler(0x08) B all(0x00), UBAMonitorSlowPlus2(0xE3), data: 01 00 01 00 00 00 00 00 00 00 00 02 48 1D 64 46 00 3A 00 00
2025-01-05 15:53:57.525 TRACE 808: [emsesp] boiler(0x08) B all(0x00), UBAMonitorWWPlus(0xE9), data: 00 80 00 80 00 00 00 00 00 46 3C 00 00 00 00 00 00 00 00 00 00 00 00 05 02 00
2025-01-05 15:53:59.024 TRACE 809: [emsesp] thermostat(0x10) B all(0x00), RC300Monitor(0x02A5), data: 80 00 01 2C 3A 00 2C 2C 05 A0 02 03 01 01 AB 01 D9 00 00 11 01 00 FF FF
2025-01-05 15:53:59.254 TRACE 810: [emsesp] thermostat(0x10) B all(0x00), RC300Monitor(0x02A5), data: 00 00 00 00 00 00 00 FF 64 4B 00 3C 01 FF 01 02 00 00 00 06 5C 09 (offset 24)
2025-01-05 15:53:59.446 TRACE 811: [emsesp] thermostat(0x10) W boiler(0x08), ?(0x02E0), data: 01 3A 64 00 01
2025-01-05 15:53:59.671 TRACE 812: [emsesp] thermostat(0x10) W boiler(0x08), ?(0x02EA), data: 00
2025-01-05 15:53:59.692 TRACE 813: [emsesp] thermostat(0x10) B all(0x00), ?(0xC4), data: 00
2025-01-05 15:53:59.879 TRACE 814: [emsesp] thermostat(0x10) B all(0x00), RC300Floordry(0x0267), data: 00 00

MichaelDvP, 2025-01-05:
This:
```
    "type": "thermostat",
      "name": "BC400-FO/UI800, BC400",
      "deviceID": "0x10",
      "productID": 4,
```
sets the selflowtemp, and will always overwrite any setting from ems-esp or SAT or HA or whatelse.

proddy, 2025-01-05:
I missed that. If you have a physical thermostat attached, SAT will not work as the thermostat will always dominate. The trick is to remove the thermostat and replace it with a simulated (remote) EMS-ESP one/

Ed1954, 2025-01-05:
Just installed a virtual Thermostat (RC100H), targetflowtemp went up to 75°C and selflowtemp to 70°C. remotetemp and remotehum are empty.

2025-01-05 21:49:47.028 INFO 35892: [command] Called command thermostat/control (Fernsteuerung) with value RC100H and id 1 on device 0x10
2025-01-05 21:49:47.028 TRACE 35893: [emsesp] boiler(0x08) B all(0x00), UBAMonitorFastPlus(0xE4), data: 00 01 C9 01 BF 33 50 (offset 35)
2025-01-05 21:49:47.233 TRACE 35894: [emsesp] boiler(0x08) B all(0x00), UBAMonitorSlowPlus2(0xE3), data: 01 00 01 00 00 00 00 00 00 00 00 02 C2 33 64 46 00 46 00 00
2025-01-05 21:49:47.507 TRACE 35895: [emsesp] boiler(0x08) B all(0x00), UBAMonitorWWPlus(0xE9), data: 00 80 00 80 00 00 00 00 00 46 3C 00 00 00 00 00 00 00 00 00 00 00 00 05 02 00
2025-01-05 21:49:47.730 TRACE 35896: [emsesp] me(0x0B) W thermostat(0x10), HPMode(0x0291), data: 03 (offset 3)
2025-01-05 21:49:48.015 TRACE 35897: [emsesp] thermostat(0x10) B all(0x00), HPMode(0x0291), data: 03 (offset 3)
2025-01-05 21:49:48.039 INFO 35898: [command] Called command thermostat/remotetemp (Raumtemperatur Remote) with value 20.4 and id 1 on device 0x10
2025-01-05 21:49:48.227 INFO 35899: [command] Called command thermostat/remotehum (Raumfeuchte Remote) with value 58 and id 1 on device 0x10
2025-01-05 21:49:48.227 INFO 35900: [command] Called command thermostat/seltemp (gewählte Raumtemperatur) with value 21 and id 1 on device 0x10
2025-01-05 21:49:48.392 TRACE 35901: [emsesp] me(0x0B) W thermostat(0x10), RC300Set(0x02B9), data: 2A (offset 10)
2025-01-05 21:49:49.942 TRACE 35902: [emsesp] thermostat(0x10) B all(0x00), RC300Set(0x02B9), data: 2A (offset 10)
2025-01-05 21:49:50.501 TRACE 35903: [emsesp] me(0x0B) R thermostat(0x10), RC300Monitor(0x02A5), length: 25
2025-01-05 21:49:50.604 TRACE 35904: [emsesp] thermostat(0x10) W me(0x0B), RC300Monitor(0x02A5), data: 00 CC 01 2A 4B 00 2A 2A 05 A0 02 03 01 00 47 03 3D 00 00 11 01 00 FF FF 00
2025-01-05 21:49:50.604 TRACE 35905: [emsesp] me(0x0B) R thermostat(0x10), RC300Monitor(0x02A5), length: 25 (offset 25)
2025-01-05 21:49:50.668 TRACE 35906: [emsesp] thermostat(0x10) W me(0x0B), RC300Monitor(0x02A5), data: 00 00 00 00 00 00 0C 64 4B 00 3C 01 3A 01 02 00 00 00 06 5C 09 (offset 25)
2025-01-05 21:49:51.975 INFO 35907: [command] Called command boiler/selflowtemp (Gewählte Vorlauftemperatur) with value 49.50
2025-01-05 21:49:52.145 TRACE 35908: [emsesp] me(0x0B) W boiler(0x08), UBASetPoints(0x1A), data: 31
2025-01-05 21:49:54.273 TRACE 35909: [emsesp] me(0x0B) R boiler(0x08), UBAMonitorFastPlus(0xE4), length: 27
2025-01-05 21:49:54.308 TRACE 35910: [emsesp] boiler(0x08) W me(0x0B), UBAMonitorFastPlus(0xE4), data: 10 20 2D 2D 00 C8 46 02 C3 64 3F 03 00 02 63 00 00 01 E7 02 F1 10 00 02 C3 00 00
2025-01-05 21:49:54.308 TRACE 35911: [emsesp] me(0x0B) R boiler(0x08), UBAMonitorFastPlus(0xE4), length: 27 (offset 27)
2025-01-05 21:49:54.308 TRACE 35912: [emsesp] boiler(0x08) W me(0x0B), UBAMonitorFastPlus(0xE4), data: 00 01 C9 01 BF 33 50 (offset 35)
2025-01-05 21:49:54.308 TRACE 35913: [emsesp] me(0x0B) R boiler(0x08), UBAMonitorFastPlus(0xE4), length: 27 (offset 42)
2025-01-05 21:49:54.308 TRACE 35914: [emsesp] boiler(0x08) W me(0x0B), UBAMonitorFastPlus(0xE4), data: <empty> (offset 42)
2025-01-05 21:49:55.158 TRACE 35915: [emsesp] thermostat(0x38) W thermostat(0x10), RemoteTemp(0x042B), data: 00 CC
2025-01-05 21:49:55.759 TRACE 35916: [emsesp] thermostat(0x38) W thermostat(0x10), RemoteHumidity(0x047B), data: 0C 3A 00 76
2025-01-05 21:49:56.854 TRACE 35917: [emsesp] boiler(0x08) B all(0x00), UBAMonitorFastPlus(0xE4), data: 10 20 2D 2D 00 C8 46 02 C3 64 3F 03 00 02 63 00 00 01 E8 02 F4 10 00 02 C3 00 00
2025-01-05 21:49:57.031 TRACE 35918: [emsesp] boiler(0x08) B all(0x00), UBAMonitorFastPlus(0xE4), data: 00 01 C9 01 BF 33 50 (offset 35)
2025-01-05 21:49:57.221 TRACE 35919: [emsesp] boiler(0x08) B all(0x00), UBAMonitorSlowPlus2(0xE3), data: 01 00 01 00 00 00 00 00 00 00 00 02 C2 33 64 46 00 46 00 00
2025-01-05 21:49:57.496 TRACE 35920: [emsesp] boiler(0x08) B all(0x00), UBAMonitorWWPlus(0xE9), data: 00 80 00 80 00 00 00 00 00 46 3C 00 00 00 00 00 00 00 00 00 00 00 00 05 02 00
2025-01-05 21:49:59.491 TRACE 35921: [emsesp] thermostat(0x10) B all(0x00), RC300Monitor(0x02A5), data: 00 CC 01 2A 4B 00 2A 2A 05 A0 02 03 01 00 47 03 3D 00 00 11 01 00 FF FF
2025-01-05 21:49:59.721 TRACE 35922: [emsesp] thermostat(0x10) B all(0x00), RC300Monitor(0x02A5), data: 00 00 00 00 00 00 00 0C 64 4B 00 3C 01 3A 01 02 00 00 00 06 5C 09 (offset 24)
2025-01-05 21:49:59.914 TRACE 35923: [emsesp] thermostat(0x10) W boiler(0x08), ?(0x02E0), data: 01 4B 64 00 01
2025-01-05 21:50:00.183 TRACE 35924: [emsesp] thermostat(0x10) W boiler(0x08), ?(0x02EA), data: 00
2025-01-05 21:50:00.204 TRACE 35925: [emsesp] thermostat(0x10) B all(0x00), ?(0xC4), data: 00
2025-01-05 21:50:00.390 TRACE 35926: [emsesp] thermostat(0x10) B all(0x00), RC300Floordry(0x0267), data: 00 00
2025-01-05 21:50:06.915 TRACE 35927: [emsesp] boiler(0x08) B all(0x00), UBAMonitorFastPlus(0xE4), data: 10 20 2D 2D 00 C8 46 02 C3 64 3F 03 00 02 63 00 00 01 E8 02 F4 10 00 02 C3 00 00
2025-01-05 21:50:07.090 TRACE 35928: [emsesp] boiler(0x08) B all(0x00), UBAMonitorFastPlus(0xE4), data: 00 01 CA 01 C0 32 50 (offset 35)
2025-01-05 21:50:07.276 TRACE 35929: [emsesp] boiler(0x08) B all(0x00), UBAMonitorSlowPlus2(0xE3), data: 01 00 01 00 00 00 00 00 00 00 00 02 C2 32 64 46 00 46 00 00
[... cut ...]
2025-01-05 21:50:19.306 TRACE 35963: [emsesp] me(0x0B) W thermostat(0x10), RC300Set(0x02B9), data: 2A (offset 10)
2025-01-05 21:50:20.433 TRACE 35964: [emsesp] thermostat(0x10) B all(0x00), RC300Set(0x02B9), data: 2A (offset 10)
2025-01-05 21:50:21.410 TRACE 35965: [emsesp] me(0x0B) R thermostat(0x10), RC300Monitor(0x02A5), length: 25
2025-01-05 21:50:21.512 TRACE 35966: [emsesp] thermostat(0x10) W me(0x0B), RC300Monitor(0x02A5), data: 00 CC 01 2A 4B 00 2A 2A 05 A0 02 03 01 00 46 03 3E 00 00 11 01 00 FF FF 00

Ed1954, 2025-01-06:
[image]
[image]
[image]

Ed1954, 2025-01-09:
I don`t see any possibility to remove the internal Thermostat BC400.  So i can`t use SAT with Buderus GB182 or i set heatingtemp instead of selflowtemp.

proddy, 2025-01-30:
correct, SAT will only work when there is no physical thermostat on the EMS bus