emsesp/EMS-ESP32 #2333: IPv6 entry is spamming logs: gets logged every three seconds with 3.7.2-dev.8

paludi, 2025-01-06:
### DESCRIPTION

IPv6 entry is spamming logs: gets logged about every three seconds with 3.7.2-dev.8 

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x ] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x ] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
  --> attached
[system.json](https://github.com/user-attachments/files/18313913/system.json)

```

### TO REPRODUCE

- upgrade to the latest dev release: 3.7.2-dev.8 
- go to log viewer
(same behavior after restart)

### EXPECTED BEHAVIOUR

IPv6 should be logged only once.

### SCREENSHOTS

example from the log viewer: 
2025-01-06 06:28:19.074 I 26: [emsesp] IPv6 (Eth) local: fe80::8ece:4eff:fe96:4e87
2025-01-06 06:28:22.074 I 27: [emsesp] IPv6 (Eth) local: fe80::8ece:4eff:fe96:4e87
2025-01-06 06:28:25.075 I 28: [emsesp] IPv6 (Eth) local: fe80::8ece:4eff:fe96:4e87
2025-01-06 06:28:28.074 I 29: [emsesp] IPv6 (Eth) local: fe80::8ece:4eff:fe96:4e87
2025-01-06 06:28:31.075 I 30: [emsesp] IPv6 (Eth) local: fe80::8ece:4eff:fe96:4e87
2025-01-06 06:28:34.074 I 31: [emsesp] IPv6 (Eth) local: fe80::8ece:4eff:fe96:4e87
2025-01-06 06:28:37.074 I 32: [emsesp] IPv6 (Eth) local: fe80::8ece:4eff:fe96:4e87
2025-01-06 06:28:40.001 I 33: [emsesp] IPv6 (Eth) local: fe80::8ece:4eff:fe96:4e87
2025-01-06 06:28:43.074 I 34: [emsesp] IPv6 (Eth) local: fe80::8ece:4eff:fe96:4e87
2025-01-06 06:28:46.074 I 35: [emsesp] IPv6 (Eth) local: fe80::8ece:4eff:fe96:4e87
2025-01-06 06:28:49.074 I 36: [emsesp] IPv6 (Eth) local: fe80::8ece:4eff:fe96:4e87
2025-01-06 06:28:52.074 I 37: [emsesp] IPv6 (Eth) local: fe80::8ece:4eff:fe96:4e87
2025-01-06 06:28:55.075 I 38: [emsesp] IPv6 (Eth) local: fe80::8ece:4eff:fe96:4e87
2025-01-06 06:28:58.074 I 39: [emsesp] IPv6 (Eth) local: fe80::8ece:4eff:fe96:4e87
2025-01-06 06:29:01.074 I 40: [emsesp] IPv6 (Eth) local: fe80::8ece:4eff:fe96:4e87
2025-01-06 06:29:04.074 I 41: [emsesp] IPv6 (Eth) local: fe80::8ece:4eff:fe96:4e87
2025-01-06 06:29:07.076 I 42: [emsesp] IPv6 (Eth) local: fe80::8ece:4eff:fe96:4e87
2025-01-06 06:29:10.002 I 43: [emsesp] IPv6 (Eth) local: fe80::8ece:4eff:fe96:4e87
2025-01-06 06:29:13.074 I 44: [emsesp] IPv6 (Eth) local: fe80::8ece:4eff:fe96:4e87
2025-01-06 06:29:16.075 I 45: [emsesp] IPv6 (Eth) local: fe80::8ece:4eff:fe96:4e87
2025-01-06 06:29:19.074 I 46: [emsesp] IPv6 (Eth) local: fe80::8ece:4eff:fe96:4e87

### ADDITIONAL CONTEXT

-

proddy, 2025-01-07:
strange. Do you know if it happened with previous versions?

paludi, 2025-01-07:
I just installed 3.7.2-dev.6 which I downloaded on Dec 21 and it does not have the issue. 
This version was before the local date/time issue that crashed the status page.
Going back to 3.7.2-dev.8, it happens again. Let me know if I can help you with anything else.

proddy, 2025-01-08:
I think it's because of the latest Espressif Arduino library (3.1.0). I'll roll this back into a dev-9 and see if that helps.

closed 2025-01-08 by proddy

paludi, 2025-01-08:
Thanks @proddy! dev-9 fixes the issue.

VlastiBroucek, 2025-01-09:
I am on dev-8 and don't see the issue... Just saying... 😁

commit 74cb23a8 (2025-01-08): replace arduino platform 3.1.0 with previous 2.0.18 to see if it fixed #2333