emsesp/EMS-ESP32 #2335: automatic firmware upload via web fails on 4MB boards

proddy, 2025-01-08:
HttpClient get returns -1. Works on 16MB so not Tasmota framework related.

mattreim, 2025-01-08:
A little feedback on the `v3.7.2-dev.9` the message is almost the same. But what's actually worse is that every time I try it, the data traffic stops and it has to be restarted manually.

```lua
2025-01-08 23:14:24.478   ERROR 11:     [system]     Firmware upload failed. URL https://github.com/emsesp/EMS-ESP32/releases/download/latest/EMS-ESP-3_7_2-dev_9-ESP32-4MB.bin, HTTP code -1
```

proddy, 2025-01-08:
yes I know, it's not fixed and there's a small chance I'll need to remove this feature for 4MB boards. Do you know if it ever worked?

mattreim, 2025-01-08:
I've tried it several times and it's never worked.

proddy, 2025-01-10:
I think this is because we use a different Arduino/Espressif library based on Tasmota which has the mbedtsl/ssl library removed and the HttpClient library needs this to connect to https://github... 

What I'll do is remove the Install feature for these boards. Users will need to manually update.

proddy, 2025-01-12:
removed in dev-10

closed 2025-01-12 by proddy