emsesp/EMS-ESP32 #2336: Radio Controlled Thermostats with MX300

Magnumandy, 2025-01-08:
Hi there,
I do have the problem that the Bosch radio controlled thermostats, which are connected to my MX300 (Buderus GB182i.2) and controllable through the MyBuderus App, do not show up through the EMS Gateway.
As you can see in the attached screenshot, there is the item "MX300/Wireless sensor base", but there are ZERO Entities found or listed. I was assuming that the Thermostats would show up here. Anyone having an idea perhaps?

Thanks a lot,
 Andy

[image]

MichaelDvP, 2025-01-10:
Thomas (@tp1de) has checked the MX300 a while ago and found that it is cloud only access and does not provide any data to bus or lan.
see: https://github.com/tp1de/ioBroker.ems-esp/issues/25

tp1de, 2025-01-10:
MX300 is not seen by Bosch/Buderus original API as well, since they do not support local LAN-access anymore.
Just cloud communication is supported.

The MX300 is the successor of KM200/IP-inside Gateways and is normally plugged into thermostats.
Some thermostats are EMS-bus connected - others are not.
@Magnumandy which thermostat are you using?

Magnumandy, 2025-01-10:
Thanks Michael for the information. That sounds sad indeed. Although I have found the thermostats being listed under "Funk-Komponenten" (wireless components) in the BC400. So I would have expected them with the EMS-ESP under "MX300/Wireless Sensor Base" (see my picture from original post).
Thanks again for your help, Andreas

Magnumandy, 2025-01-10:
Hi Thomas, thanks for your message, which overlapped with my response to Michael. I was hoping to find a solution to control my connected thermostats using the Gateway. Sad situation that Bosch/Buderus is not giving any possibilities here...

tp1de, 2025-01-10:
You are talking about Bosch Smart Home Thermostats for radiators? 
These are not ems devices but might be integrated by home automation systems like homeassistant or ioBroker.

Magnumandy, 2025-01-10:
Yes, these are the ones I am referring to. However, mine are connected with my MX300, BC400, GB182.2i Heating, so my thinking was that one can pull information out through the EMS Gateway for these devices. Obviously, this does not work. My research on controlling the Radiator Thermostats any other way (an API or something similar) has not been successful so far. In the old days, there was an API (for my former GAS Heating), but seemingly not for the new one...

tp1de, 2025-01-10:
The Bosch Smart Home Controller which is seen by ems-esp can be integrated in homeassistant and ioBroker with the appropriate integrations / adapters. Do you use any home automation system?

Magnumandy, 2025-01-10:
Yes, I use Homeseer HS4 (for legacy reasons) and I want to setup a Home Assistant environment in parallel. So far, I did not find any way to connect to the thermostat devices without decoupling them from my Buderus environment. That sounds interesting indeed. Do you have more information on that?

tp1de, 2025-01-10:
https://www.home-assistant.io/integrations/bosch_shc/

Magnumandy, 2025-01-10:
I do have the Bosch smart Home controller, but that one is not able to communicate with my Buderus Heating. So the brilliant idea to control the Gas Heater specifically depending on the heating demand of every single room is not working with that device. I tried that a year ago and at that time, there was no possibility at least...

tp1de, 2025-01-10:
That is what I am doing using ems-esp and smart thermostats from homematic integrated into homeassistant.
(I was using ioBroker before).
Using the mentioned integration it should work wit bosch thermostats as well.

tp1de, 2025-01-10:
> Do you do something similar?

No I use outside temp controlled flow temp and then calculate the actual heatdemand per room.
This heatdemand is different per room (factor depending on size of radiator).
When sum of heatdemand is > defined level I switch the heating circuit on (automatic mode) - otherwise off.
I did a hydraulic optimization on my heating system some years ago. So the heating curve works as desired.

If it really works with the Buderus function to align the boiler flowtemp according to the thermostats this might be an advantage.
But you can still implement a function to switch off if there is  too low heat demand to avoid takting.

Magnumandy, 2025-01-10:
Yes, you are right. Thanks a lot for your thoughts and information. I might still be able, as you say, to switch off the heater if, e.g. the weather forecast predicts sunshine in the next hour, or so. Because this is, what I wanted to do that, depending on the hourly forecast, the temperature in the HC is adjusted.
Thanks for your good feedback!
All the best, Andreas

tp1de, 2025-01-10:
you are welcome

closed 2025-03-22 by proddy