emsesp/EMS-ESP32 #2339: Issue with commands when using MQTT base with paths

Hisselmissel, 2025-01-09:
### DESCRIPTION

I’m encountering an issue with the MQTT integration where commands fail to work if an MQTT base with paths is used (e.g. ems-esp/esp32-12345). Receiving data works perfectly in this configuration, but commands result in an error.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [x] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
  {"system":{"version":"3.7.2-dev.9","uptime":"000+00:14:56.581","uptimeSec":896,"platform":"ESP32","cpuType":"ESP32-D0WD-V3","arduino":"ESP32 Arduino v2.0.17","sdk":"v4.4.7-dirty","freeMem":140,"maxAlloc":99,"freeCaps":4165,"usedApp":2394,"freeApp":2278,"partition":"app0","resetReason":"Software reset CPU / Software reset CPU","psram":true,"psramSize":4093,"freePsram":4060,"model":""},"network":{"network":"WiFi","hostname":"ems-esp","RSSI":-58,"WIFIReconnects":0,"TxPowerSetting":0,"staticIP":false,"lowBandwidth":false,"disableSleep":true,"enableMDNS":true,"enableCORS":false,"APProvisionMode":"disconnected","APSecurity":"wpa2","APSSID":"ems-esp"},"ntp":{"NTPStatus":"connected","enabled":true,"server":"time.google.com","tzLabel":"Europe/Amsterdam"},"mqtt":{"MQTTStatus":"connected","MQTTPublishes":50,"MQTTQueued":0,"MQTTPublishFails":0,"MQTTReconnects":0,"enabled":true,"clientID":"esp32-8fc964ec","keepAlive":60,"cleanSession":false,"entityFormat":3,"base":"ems-esp/esp32-8fc964ec","discoveryPrefix":"gee","discoveryType":0,"nestedFormat":1,"haEnabled":false,"mqttQos":1,"mqttRetain":false,"publishTimeHeartbeat":60,"publishTimeBoiler":60,"publishTimeThermostat":60,"publishTimeSolar":300,"publishTimeMixer":300,"publishTimeWater":300,"publishTimeOther":300,"publishTimeSensor":300,"publishSingle":false,"publish2command":false,"sendResponse":false},"syslog":{"enabled":false},"sensor":{"temperatureSensors":0,"temperatureSensorReads":0,"temperatureSensorFails":0,"analogSensors":0,"analogSensorReads":0,"analogSensorFails":0},"api":{"APICalls":0,"APIFails":0},"bus":{"busStatus":"connected","busProtocol":"HT3","busTelegramsReceived":1362,"busReads":440,"busWrites":0,"busIncompleteTelegrams":0,"busReadsFailed":0,"busWritesFailed":0,"busRxLineQuality":100,"busTxLineQuality":100},"settings":{"boardProfile":"E32V2","locale":"en","txMode":1,"emsBusID":11,"showerTimer":false,"showerMinDuration":180,"showerAlert":false,"hideLed":false,"noTokenApi":false,"readonlyMode":false,"fahrenheit":false,"dallasParasite":false,"boolFormat":4,"boolDashboard":3,"enumFormat":1,"analogEnabled":true,"telnetEnabled":false,"maxWebLogBuffer":50,"modbusEnabled":false,"forceHeatingOff":false,"developerMode":false},"devices":[{"type":"boiler","name":"CBS/GB*72, Trendline, Cerapur, Greenstar Si","deviceID":"0x08","productID":123,"brand":"","version":"06.03","entities":80,"handlersReceived":"0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A 0x04","handlersFetched":"0x14 0x16 0x33 0x26 0x28","handlersPending":"0xBF 0xC2 0x1A 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x2E 0x3B","handlersIgnored":"0x23"},{"type":"thermostat","name":"CW400/RC3*0, Moduline 3000/1010H, CW400, Sense II, HPC410","deviceID":"0x10","productID":158,"brand":"","version":"74.02","entities":59,"handlersReceived":"0x06 0xA2 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267","handlersFetched":"0x02A5 0x02B9 0x02AF 0x029B 0x0291 0x0292 0x0293 0x0294 0x02F5 0x02F6 0x031B 0x023A 0x0240 0x0269","handlersPending":"0xA3 0x12 0x13 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 0x02A2 0x0478 0x02CC 0x0467 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031E 0xBB 0x023E 0x016E","handlersIgnored":"0xBF"},{"type":"controller","name":"Controller","deviceID":"0x09","productID":152,"brand":"","version":"03.01","entities":0}]}
```

### TO REPRODUCE

- Configure MQTT base: ems-esp/esp32-12345
- Check if data is published correctly to topic ems-esp/esp32-12345/boiler_data (via MQTT Explorer)
- Publish command to ems-esp/esp32-12345/boiler/seltemp with payload 28.0 (via MQTT Explorer)
- System log shows the following line:
 ```2025-01-09 15:19:36.818 E 60: [mqtt] MQTT command failed with error invalid path ems-esp/esp32-12345/boiler/seltemp (Error) ```

### EXPECTED BEHAVIOUR

- System log should show the following line after publishing the command:
```2025-01-09 15:15:41.993 I 56: [command] Called command boiler/seltemp (selected temperature) with value 28```
Which works fine if I use a base like esp32-12345, but not with paths.

### SCREENSHOTS

[image]

proddy, 2025-01-09:
very concise, thanks! We'll look into it.

Hisselmissel, 2025-01-10:
It works. Thank you very much for the quick fix!

closed 2025-01-10 by Hisselmissel