emsesp/EMS-ESP32 #2342: Bosch heat pump parameter silentMode is defined as boolean but has also information about level which is missing when transferred to boolean

ajustus83, 2025-01-10:
currently the parameter silent Mode of device XCU_THH/CS*800i, Logatherm WLW* (Boiler/HP) is defined as boolean.
Also in this parameter is the level of reduction:
6: 60% Reduction
5: 50% Reduction
4: 40% Reduction
3: 30% Reduction

the parameter is writable
parameter details:
device-ID: 0x08 Type-ID: 0x484, Position: 64
000+01:19:02.348 TRACE 3218: [emsesp] boiler(0x08) -> all(0x00), B, HPSilentMode(0x0484), data: 06 (offset 64)

MichaelDvP, 2025-01-10:
Implemented as `powereduction`

ajustus83, 2025-01-10:
I see the parameter, but it is not changeable, for this I reopened this issue.
the parameter should be changable. I changed the title

MichaelDvP, 2025-01-10:
I see a command here:
https://github.com/emsesp/EMS-ESP32/blob/fb44bc33b98514c2496d0527bd45c5b8678eb978/src/devices/boiler.cpp#L558-L566
Please attach a log when changing the parameter with the thermostat and a second changing with emsesp.

ajustus83, 2025-01-10:
000+01:17:41.731 TRACE 2754: [emsesp] thermostat(0x10) -> boiler(0x08), W, HPSilentMode(0x0484), data: 05 (offset 64)
000+01:17:41.806 TRACE 2755: [emsesp] boiler(0x08) -> all(0x00), B, HPSilentMode(0x0484), data: 05 (offset 64)

this is when I do a change with the thermostat

MichaelDvP, 2025-01-11:
Please specify exactly what you mean with "is not changeable". I can't find an error in code and have no heatpump to test.

BastyJuice, 2025-02-06:
i can confirm, i also have CS5800i with XCU_THH/CS*800i, Logatherm WLW* (Boiler/HP) a lot of Features are not changeable

also not changeable

- selected flow temperature
- dhw circulation active

proddy, 2025-02-09:
> i can confirm, i also have CS5800i with XCU_THH/CS_800i, Logatherm WLW_ (Boiler/HP) a lot of Features are not changeable
> 
> also not changeable
> 
> * selected flow temperature
> * dhw circulation active

can you describe what happens?

BastyJuice, 2025-02-14:
hey @proddy nothing it does not change the value

proddy, 2025-02-15:
https://docs.emsesp.org/Troubleshooting/#changing-a-value-works-at-first-but-is-then-reset-to-its-original-value

closed 2025-04-05 by github-actions[bot]