emsesp/EMS-ESP32 #2349: Change (temporarily) Heating Temperature (but not the permanent heat/eco/nofrost temperatures)

bergteam, 2025-01-12:
I am not sure if this a Bug or a Request for Improvement.

### DESCRIPTION
I have a Junkers Boiler with FW120 Thermostat.
The Thermostat defines the heat/eco/nofrost temperatures and changes between them based on the daily schedule. The result is stored under "hc1 selected room temperature". I include sample temperatures in order to better describe my point.

a) Current Setting
    + hc1 heat temperature 21,5 °C
    + hc1 eco temperature 14,0 °C
    + hc1 nofrost temperature 5,0 °C
    + hc1 selected room temperature 21,5 °C

I would like to **temporarily set the temperature to 21,0°C** but not change the permanent (heat) temperature.
b) on the Thermostat FW120 I turn the knob and the following setting appears.
    + hc1 heat temperature 21,5 °C                   <= this does not change
    + hc1 eco temperature 14,0 °C
    + hc1 nofrost temperature 5,0 °C
    _+ hc1 selected room temperature 21,0 °C    <=only this_
=> wanted behavior.

c) if I try the same on EMS-ESP by changing "hc1 selected room temperature" the following appears. Note the programm is currently in mode "heat".
    _+ hc1 heat temperature 21,0 °C                   <= temp also changed_
    + hc1 eco temperature 14,0 °C
    + hc1 nofrost temperature 5,0 °C
    _+ hc1 selected room temperature 21,0 °C_
=> not wanted behavior. Why was permanent "hc1 heat temperature" changed?

### CURRENT BEHAVIOR
When changing the "hc1 selected room temperature" value also the permanent value of the current heating mode is changed.

### EXPECTED BEHAVIOUR
When changing the "hc1 selected room temperature" only this value is changed.

### SOFTWARE/HARDWARE
Version            3.7.2-dev.10
Platform          ESP32-16MB  (-PSRAM)
Release Type   Development

MichaelDvP, 2025-01-12:
Most thermostats have a special field in the setting telegrams to set the temporary value and we use it for seltemp. But sadly for the junkers thermostats we have not found a way to set the temporary setpoint. 
It's published in telegram 0x16F, offset 2 (int16), but the settings 0x165, 0x179 only have the settings for the permanent values. 
If you find a way to set the temporary value via ems-bus we are happy to implement it.

bergteam, 2025-01-13:
**Maybee someone from the community has the FB10 - Remote Thermostat - installed and could log it.** According to the User Manual this device does control "hc1 selected room temperature" on the main thermostat (e.g FW120). I will also try to get hold of one.

Screenshot from FB10 - Installations- und Bedienungsanleitung
[image]

MichaelDvP, 2025-01-13:
Afair this is telegram 0x123 offset 2, but it has to be sent from a remote device: 0x18 for hc1, 0x19 for gc2, etc.
Don't know if we have to register the remote to the bus before sending.

You can try with send raw telegram: `call system send "18 10 FF 02 00 23 2A"` for 21 degrees seltemp (0x2A = 42 => 21°)

Sbried, 2025-01-13:
What I know, maybe that helps:
0x0122 is the control telegram sent from FWxxx to the FB10.
Offset 0: RoomTempInfluence enabled=-1, disabled=0
Offset 1: Requested room temp used by FB10 with active room influence, int8 range 0...60, resolution 0,5°C
Offset 2: Requested room temperature offset used by FB10 with inactive room influence, int8 range -10...10, resolution 0,5°C

0x0123 is the control telegram sent from FB10 to the FW1xxx heating circuit.
Offset 0: Current room temperature int16, resolution 0,1°C
Offset 1: Requested room temp sent by FB10 with active room influence, int8 range 0...60, resolution 0,5°C
Offset 2: Requested room temperature offset sent by FB10 with inactive room influence, int8 range -10...10, resolution 0,5°C

bergteam, 2025-01-13:
That sounds promising. But unfortunately made no difference in any temp values set.

My Settings
+ hc1 control device = FB10
+ hc1 room influence = intern (also tried setting to extern, not sure about the impact, nothing observed)
+ FB10 => hc1 RF room temperature sensor 21,8 °C (ems-esp.0.thermostat.hc1.remotetemp) filled with iobroker 

[image]

Configured as suggested
[image]

Settings done. 
1. Send 0x0123 with Offset 1 and 2 set to 5 => Succesfully executed. No change on any settings
2. Changed "hc1 room influence" from intern (default) to extern
3. Send 0x0123 with Offset 1 and 2 set to 5 => Succesfully executed. No change on any settings
[image]

Sbried, 2025-01-14:
It would be better to log the raw data to see what happens.
Before/after sending to 0x123 you could read 0x0122 from the FW120 to see if something changed.

closed 2025-03-22 by proddy