emsesp/EMS-ESP32 #2369: Found few missing translation

misa1515, 2025-01-22:
### DESCRIPTION

_A clear and concise description of what the problem is or the change requested._

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [x] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json

### SCREENSHOTS  

![Image](https://github.com/user-attachments/assets/31454f0c-ba44-4afa-9942-d6e72676dc60)

![Image](https://github.com/user-attachments/assets/1ca73368-d9b0-4988-a34b-46bd3fd26fc4)

```

proddy, 2025-01-30:
this is fixed in later 3.7.2 releases. See demo at https://demo.emsesp.org/settings/version

misa1515, 2025-02-02:
Please check it again, still missing translation for marked text on first picture. 
"pletform" and "release type"

proddy, 2025-02-02:
> Please check it again, still missing translation for marked text on first picture. "pletform" and "release type"

you're right. I'll add them!

misa1515, 2025-02-02:
thank you for feedback.

proddy, 2025-02-03:
added

closed 2025-02-03 by proddy

proddy, 2025-02-11:
oh yeah, will fix that