emsesp/EMS-ESP32 #2372: SYSLOG: TX status of message sending

mirino75, 2025-01-25:
### DESCRIPTION

Different TX Status of message sending in SYSLOG server and EMS log

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
  Paste System information here....

```

### TO REPRODUCE

If I set the temperature via HA, everything goes OK, the temperature changes on the boiler and thermostat, but in the syslog server the TX status is ERROR (rejected). In the EMS status log the TX log is OK.

### EXPECTED BEHAVIOUR

_A clear and concise description of what you expected to happen._

### SCREENSHOTS

[image]
[image]

### ADDITIONAL CONTEXT

_Add any other context about the issue here._

**(Please remember to close the issue when it has been addressed)**

proddy, 2025-01-25:
will take a look

MichaelDvP, 2025-01-25:
If a write is not successful, emsesp retries 3 times and then set a tx-error. Here we don't see retry fails, the  command is executed, so error counter is still zero.

proddy, 2025-01-25:
maybe those errors should be warnings to avoid panic

closed 2025-01-30 by proddy