emsesp/EMS-ESP32 #2374: Boiler temperature is not displayed correctly

fumagallis4ever, 2025-01-25:
###DESCRIPTION

I have a Junkers Cerastar and was previously using the KM 200 LAN Gateway but have switched to the EMS Gateway. In iobroker, I am getting an incorrect value for the boiler temperature.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [x] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the System information in the area below, taken from `http://<IP>/api/system`

### SCREENSHOTS
KM200
[image]

EMS-Gateway

[image]

proddy, 2025-01-25:
which value is correct? 380 looks large.

What does EMS-ESP say (not ioBroker)?

BTW you ticked the box "Provide the System information in the area below, taken from http://<IP>/api/system" but it's missing.

fumagallis4ever, 2025-01-27:
I think 380 degrees is the correct value

[image]

{
  "system": {
    "version": "3.7.2-dev.15",
    "uptime": "002+06:36:45.362",
    "uptimeSec": 196605,
    "platform": "ESP32",
    "cpuType": "ESP32-D0WD-V3",
    "arduino": "ESP32 Arduino v2.0.17",
    "sdk": "v4.4.7-dirty",
    "freeMem": 161,
    "maxAlloc": 95,
    "freeCaps": 4197,
    "usedApp": 2407,
    "freeApp": 2265,
    "partition": "app1",
    "flash_chip_size": 16384,
    "resetReason": "Power on reset / APP CPU reset by PRO CPU",
    "psram": true,
    "psramSize": 4093,
    "freePsram": 4070,
    "model": ""
  },
  "network": {
    "network": "Ethernet",
    "hostname": "ems-esp",
    "TxPowerSetting": 0,
    "staticIP": true,
    "lowBandwidth": false,
    "disableSleep": false,
    "enableMDNS": true,
    "enableCORS": false,
    "APProvisionMode": "disconnected",
    "APSecurity": "wpa2",
    "APSSID": "ems-esp"
  },
  "ntp": {
    "NTPStatus": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tzLabel": "Europe/Berlin"
  },
  "mqtt": {
    "MQTTStatus": "disconnected",
    "MQTTPublishes": 4769,
    "MQTTQueued": 0,
    "MQTTPublishFails": 0,
    "MQTTReconnects": 1,
    "enabled": true,
    "clientID": "",
    "keepAlive": 60,
    "cleanSession": true,
    "entityFormat": 0,
    "base": "ems-esp",
    "discoveryPrefix": "homeassistant",
    "discoveryType": 0,
    "nestedFormat": 1,
    "haEnabled": true,
    "mqttQos": 0,
    "mqttRetain": false,
    "publishTimeHeartbeat": 60,
    "publishTimeBoiler": 10,
    "publishTimeThermostat": 10,
    "publishTimeSolar": 10,
    "publishTimeMixer": 10,
    "publishTimeWater": 10,
    "publishTimeOther": 60,
    "publishTimeSensor": 10,
    "publishSingle": false,
    "publish2command": false,
    "sendResponse": false
  },
  "syslog": {
    "enabled": false
  },
  "sensor": {
    "temperatureSensors": 0,
    "temperatureSensorReads": 0,
    "temperatureSensorFails": 0
  },
  "api": {
    "APICalls": 9912,
    "APIFails": 0
  },
  "bus": {
    "busStatus": "connected",
    "busProtocol": "HT3",
    "busTelegramsReceived": 146847,
    "busReads": 26211,
    "busWrites": 2,
    "busIncompleteTelegrams": 0,
    "busReadsFailed": 3,
    "busWritesFailed": 0,
    "busRxLineQuality": 100,
    "busTxLineQuality": 100
  },
  "settings": {
    "boardProfile": "E32V2",
    "locale": "de",
    "txMode": 1,
    "emsBusID": 11,
    "showerTimer": false,
    "showerMinDuration": 180,
    "showerAlert": false,
    "hideLed": false,
    "noTokenApi": false,
    "readonlyMode": false,
    "fahrenheit": false,
    "dallasParasite": false,
    "boolFormat": 6,
    "boolDashboard": 5,
    "enumFormat": 1,
    "analogEnabled": false,
    "telnetEnabled": false,
    "maxWebLogBuffer": 50,
    "modbusEnabled": false,
    "forceHeatingOff": false,
    "developerMode": false
  },
  "devices": [
    {
      "type": "boiler",
      "name": "Condens, Logamax/Logomatic, Cerapur Top, Greenstar, Generic HT3",
      "deviceID": "0x08",
      "productID": 95,
      "brand": "",
      "version": "35.01",
      "entities": 55,
      "handlersReceived": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A 0x04",
      "handlersFetched": "0x14 0x16 0x33",
      "handlersPending": "0xBF 0xC2 0x1A 0x35 0x26 0x2E 0x3B",
      "handlersIgnored": "0x23"
    },
    {
      "type": "thermostat",
      "name": "FW200",
      "deviceID": "0x10",
      "productID": 106,
      "brand": "Junkers",
      "version": "12.14",
      "entities": 31,
      "handlersReceived": "0x06 0x0166",
      "handlersFetched": "0x016F 0x0165 0xBB 0x01D3",
      "handlersPending": "0xA3 0xA2 0x12 0x13 0x0170 0x0171 0x0172 0x0167 0x0168 0x23",
      "handlersIgnored": "0xBE"
    },
    {
      "type": "gateway",
      "name": "KM200, MB LAN 2",
      "deviceID": "0x0D",
      "productID": 189,
      "brand": "",
      "version": "01.10",
      "entities": 0,
      "handlersIgnored": "0x18 0x1C"
    },
    {
      "type": "controller",
      "name": "HT3",
      "deviceID": "0x09",
      "productID": 95,
      "brand": "",
      "version": "35.01",
      "entities": 0
    }
  ]
}

bbqkees, 2025-01-27:
When you see 380 in the KM200, what exact value do you see for this entity in the web interface of the Gateway? Probably 38,0.
ActualSupplyTemperature reads on the current flow temperature entity.
A flow temperature of 380 degrees is incorrect of course.  Likely 38,0 degrees.
The KM200 value is wrong, EMS-ESP is correct.

[image]

MichaelDvP, 2025-01-28:
Critical point of water is 374°C/221bar. So the heating system is well pressurized for a 2200m tall tower and supercritical, wow. 
Send pics.

proddy, 2025-03-22:
closing, stale issue

closed 2025-03-22 by proddy