emsesp/EMS-ESP32 #2378: Duplication of devices

mirino75, 2025-01-28:
### DESCRIPTION

_A clear and concise description of what the problem is or the change requested._

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
  {
  "system": {
    "version": "3.7.1",
    "uptime": "002+23:00:04.857",
    "uptimeSec": 255604,
    "platform": "ESP32S3",
    "cpuType": "ESP32-S3",
    "arduino": "ESP32 Arduino v2.0.17",
    "sdk": "v4.4.7-dirty",
    "freeMem": 183,
    "maxAlloc": 135,
    "freeCaps": 8280,
    "usedApp": 2292,
    "freeApp": 2380,
    "partition": "app0",
    "resetReason": "Software reset CPU / Software reset CPU",
    "psram": true,
    "psramSize": 8189,
    "freePsram": 8098,
    "model": "",
    "temperature": 37.5
  },
  "network": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -72,
    "WIFIReconnects": 0,
    "TxPowerSetting": 0,
    "staticIP": false,
    "lowBandwidth": false,
    "disableSleep": true,
    "enableMDNS": true,
    "enableCORS": false,
    "APProvisionMode": "disconnected",
    "APSecurity": "wpa2",
    "APSSID": "ems-esp"
  },
  "ntp": {
    "NTPStatus": "disconnected",
    "enabled": false,
    "server": "time.google.com",
    "tzLabel": "Europe/Amsterdam"
  },
  "mqtt": {
    "MQTTStatus": "connected",
    "MQTTPublishes": 106779,
    "MQTTQueued": 0,
    "MQTTPublishFails": 0,
    "MQTTReconnects": 0,
    "enabled": true,
    "clientID": "esp32-37304ec0",
    "keepAlive": 60,
    "cleanSession": false,
    "entityFormat": 0,
    "base": "ems-esp",
    "discoveryPrefix": "homeassistant",
    "discoveryType": 0,
    "nestedFormat": 1,
    "haEnabled": true,
    "mqttQos": 0,
    "mqttRetain": false,
    "publishTimeHeartbeat": 60,
    "publishTimeBoiler": 5,
    "publishTimeThermostat": 5,
    "publishTimeSolar": 10,
    "publishTimeMixer": 10,
    "publishTimeWater": 5,
    "publishTimeOther": 60,
    "publishTimeSensor": 5,
    "publishSingle": false,
    "publish2command": false,
    "sendResponse": false
  },
  "syslog": {
    "enabled": true,
    "syslogStarted": true,
    "syslogLevel": "info",
    "syslogIP": "192.168.2.10",
    "syslogQueue": 0
  },
  "sensor": {
    "temperatureSensors": 0,
    "temperatureSensorReads": 0,
    "temperatureSensorFails": 0,
    "analogSensors": 0,
    "analogSensorReads": 0,
    "analogSensorFails": 0
  },
  "api": {
    "APICalls": 166111,
    "APIFails": 0
  },
  "bus": {
    "busStatus": "connected",
    "busProtocol": "Buderus",
    "busTelegramsReceived": 309579,
    "busReads": 81536,
    "busWrites": 19,
    "busIncompleteTelegrams": 264,
    "busReadsFailed": 20,
    "busWritesFailed": 0,
    "busRxLineQuality": 100,
    "busTxLineQuality": 100
  },
  "settings": {
    "boardProfile": "CUSTOM",
    "locale": "en",
    "txMode": 1,
    "emsBusID": 11,
    "showerTimer": true,
    "showerMinDuration": 180,
    "showerAlert": true,
    "showerAlertColdshot": 10,
    "showerAlertTrigger": 7,
    "phyType": 0,
    "rxGPIO": 5,
    "txGPIO": 17,
    "dallasGPIO": 18,
    "pbuttonGPIO": 0,
    "ledGPIO": 2,
    "hideLed": false,
    "noTokenApi": false,
    "readonlyMode": false,
    "fahrenheit": false,
    "dallasParasite": false,
    "boolFormat": 1,
    "boolDashboard": 1,
    "enumFormat": 1,
    "analogEnabled": true,
    "telnetEnabled": false,
    "maxWebLogBuffer": 50,
    "webLogBuffer": 50,
    "modbusEnabled": false,
    "forceHeatingOff": false,
    "developerMode": false
  },
  "devices": [
    {
      "type": "boiler",
      "name": "CBS/GB*72, Trendline, Cerapur, Greenstar Si",
      "deviceID": "0x08",
      "productID": 123,
      "brand": "Buderus",
      "version": "04.09",
      "entities": 79,
      "handlersReceived": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x2A 0x04",
      "handlersFetched": "0x14 0x16 0x33 0x26 0x28",
      "handlersPending": "0xBF 0xC2 0x1A 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x2E 0x3B"
    },
    {
      "type": "thermostat",
      "name": "RC300/RC3*0, Moduline 3000/1010H, CW400, Sense II, HPC410",
      "deviceID": "0x10",
      "productID": 158,
      "brand": "",
      "version": "11.07",
      "entities": 53,
      "handlersReceived": "0x06 0xA2 0x02BA 0x02BB 0x02BC 0x031D 0x0267",
      "handlersFetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0291 0x0292 0x0293 0x0294 0x02F5 0x02F6 0x023A 0x0240 0x0269",
      "handlersPending": "0xA3 0x12 0x13 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CC 0x0467 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031B 0x031E 0xBB 0x023E",
      "handlersIgnored": "0xBF 0xF7"
    },
    {
      "type": "thermostat",
      "name": "RC3*0, Moduline 3000/1010H, CW400, Sense II, HPC410",
      "deviceID": "0x00",
      "productID": 158,
      "brand": "",
      "version": "11.07",
      "entities": 1,
      "handlersReceived": "0x06",
      "handlersPending": "0xA3 0xA2 0x12 0x13 0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02A6 0x02BA 0x02B0 0x029C 0x0472 0x02A7 0x02BB 0x02B1 0x029D 0x0473 0x02A8 0x02BC 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CC 0x0467 0x0291 0x02CE 0x0468 0x0292 0x02D0 0x0469 0x0293 0x02D2 0x046A 0x0294 0x02F5 0x02F6 0x031B 0x031D 0x031E 0x023A 0x0267 0x0240 0xBB 0x023E 0x0269",
      "handlersIgnored": "0xBF"
    },
    {
      "type": "controller",
      "name": "BC25",
      "deviceID": "0x09",
      "productID": 125,
      "brand": "",
      "version": "01.06",
      "entities": 0
    }
  ]
}

```

### TO REPRODUCE

I have 1 boiler and 1 thermostat. The system detected it correctly, but after 4 days I found out that it found another thermostat that only has one entity, the date.

[api_system_new.json](https://github.com/user-attachments/files/18570679/api_system_new.json)
[api_system_old.json](https://github.com/user-attachments/files/18570680/api_system_old.json)

### EXPECTED BEHAVIOUR

_A clear and concise description of what you expected to happen._

### SCREENSHOTS

[image]

[image]

### ADDITIONAL CONTEXT

_Add any other context about the issue here._

**(Please remember to close the issue when it has been addressed)**

MichaelDvP, 2025-01-28:
This is a strange error. Could you check the syslog what is around the new detection. (or send the complete syslog from emsesp)
I guess the date entity have not changed since the wrong detection.

A device with id 0x00 could not exist, we can filter telegrams from 0x00. But i can't imagine how ems-esp get it.
There must be a telegram with source 0x00, type 0x02, and valid data from thermostat 0x10 and valid checksum, also a one with source 0x00, type 0x06, valid data from thermostat 0x10 and valid checksum. Telegrams with invalid checksum are not processed.

mirino75, 2025-01-28:
[image]

[syslog.zip](https://github.com/user-attachments/files/18571942/syslog.zip)

I figured it out after Martin warned me that there was a problem with my devices. I'm trying to export data via Prometheus...
https://github.com/poelstra/ems-esp-exporter
[image]

commit a8195665 (2025-01-29): ignore src==0, fixes #2378
closed 2025-01-29 by proddy