emsesp/EMS-ESP32 #2379: buderus boiler freezed with double dash (--) error

voborl00, 2025-01-28:
after connecting (with original service jack) bbqkees ems gateway e32 v2 to buderus logamax gb072 boiler is repeatedly (few days interval) falling to freezed state with double dash (--) error and only turning it off and on again could be performed (no buttons, no gateway disconnection, no nothing works). gateway also does not send any usable data to home assistant but i have debug syslog fragment from this event interval. it was in the middle of the night, there was no thermostat/boiler fiddling, so if you please are able to detect the source of the problem, it will be very helpful, as it makes tenant's flat completely cold since the event to the morning and as it is not for the first time, they're really pissed off :-(

Jan 28 00:15:00Z ems-esp emsesp 010+10:56:18.906 T 962413: me(0x0B) R boiler(0x08), WeatherComp(0x28), length: 27
 Jan 28 00:15:00Z ems-esp emsesp 010+10:56:18.933 T 962414: boiler(0x08) W me(0x0B), WeatherComp(0x28), data: 00 5A 14 10 00 05
 Jan 28 00:15:00Z ems-esp emsesp 010+10:56:19.110 T 962415: me(0x0B) R thermostat(0x18), RC300Monitor(0x02A5), length: 25
 Jan 28 00:15:00Z ems-esp emsesp 010+10:56:19.219 T 962416: thermostat(0x18) W me(0x0B), RC300Monitor(0x02A5), data: 00 C9 21 24 17 00 24 29 01 59 01 01 03 01 59 01 3B 00 00 11 01 04 07 DA 00
 Jan 28 00:15:00Z ems-esp emsesp 010+10:56:19.244 T 962417: me(0x0B) R thermostat(0x18), RC300Monitor(0x02A5), length: 25 (offset 25)
 Jan 28 00:15:00Z ems-esp emsesp 010+10:56:19.309 T 962418: thermostat(0x18) W me(0x0B), RC300Monitor(0x02A5), data: 00 04 FF AC 00 00 00 64 46 00 3C 01 FF (offset 25)
 Jan 28 00:15:00Z ems-esp emsesp 010+10:56:19.604 T 962419: me(0x0B) R thermostat(0x18), RC300Set(0x02B9), length: 25
 Jan 28 00:15:01Z ems-esp emsesp 010+10:56:19.697 T 962420: thermostat(0x18) W me(0x0B), RC300Set(0x02B9), data: FF 2E 29 26 24 03 00 FF FF 05 2C 01 E1 20 01 0F 05
 Jan 28 00:15:01Z ems-esp emsesp 010+10:56:19.931 T 962421: me(0x0B) R thermostat(0x18), RC300Summer(0x02AF), length: 25
 Jan 28 00:15:01Z ems-esp emsesp 010+10:56:19.997 T 962422: thermostat(0x18) W me(0x0B), RC300Summer(0x02AF), data: 0A 28 00 00 46 2D 12 02 19 1E 00
 Jan 28 00:15:01Z ems-esp telegram 010+10:56:20.434 T 962423: Incomplete Rx: 0B 98 FF 00 19 01 9B FF 89
 Jan 28 00:15:02Z ems-esp telegram 010+10:56:20.907 T 962424: Incomplete Rx: 0B 98 FF 00 19 01 9B FF CA
 Jan 28 00:15:03Z ems-esp emsesp 010+10:56:22.147 T 962425: me(0x0B) R thermostat(0x18), RC300Curves(0x029B), length: 25
 Jan 28 00:15:03Z ems-esp emsesp 010+10:56:22.200 T 962426: thermostat(0x18) W me(0x0B), RC300Curves(0x029B), data: 01 01 00 00 00 02 05 30 46
 Jan 28 00:15:03Z ems-esp emsesp 010+10:56:22.296 T 962427: me(0x0B) R thermostat(0x18), HPMode(0x0291), length: 25
 Jan 28 00:15:03Z ems-esp emsesp 010+10:56:22.346 T 962428: thermostat(0x18) W me(0x0B), HPMode(0x0291), data: 01 00 00 00
 Jan 28 00:15:03Z ems-esp emsesp 010+10:56:22.394 T 962429: boiler(0x08) B all(0x00), UBADevices(0x07), data: 0B 00 00 00 00 00 00 00 00 00 00 00 00
 Jan 28 00:15:04Z ems-esp emsesp 010+10:56:22.719 T 962430: me(0x0B) R thermostat(0x18), RC300WWmode(0x02F5), length: 25
 Jan 28 00:15:04Z ems-esp emsesp 010+10:56:22.794 T 962431: thermostat(0x18) W me(0x0B), RC300WWmode(0x02F5), data: 01 00 02 00 00 FF 08 01 00 08 04 00
 Jan 28 00:15:04Z ems-esp emsesp 010+10:56:22.923 T 962432: me(0x0B) R thermostat(0x18), RC300OutdoorTemp(0x023A), length: 25
 Jan 28 00:15:04Z ems-esp emsesp 010+10:56:22.948 T 962433: thermostat(0x18) W me(0x0B), RC300OutdoorTemp(0x023A), data: 00 4E
 Jan 28 00:15:04Z ems-esp telegram 010+10:56:23.420 T 962434: Incomplete Rx: 0B 98 FF 00 19 01 41 FF 92
 Jan 28 00:15:07Z ems-esp emsesp 010+10:56:25.686 T 962435: me(0x0B) R thermostat(0x18), RC300Settings(0x0241), length: 25
 Jan 28 00:15:07Z ems-esp emsesp 010+10:56:25.758 T 962436: thermostat(0x18) W me(0x0B), RC300Settings(0x0241), data: 32 00 03 00 00 00 00 00 FF 02 F0 06 FF 00
 Jan 28 00:15:07Z ems-esp emsesp 010+10:56:25.828 T 962437: boiler(0x08) B all(0x00), UBADevices(0x07), data: 0B 00 00 00 00 00 00 00 00 00 00 00 00
 Jan 28 00:15:07Z ems-esp emsesp 010+10:56:26.052 T 962438: boiler(0x08) B all(0x00), UBAMonitorFast(0x18), data: 17 00 D3 64 00 01 01 20 40 80 00 02 58 80 00 00 00 FF 30 59 00 CC 00 00 00
 Jan 28 00:15:07Z ems-esp emsesp 010+10:56:26.342 T 962439: boiler(0x08) B all(0x00), UBAMonitorSlow(0x19), data: 00 49 80 00 80 00 00 00 00 64 00 7B 94 0D F9 D3 00 00 00 0C 37 6B 00 5A 6E 80 00
 Jan 28 00:15:07Z ems-esp emsesp 010+10:56:26.604 T 962440: boiler(0x08) B all(0x00), UBAMaintenanceStatus(0x1C), data: 98 06 12 10 35 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 Jan 28 00:15:08Z ems-esp emsesp 010+10:56:26.807 T 962441: boiler(0x08) B all(0x00), MC110Status(0x2A), data: 00 00 00 00 00 00 00 00 CF 00 00 80 00 00 80 00 80 00 80 00 00
 Jan 28 00:15:08Z ems-esp emsesp 010+10:56:27.095 T 962442: boiler(0x08) B all(0x00), UBAMonitorWW(0x34), data: 3C 02 58 02 58 21 00 00 03 00 01 C2 68 00 21 26 00 80 00
 Jan 28 00:15:08Z ems-esp emsesp 010+10:56:27.371 T 962443: boiler(0x08) B all(0x00), UBADevices(0x07), data: 0B 00 00 00 00 00 00 00 00 00 00 00 00
 Jan 28 00:15:12Z ems-esp emsesp 010+10:56:30.902 T 962444: thermostat(0x18) B all(0x00), RC300Monitor(0x02A5), data: 00 C9 21 24 17 00 24 29 01 59 01 01 03 01 59 01 3B 00 00 11 01 04 07 DA 00
 Jan 28 00:15:12Z ems-esp emsesp 010+10:56:31.159 T 962445: boiler(0x08) B all(0x00), UBADevices(0x07), data: 0B 00 01 00 00 00 00 00 00 00 00 00 00
 Jan 28 00:15:12Z ems-esp emsesp 010+10:56:31.422 T 962446: thermostat(0x18) B all(0x00), RC300Monitor(0x02A5), data: 00 04 FF AC 00 00 00 64 46 00 3C 01 FF (offset 25)
 Jan 28 00:15:12Z ems-esp emsesp 010+10:56:31.619 T 962447: thermostat(0x18) B all(0x00), RC300WWmode2(0x031D), data: 00 00 05 07
 Jan 28 00:15:13Z ems-esp emsesp 010+10:56:32.057 T 962448: thermostat(0x18) W boiler(0x08), UBASetPoints(0x1A), data: 17 64 64
 Jan 28 00:15:17Z ems-esp emsesp 010+10:56:36.046 T 962449: boiler(0x08) B all(0x00), UBAMonitorFast(0x18), data: 17 00 D3 64 00 01 01 20 40 80 00 02 58 80 00 00 00 FF 30 59 00 CC 00 00 00
 Jan 28 00:15:17Z ems-esp emsesp 010+10:56:36.264 T 962450: boiler(0x08) B all(0x00), MC110Status(0x2A), data: 00 00 00 00 00 00 00 00 CF 00 00 80 00 00 80 00 80 00 80 00 00
 Jan 28 00:15:17Z ems-esp emsesp 010+10:56:36.585 T 962451: boiler(0x08) B all(0x00), UBAMonitorWW(0x34), data: 3C 02 58 02 58 21 00 00 03 00 01 C2 68 00 21 26 00 80 00
 Jan 28 00:15:27Z ems-esp emsesp 010+10:56:45.957 T 962452: boiler(0x08) B all(0x00), UBAMonitorFast(0x18), data: 17 00 D3 64 00 01 01 20 40 80 00 02 58 80 00 00 00 FF 30 59 00 CC 00 00 00
 Jan 28 00:15:27Z ems-esp emsesp 010+10:56:46.207 T 962453: boiler(0x08) B all(0x00), MC110Status(0x2A), data: 00 00 00 00 00 00 00 00 CF 00 00 80 00 00 80 00 80 00 80 00 00
 Jan 28 00:15:27Z ems-esp emsesp 010+10:56:46.484 T 962454: boiler(0x08) B all(0x00), UBAMonitorWW(0x34), data: 3C 02 58 02 58 21 00 00 03 00 01 C2 68 00 21 26 00 80 00
 Jan 28 00:15:31Z ems-esp emsesp 010+10:56:49.957 T 962455: boiler(0x08) B all(0x00), UBAMonitorFast(0x18), data: 17 00 D3 64 00 01 01 20 40 80 00 02 58 80 00 00 00 FF 30 59 00 CC 00 02 00
 Jan 28 00:15:34Z ems-esp emsesp 010+10:56:52.992 T 962456: boiler(0x08) B all(0x00), UBAMonitorFast(0x18), data: 17 00 D3 64 00 01 01 2D 40 80 00 02 58 80 00 00 00 FF 30 4C 01 1C 00 02 00
 Jan 28 00:15:36Z ems-esp emsesp 010+10:56:54.982 T 962457: boiler(0x08) B all(0x00), UBAMonitorFast(0x18), data: 17 00 D3 64 00 09 01 2D 40 80 00 02 58 80 00 00 8C FF 30 4C 01 1C 00 02 00
 Jan 28 00:15:37Z ems-esp emsesp 010+10:56:55.973 T 962458: boiler(0x08) B all(0x00), MC110Status(0x2A), data: 00 00 00 00 00 00 00 01 67 01 66 80 00 00 80 00 80 00 80 00 00
 Jan 28 00:15:37Z ems-esp emsesp 010+10:56:56.197 T 962459: boiler(0x08) B all(0x00), UBAMonitorWW(0x34), data: 3C 02 58 02 58 21 00 00 03 00 01 C2 68 00 21 26 00 80 00
 Jan 28 00:15:37Z ems-esp emsesp 010+10:56:56.494 T 962460: boiler(0x08) B all(0x00), UBAMonitorFast(0x18), data: 17 00 D3 64 3B 09 01 25 40 80 00 02 58 80 00 00 AD FF 2D 48 00 C8 00 02 00
 Jan 28 00:15:38Z ems-esp emsesp 010+10:56:56.973 T 962461: boiler(0x08) B all(0x00), UBAMonitorFast(0x18), data: 17 00 D5 64 3B 09 01 25 40 80 00 02 58 80 00 00 B7 FF 2D 48 00 C8 00 02 00
 Jan 28 00:15:41Z ems-esp emsesp 010+10:56:59.960 T 962462: boiler(0x08) B all(0x00), UBAMonitorFast(0x18), data: 17 00 DD 64 34 09 01 25 40 80 00 02 58 80 00 00 C3 FF 2D 48 00 C8 00 02 00
[... cut ...]
 Jan 28 00:16:27Z ems-esp emsesp 010+10:57:43.194 T 962710: controller(0x09) R boiler(0x08), ?(0xF8), length: 255, data: 05 00 00
 Jan 28 00:16:27Z ems-esp emsesp 010+10:57:43.286 T 962714: controller(0x09) R boiler(0x08), ?(0xF8), length: 255, data: 05 00 00
 Jan 28 00:16:27Z ems-esp emsesp 010+10:57:43.387 T 962718: thermostat(0x18) R boiler(0x08), UBADevices(0x07), length: 14
 Jan 28 00:16:27Z ems-esp emsesp 010+10:57:43.419 T 962719: boiler(0x08) W thermostat(0x18), UBADevices(0x07), data: 0B 00 00 00 00 00 00 00 00 00 00 00 00

bbqkees, 2025-01-28:
I assume you are on a recent development firmware and not on 3.7.0, which had a memory leak which bricked up the Gateway after a while.

I think the EMS service port in the boiler may be giving an intermittent connection.
The Buderus GB072 has the same front panel as my own Nefit Trendline.
So I know from experience that it has a very 'loose' EMS service port. It is also a bit recessed so you have to insert the cable properly.

If you can revert to the screw terminal again to see if that solves it.
In the mean time I can send you a new EMS service cable to rule out the issue is with the cable itself.

voborl00, 2025-01-28:
tried latest firmware, it was running smooth approx a week and then this. i think cable is fitted at its max, but can only validate it visually...

i can tolerate if the connection drops and the gateway stop sending data (to HA), but this completele freezes the boiler (and the flat as well), which is after this last event unacceptable - can loose cable or similar issue produce such heavy boiler failure?

when i had it connected to screw terminal, it stops communication after day or two as well, but instead of freezing whole boiler, it produced EMS comm failure error (which in the end leads to safety mode on boiler and frozen flat as well)

so i think, if there is no other way than retries, i had to wait to summer when the boiler is not needed 100% functional

voborl00, 2025-01-28:
understood. i have only one thermostat as a module, this boiler-thermostat double worked with no issues many years. now with using service jack it worked +-week with no issues in locked room, so no one touched the gateway or the service cable and after a week this double dash error occured from blue sky - can the cable loose itself (it was fitted as deep as it can go on both sides) by time?

about screw terminal - is it better to fit both ems twin wire from thermostat and from gateway into one twin terminal port? or is it better to use two wago connectors (https://www.wago.com/us/wire-splicing-connectors/compact-splicing-connector/p/221-413) which will both input one wire from thermostat and gateway and output one wire to screw terminal?

voborl00, 2025-01-29:
more to add, after this event and boiler restart, it has (again like in comm error when using wired screw terminal instead of service jack) disabled hot water for storage tank. didn't notice it, as there is slow temperature degradation in tank. i have to set it up in service menu, as it cannot be set just with direct hot water button on control panel. standard heating to radiators hasn't been affected. is this also expected behaviour? tenants are mad now :((

voborl00, 2025-01-29:
firmware 3.7.2-dev.10

proddy, 2025-01-29:
sorry to hear that. to rule out its a software issue in EMS-ESP what happens if your make EMS-ESP read-only by enabling this:

[image]

then EMS-ESP will not sent packets on the EMS bus, and will only be in listen mode.

voborl00, 2025-01-29:
it was in read only mode since it was installed and i checked it already after firmware update and still it disables hot water in each this (service cable/direct wiring) comm trouble

bbqkees, 2025-01-30:
@voborl00 Let's swap out the Gateway for a new one before you run new tests.
I'll send one tomorrow, you can test with the new unit to see if it makes a difference.
If it does, you use the new one in your site and if it doesn't we excluded the hardware being an issue and have to look elsewhere for the cause. In the latter case you just return the second Gateway to me.
Did you connect DS18B20 sensors to it by chance?

MichaelDvP, 2025-01-30:
You can split in the screw-terminals or use wagos in between. Each cable length from terminal to terminal or wago should be >20cm. 
Use > 0.5mm^2 cables for all bus connections, The 2 bus lines should be parallel or twisted, and with distance to mains cables. 
Try which connection give you lower rates of incomplete telegrams.

You have not posted the system info, so we don't know what is configured. 
Ems-esp is a interface from ems-bus to mqtt/api/web, it just translates entities from one to another, but never changes something on the heating system without a vommand from api/mqtt. In readonly-mode no commands are sent to the bus (actually except remote thermostat emulation if configured). 
If the dhw is disabled via ems-esp there is a) not readonly-mode configured and b) a command to disable dhw sent from your system (homeassistent or whatever you use),  

So before connecting to the bus:
- check ems-esp settings, or reset to default. Change settings only if you know what they do.
- check homeassistant (or what system you use) and disable writes to ems-esp.
- don't use special settings (force heating off, remote thermostat, mqtt publish to command) if you are unsure that you really need this.

voborl00, 2025-07-18:
ok, so here we go again, after running ok for around 10 days it again goes into "--" state of the boiler. this time i doublechecked everything, firmware version, readonly mode, wiring was at best achievable, but again out of the blue this error state and again disabled hot water (even in the readonly mode!) which had to be set up manually in boiler's service menu. here are all the logs from the last minutes before crashing, there are many lines like this which i removed

Jul 17 17:25:13Z ems-esp emsesp 013+06:41:26.644 T 1304689: boiler(0x08) W controller(0x09), ?(0xF8), data: 05 00 00
Jul 17 17:25:13Z ems-esp emsesp 013+06:41:26.667 T 1304690: controller(0x09) R boiler(0x08), ?(0xF8), length: 255, data: 05 00 00

 Jul 17 17:22:50Z ems-esp emsesp 013+06:39:05.284 T 1300122: thermostat(0x18) W me(0x0B), RC300Settings(0x0241), data: 32 00 03 00 00 00 00 00 FF 02 F0 06 FF 00
 Jul 17 17:23:01Z ems-esp emsesp 013+06:39:15.031 T 1300430: boiler(0x08) B all(0x00), MC110Status(0x2A), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 Jul 17 17:23:11Z ems-esp emsesp 013+06:39:25.240 T 1300787: me(0x0B) R boiler(0x08), UBAParameterWW(0x33), length: 27
 Jul 17 17:23:12Z ems-esp emsesp 013+06:39:26.781 T 1300828: boiler(0x08) B all(0x00), MC110Status(0x2A), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 Jul 17 17:23:13Z ems-esp emsesp 013+06:39:28.266 T 1300864: me(0x0B) R boiler(0x08), WeatherComp(0x28), length: 27
 Jul 17 17:23:15Z ems-esp emsesp 013+06:39:29.970 T 1300918: thermostat(0x18) W me(0x0B), RC300Monitor(0x02A5), data: 00 E4 32 25 00 00 25 24 00 23 03 03 01 00 23 01 27 00 00 11 01 01 08 E8 00
 Jul 17 17:23:18Z ems-esp emsesp 013+06:39:32.528 T 1301001: me(0x0B) R thermostat(0x18), RC300Set(0x02B9), length: 25
 Jul 17 17:23:26Z ems-esp emsesp 013+06:39:40.245 T 1301206: boiler(0x08) B all(0x00), UBAMonitorFast(0x18), data: 00 00 00 00 00 20 00 20 00 00 00 00 00 00 00 00 00 00 39 55 00 E9 00 00 00
 Jul 17 17:23:26Z ems-esp emsesp 013+06:39:41.143 T 1301216: me(0x0B) R thermostat(0x18), HPMode(0x0291), length: 25
 Jul 17 17:23:29Z ems-esp emsesp 013+06:39:44.460 T 1301327: me(0x0B) R thermostat(0x18), RC300OutdoorTemp(0x023A), length: 25
 Jul 17 17:23:31Z ems-esp emsesp 013+06:39:46.215 T 1301391: me(0x0B) R thermostat(0x18), RC300Settings(0x0241), length: 25
 Jul 17 17:23:40Z ems-esp emsesp 013+06:39:54.549 T 1301664: boiler(0x08) B all(0x00), UBAMonitorWW(0x34), data: 00 00 00 00 00 00 00 00 00 00 01 D6 A0 00 22 DE 00 00 00
 Jul 17 17:23:53Z ems-esp emsesp 013+06:40:07.291 T 1302120: boiler(0x08) B all(0x00), MC110Status(0x2A), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 Jul 17 17:24:07Z ems-esp emsesp 013+06:40:21.579 T 1302603: thermostat(0x18) B all(0x00), RC300Monitor(0x02A5), data: 00 22 (offset 8)
 Jul 17 17:24:18Z ems-esp emsesp 013+06:40:32.542 T 1302960: me(0x0B) R thermostat(0x18), RC300Monitor(0x02A5), length: 25
 Jul 17 17:24:18Z ems-esp emsesp 013+06:40:32.676 T 1302961: thermostat(0x18) W me(0x0B), RC300Monitor(0x02A5), data: 00 E4 32 25 00 00 25 24 00 22 03 03 01 00 22 01 28 00 00 11 01 01 08 E8 00
 Jul 17 17:24:18Z ems-esp emsesp 013+06:40:32.676 T 1302962: me(0x0B) R thermostat(0x18), RC300Monitor(0x02A5), length: 25 (offset 25)
 Jul 17 17:24:18Z ems-esp emsesp 013+06:40:32.736 T 1302963: thermostat(0x18) W me(0x0B), RC300Monitor(0x02A5), data: 00 05 FF 9C 00 00 00 00 46 00 3C 01 FF (offset 25)
 Jul 17 17:24:22Z ems-esp emsesp 013+06:40:36.474 T 1303013: thermostat(0x18) B all(0x00), SystemError(0xBF), data: 18 9D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 Jul 17 17:24:28Z ems-esp emsesp 013+06:40:43.443 T 1303239: thermostat(0x18) W me(0x0B), RC300WWmode(0x02F5), data: 01 00 02 00 00 FF 08 01 00 08 04 00
 Jul 17 17:24:30Z ems-esp emsesp 013+06:40:45.001 T 1303297: thermostat(0x18) W me(0x0B), RC300OutdoorTemp(0x023A), data: 00 00
 Jul 17 17:24:34Z ems-esp emsesp 013+06:40:47.879 T 1303404: boiler(0x08) B all(0x00), UBAMonitorFast(0x18), data: 00 00 00 00 00 20 00 20 00 00 00 00 00 00 00 00 00 00 39 55 00 E9 00 00 00
 Jul 17 17:24:47Z ems-esp emsesp 013+06:41:01.527 T 1303864: boiler(0x08) B all(0x00), UBAMonitorFast(0x18), data: 00 00 00 00 00 20 00 20 00 00 00 00 00 00 00 00 00 00 39 55 00 E9 00 00 00
 Jul 17 17:25:01Z ems-esp emsesp 013+06:41:15.538 T 1304326: boiler(0x08) B all(0x00), UBAMonitorWW(0x34), data: 00 00 00 00 00 00 00 00 00 00 01 D6 A0 00 22 DE 00 00 00
 Jul 17 17:25:07Z ems-esp telegram 013+06:41:21.034 T 1304513: Incomplete Rx: 0B 88 14 00 1B DF 08 0B 14 00 52 D7 30 C4
 Jul 17 17:25:07Z ems-esp emsesp 013+06:41:21.425 T 1304514: thermostat(0x18) B all(0x00), RC300Monitor(0x02A5), data: 00 21 01 29 (offset 13) 
 Jul 17 17:25:11Z ems-esp emsesp 013+06:41:26.094 T 1304668: me(0x0B) R boiler(0x08), UBAParameters(0x16), length: 27 (offset 27)

@bbqkees any ideas as i am at the end of my options (connection, wiring, configuration, ...) ?

voborl00, 2025-07-18:
obviously there is that SystemError followed by WWMode which could lead to "--" error and hot water mode set to disabled (or "1" for given buderus boiler - "8" sets it to use hot water tank). but i am not able to decode the messages' values and there was no obvious error noticable at the moment of crash @bbqkees

MichaelDvP, 2025-07-18:
The 0xBF telegram from thermostat shows no error, but telegram 0x18 fom boiler shows error 9U (233), indicating an defect of UBA or KIM, Switch off, pull the coding key (KIM) and reinsert, maybe this helps. The DHW telegrams look ok, see https://docs.emsesp.org/ems-library/#telegram-0x02f5-0x02f6

The log is confusing, it starts 17:22:50 with T 1300122 and ends 17:25:11 with T 1304668, This are 4500 messages in 2:20 minutes, ~32 logs per second? Where are the other messages? Also no system info, no idea what gateway/version/settings are used. But 9U error have nothing to do with ems-bus communication, the gateway and it's settings seems not involved.

AKreisel, 2025-09-07:
Hi,

I have the exact same problem as documented in this issue.

I have a combination of Buderus GB172-14/RC310 running since 2016 without any problems. Last year I bought an ems-esp and connected it via service port. First it worked fine but every now and then it crashed, sometimes after a few weeks, sometimes after only two days. Every time the boiler afterwards showed error code EH/258 and shut down / stopped working.

After about six occurences I stopped using the ems-esp device and the boiler ran fine for a few months. But I came back to it and tried again, still experiencing the aforementioned outcome. This time I played around with the settings for a bit and also set up a syslog server for analysis. The latter actually seemed to stabilize the sitution instead of just providing debug data and I did not experience crashes anymore. After a few month time I had to shut down my syslog server for two weeks because of construction work in the basement and immediately the crashes came back (but as my syslog server was down, I had no logs). After reestablishing the syslog server, everything went smooth again for about two months.

But yesterday evening I noticed the double dash error, mentioned in this thread, on the dashboard of the boiler, accompanied by the error code 9U/233. And similarly to the issue of the author, hot water was disabled for me, too, after rebooting the boiler, and I had to manually enable it again through the service menus.

I enclose the syslog of my ems-esp device, firmware version is 3.7.2. Is there any additional information I could provide to help narrowing down the problem? I saw the last comment, saying it is unlikely the gateway is causing the problem, but as I am the second person now, experiencing said issues and also my boiler has been running without any problems for eight years, and the troubles only began after using the ems-esp device, maybe it's worth another look.

I currently don't use the ems-esp, because I am a bit concerned about further problems.

Regards,
Andreas

PS: I also have tried 'read-only mode' without any success before configuring syslog

[ems-esp.log](https://github.com/user-attachments/files/22197766/ems-esp.log)

[image]

proddy, 2025-09-15:
@AKreisel sorry to hear that. A few clarifying questions if I may:

- How soon after you plug in the EMS-ESP until you see the error code?
- Does it also happen when you're not writing to the EMS bus, i.e. it's just idling and reading data?
- Which type of gateway/interface board are you using and which model and batch version? It could be an electronic mishap where EMS-ESP is sending noise down the line and confusing the devices.

MichaelDvP, 2025-09-16:
I can't really help with the issue, but can help to analyze the log. 
Communication looks normal until
`Sep  6 20:23:04 ems-esp emsesp 040+13:36:19.354` when suddely the controller starts an interview of the boiler, reading all parameters and values. After controller has read all values it tries to read 0xF8 for the ems telegram 0x05
`Sep  6 20:23:08 ems-esp emsesp 040+13:36:22.084 T 4594533: controller(0x09) R boiler(0x08), ?(0xF8), length: 255, data: 05 00 00`
Boiler reply a empty value and controller loops with repeated reading the 0xF8. The controller reads are every 30ms, so the log is flooded, but other communication still is there and stable. 
Telegrams 0xF. contains data about  telegram adresses sent as fist 3 bytes in data body. `data 05 00 00` means telegram 05 EMS (00 00 EMS+ address). The known 0xF. telegrams are: 0xF7 is used to show menu entries in the thermostat, 0xF9 for value limits. But i don't know what 0xF8 means.
Telegram 0x05 is a command telegram for the controller, used for push button commands like reset, chimneysweeper, disinfection, wwonetime, etc. 
The telegram is also read by the contoller in
```
Sep  6 20:23:06 ems-esp emsesp 040+13:36:20.716 T 4594516: controller(0x09) R boiler(0x08), ?(0x05), length: 27
Sep  6 20:23:06 ems-esp emsesp 040+13:36:20.773 T 4594517: boiler(0x08) W controller(0x09), ?(0x05), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```
showing only zeros. Maybe the controller expects something other and tries to read/configure the KIM (Kessel Id-Module). 

Questions:
- Is the controller interview a periodic process? 
- Try `read 08 05` while the system is running  normal. What's the output? 
- Is there a controller interview on statup (restart whole system) with a different 0x05 telegram?
- Can you try to set tx-mode 0 in terminal. This will stop any write/read-requests from ems-esp to the bus.

AKreisel, 2025-09-26:
Thank you for the explanations. I will try the bus clamps.

Although I find it hard to believe that this EH/258 is caused by a loose connection in my case. As I wrote before it happened every few days when I ran the gateway without a syslog server, then didn't occur for several month with the syslog server configured, then was back again, when I had to shutdown my syslog server for two weeks, but now hasn't happened since May when I put the syslog server back in place. And I didn't even touch the service plug in the time I had to shut off the syslog server, it was plugged in the whole time. Still it broke and also recovered just by (un)configuring this syslog server.

As for the 9U/233, I can image that to be caused by a loose contact, and not the gateway. Especially as it only occured one single time. I just was intrigued by the fact that the OP apparently has a setup very similar to mine and also experienced the same error. So I hoped for an easy gateway fix. ;)