emsesp/EMS-ESP32 #2383: ems-esp Buderus WPS196.i2 does not have device PKS9 Passive Cooling station

a-lenkov, 2025-01-30:
### DESCRIPTION

ems-esp for HP Buderus WPS196.i2 with the Passive Cooling station PKS9 does not have device PKS9.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
  Paste System information here....

```

### TO REPRODUCE

_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR

_A clear and concise description of what you expected to happen._

### SCREENSHOTS

_If applicable, add screenshots to help explain your issue._

### ADDITIONAL CONTEXT

_Add any other context about the issue here._

**(Please remember to close the issue when it has been addressed)**

a-lenkov, 2025-01-30:
What kind of information do you need more? What shall I add?

proddy, 2025-01-30:
Have a look at https://docs.emsesp.org/Troubleshooting/#im-missing-certain-data-from-an-ems-device

See if EMS-ESP detects the device on PKS9, if it's indeed an EMS device. It'll be shown in the logs. Then we'll need to help to decipher the telegrams. What kind of information do you want to see? Can you make screenshots of what is displayed now?