emsesp/EMS-ESP32 #2394: Change WiFi failed - Bootloop

HerrRiebmann, 2025-02-04:
V3.7.1 ESP32 16 MB

### DESCRIPTION

I got a new router with new WiFi credentials.
The ESP will create a Hotspot, where I entered the new SSID and password. Then it still starts with the hotspot.
Using the "Search WiFi" won´t find any Networks. At least there were ~8 available.

Then I changed my 2.4/5GHz to 2.4 only, without any change.

Now it seems that also the Hotspot is not very reliable. It often stops. The (original) Hardware is left untouched, I only unplugged it from power.
Finally it does not start the Hotspot anymore...

Then conneted it to the PC and did the upgrade again. First with Settings, secondly without.

### REQUESTED INFORMATION

Started with V3.7.0, then upgraded to V3.7.1

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [X] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [X] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [X] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
  Not accessible anymore...

```

### TO REPRODUCE

Change PW from old WiFi. Introduce new WiFi with different Name and PW.
Connect to ESP via AP, then enter new Credentials.

### EXPECTED BEHAVIOUR

The device should connect to the new AP.

### SCREENSHOTS

_If applicable, add screenshots to help explain your issue._

### ADDITIONAL CONTEXT

What might happen, as I saw when I changed another Device to the new WiFi:
My 3D Printer Camera also failed. The output was that the WiFi channel number was to "big". As I saw, the Router took Channel 11 for 2,4 and 36 for 5GHz.
Off course I already deactivated the 5GHz Band, without any success.

Log after Flashing (EMS-ESP Flasher 1.1.1) with constant reboots:

> Recent Log (level INFO, max 25 messages):
> 000+00:00:00.000 I 1: [emsesp] EMS-ESP version 3.7.1 (boot partition)
> 000+00:00:00.000 I 2: [emsesp] Last system reset reason Core0: Software reset CPU, Core1: Software reset CPU
> 000+00:00:00.000 I 3: [emsesp] Board profile set to S32
> 000+00:00:00.000 I 4: [emsesp] Loaded EMS device library (121 entries)
> 000+00:00:00.000 I 5: [emsesp] Starting Scheduler service
> 000+00:00:00.000 I 6: [emsesp] Starting Custom Entity service
> 000+00:00:00.000 I 7: [temperaturesensor] Starting Temperature Sensor service
> 000+00:00:00.000 I 8: [analogsensor] Starting Analog Sensor service
> 000+00:00:00.000 I 9: [emsesp] Starting Modules service
> 000+00:00:00.000 I 10: [module] No external library modules found
> 000+00:00:00.000 I 11: [emsesp] Starting Web Server
> ems-esp:$ ems-esp:$ 000+00:00:10.000 I 1: [emsesp] Starting Access Point with captive portal on 192.168.4.1
> ems-esp:$ ets Jul 29 2019 12:21:46
> rst:0x8 (TG1WDT_SYS_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0030,len:1184
> load:0x40078000,len:13132
> load:0x40080400,len:3036
> entry 0x400805e4
> E (768) psram: PSRAM ID read error: 0xffffffff
> ems-esp:$ ems-esp:$ sems-esp:$ shems-esp:$ shoems-esp:$ showems-esp:$ show ems-esp:$ show lems-esp:$ show loems-esp:$ show log
> Recent Log (level INFO, max 25 messages):
> 000+00:00:00.000 I 1: [emsesp] EMS-ESP version 3.7.1 (boot partition)
> 000+00:00:00.000 I 2: [emsesp] Last system reset reason Core0: Timer group1 watch dog reset, Core1: APP CPU reset by PRO CPU
> 000+00:00:00.000 I 3: [emsesp] Board profile set to S32
> 000+00:00:00.000 I 4: [emsesp] Loaded EMS device library (121 entries)
> 000+00:00:00.000 I 5: [emsesp] Starting Scheduler service
> 000+00:00:00.000 I 6: [emsesp] Starting Custom Entity service
> 000+00:00:00.000 I 7: [temperaturesensor] Starting Temperature Sensor service
> 000+00:00:00.000 I 8: [analogsensor] Starting Analog Sensor service
> 000+00:00:00.000 I 9: [emsesp] Starting Modules service
> 000+00:00:00.000 I 10: [module] No external library modules found
> 000+00:00:00.000 I 11: [emsesp] Starting Web Server
> ems-esp:$ ems-esp:$ 000+00:00:10.000 I 1: [emsesp] Starting Access Point with captive portal on 192.168.4.1
> ems-esp:$ ets Jul 29 2019 12:21:46
> rst:0x8 (TG1WDT_SYS_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0030,len:1184
> load:0x40078000,len:13132
> load:0x40080400,len:3036
> entry 0x400805e4
> E (768) psram: PSRAM ID read error: 0xffffffff
> ems-esp:$ ems-esp:$ sems-esp:$ shems-esp:$ shoems-esp:$ showems-esp:$ show ems-esp:$ show lems-esp:$ show loems-esp:$ show log
> Recent Log (level INFO, max 25 messages):
> 000+00:00:00.000 I 1: [emsesp] EMS-ESP version 3.7.1 (boot partition)
> 000+00:00:00.000 I 2: [emsesp] Last system reset reason Core0: Timer group1 watch dog reset, Core1: APP CPU reset by PRO CPU
> 000+00:00:00.000 I 3: [emsesp] Board profile set to S32
> 

I´ll provide any information, you need! Got knowledge about Arduino, ESP´s and everything available from a logic analyzer to an osziloscope.

The router is a brand new Vodafone router (connected to the TV cable) with 1000/50mbit.
Firmware-Version AR01.04.137.07_060724_7249.SIP.20
Hardware-Typ & Version 9

MichaelDvP, 2025-02-04:
Looks to me as you have a S32 gateway with 16M ram and have flashed the psram version (16MB+). 
Flash the right version via USB. 
Or try to disconnect the ems-bus. This needs less ram and the AP should start and you can configure/update to the 16M version (without plus).

HerrRiebmann, 2025-02-04:
Oh, no! I really oversaw the " + ". Thanks a lot for figuring that out!

Now I can connect to the ESP again. The network still cannot be found.
And not connected after entering the credentials manually:

> ems-esp:$ 000+00:04:23.880 I 2: [system] Preparing to restart system
> 000+00:04:23.880 D 3: [command] Called command system/restart (restart EMS-ESP)
> ems-esp:$ 000+00:04:24.917 I 4: [system] Restarting EMS-ESP...
> ems-esp:$ ets Jul 29 2019 12:21:46
> rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0030,len:1184
> load:0x40078000,len:13132
> load:0x40080400,len:3036
> entry 0x400805e4
> ets Jul 29 2019 12:21:46
> rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0030,len:1184
> load:0x40078000,len:13132
> load:0x40080400,len:3036
> entry 0x400805e4
> ems-esp:$ ems-esp:$ sems-esp:$ shems-esp:$ shoems-esp:$ showems-esp:$ show ems-esp:$ show lems-esp:$ show loems-esp:$ show log
> Recent Log (level INFO, max 25 messages):
> 000+00:00:00.000 I 1: [emsesp] EMS-ESP version 3.7.1 (boot partition)
> 000+00:00:00.000 I 2: [emsesp] Last system reset reason Core0: Software reset CPU, Core1: Software reset CPU
> 000+00:00:00.000 I 3: [emsesp] Board profile set to S32
> 000+00:00:00.000 I 4: [emsesp] Loaded EMS device library (121 entries)
> 000+00:00:00.000 I 5: [emsesp] Starting Scheduler service
> 000+00:00:00.000 I 6: [emsesp] Starting Custom Entity service
> 000+00:00:00.000 I 7: [temperaturesensor] Starting Temperature Sensor service
> 000+00:00:00.000 I 8: [analogsensor] Starting Analog Sensor service
> 000+00:00:00.000 I 9: [emsesp] Starting Modules service
> 000+00:00:00.000 I 10: [module] No external library modules found
> 000+00:00:00.000 I 11: [emsesp] Starting Web Server
> ems-esp:$ ems-esp:$ 000+00:00:03.553 W 1: [emsesp] WiFi disconnected (#1). Reason: no AP found (201)
> ems-esp:$ 000+00:00:09.492 W 2: [emsesp] WiFi disconnected (#2). Reason: no AP found (201)
> ems-esp:$ 000+00:00:10.000 I 3: [emsesp] Starting Access Point with captive portal on 192.168.4.1
> ems-esp:$ 000+00:00:13.556 W 4: [emsesp] WiFi disconnected (#3). Reason: no AP found (201)
> ems-esp:$ 000+00:00:16.557 W 5: [emsesp] WiFi disconnected (#4). Reason: no AP found (201)
> ems-esp:$ 000+00:00:19.563 W 6: [emsesp] WiFi disconnected (#5). Reason: no AP found (201)
> ems-esp:$ 000+00:00:22.570 W 7: [emsesp] WiFi disconnected (#6). Reason: no AP found (201)
> ems-esp:$ 000+00:00:25.572 W 8: [emsesp] WiFi disconnected (#7). Reason: no AP found (201)
> ems-esp:$ 000+00:00:28.577 W 9: [emsesp] WiFi disconnected (#8). Reason: no AP found (201)
> ems-esp:$ 000+00:00:31.581 W 10: [emsesp] WiFi disconnected (#9). Reason: no AP found (201)

SSID and PW are 100% correct. 

Then I tried to restore my old settings with the outdated Wifi. Then the console printes out

> 00+00:00:46.820 W 15: [emsesp] WiFi disconnected (#14). Reason: auth fail (202)
> ems-esp:$ 000+00:00:49.842 W 16: [emsesp] WiFi disconnected (#15). Reason: auth fail (202)
> ems-esp:$ 000+00:00:52.857 W 17: [emsesp] WiFi disconnected (#16). Reason: assoc leave (8)
> ems-esp:$ 000+00:00:55.865 W 18: [emsesp] WiFi disconnected (#17). Reason: assoc leave (8)
> ems-esp:$ 000+00:00:58.872 W 19: [emsesp] WiFi disconnected (#18). Reason: assoc leave (8)
> ems-esp:$ 000+00:01:01.881 W 20: [emsesp] WiFi disconnected (#19). Reason: auth fail (202)
> ems-esp:$ 000+00:01:04.891 W 21: [emsesp] WiFi disconnected (#20). Reason: assoc leave (8)
> ems-esp:$ 000+00:01:07.898 W 22: [emsesp] WiFi disconnected (#21). Reason: auth fail (202)
> ems-esp:$ 000+00:01:10.916 W 23: [emsesp] WiFi disconnected (#22). Reason: auth fail (202)
> ems-esp:$ 000+00:01:13.925 W 24: [emsesp] WiFi disconnected (#23). Reason: auth fail (202)
> ems-esp:$ 000+00:01:16.936 W 25: [emsesp] WiFi disconnected (#24). Reason: assoc leave (8)
> ems-esp:$ 000+00:01:19.945 W 26: [emsesp] WiFi disconnected (#25). Reason: assoc leave (8)

and so on. In this case, I cannot use the "find networks"-tab. This will always fail with "no networks found". Not even the old one, with the wrong password.

I will now take a regular ESP32 and try to connect to the WiFi, with full debug options. Maybe I can find a general issue with my network settings.
Sadly, and this would have been my first approach, the Router cannot be set to use different channels. Or be configured in any meaningful way :-/

MichaelDvP, 2025-02-04:
ESP32 is 2.4GHz only, Channel 11 works here, the german ch12/13 should also work, but i have never tested.
Some users reported connection issues with WPA3/WPA2 mixed mode and mesh networks.
I've tested a while ago WPA3/WPA2 mixed and got WPA3 connection, but never tested with mesh.
Is the SSID hidden? Is DHCP not enabled?

HerrRiebmann, 2025-02-04:
Allright, coming closer, but not to an end!
Using the classic "WiFiScan" example shows the WiFi and all the other ones in the area (14 in total).

> 2 | Vodafone-Router-SSID                    |  -49 | 11 | WPA2

Connection is still not possible.
Then I finally found a setting which can be changed! 2.4GHz Network 802.11 n/ax (_where ax is not supported_).
Changed it to "N" only -> Not working
Deactivated 5GHz Mixed 802.11 n/ac/ax -> Not working
BandSteering also deactivated (_automatically, since I deactivated the 5GHz network_).
WPA2
No MAC Filter
SSID visible
The SSID contains a " - " (minus) but as I read this should work on the ESP´s.
Password only contains letters and numbers.
Local DHCPv4 Server: Active
No mesh.
AFAIK there are no more settings. This Router (Vodafone Station Wi-Fi 6) is crap...

Also the 3D Printer-Cam still cannot connect (Linux based device). That was where I got the log with the channel exeeding from.
All other devices, notebooks, TV, smartplugs, (old) phones, etc. work out of the box.
I´ll now try to check the Cam again and see if I can see more from it´s log.

Thank you Michael for trying to help me out!

HerrRiebmann, 2025-02-04:
It is allways what it is :( My bad!
Error 45 kicked in...

A QR code came with the router. So I scanned the SSID & Password wherever I could.
On any other device, I selected the SSID and entered the PW.
Only on the EMS-ESP and the Cam I entered it manually.
The Font was really misleading! Until I found out, that the SSID is a combination of **Vodafone-** and a two digit HEX.
And Hex does not contain an "O"!

But maybe we can at least take a benefit from my stupidity.
If the SSID is entered wrong, or the SSID is correct but the password is wrong, the "Find Networks" seems no to work properly.

Sorry!!!

closed 2025-02-04 by HerrRiebmann