emsesp/EMS-ESP32 #2403: Improvement: CR50 thermostat

VlastiBroucek, 2025-02-15:
### DESCRIPTION

Holiday mode in CR50 thermostat

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [X] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [X] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [  ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [X] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
  {"system":{"version":"3.7.2-dev.19","uptime":"000+01:02:15.356","uptimeSec":3735,"platform":"ESP32","cpuType":"ESP32-D0WD-V3","arduino":"ESP32 Arduino v2.0.17","sdk":"v4.4.7-dirty","freeMem":135,"maxAlloc":95,"freeCaps":4161,"usedApp":2236,"freeApp":3460,"partition":"app1","flash_chip_size":16384,"resetReason":"Software reset CPU / Software reset CPU","psram":true,"psramSize":4093,"freePsram":4060,"model":""},"network":{"network":"Ethernet","hostname":"ems-esp","TxPowerSetting":0,"staticIP":false,"lowBandwidth":false,"disableSleep":true,"enableMDNS":true,"enableCORS":false,"APProvisionMode":"disconnected","APSecurity":"wpa2","APSSID":"ems-esp"},"ntp":{"NTPStatus":"connected","enabled":true,"server":"time.cloudflare.com","tzLabel":"Australia/Hobart"},"mqtt":{"MQTTStatus":"connected","MQTTPublishes":1386,"MQTTQueued":0,"MQTTPublishFails":0,"MQTTReconnects":0,"enabled":true,"clientID":"esp32-d767b4fc","keepAlive":60,"cleanSession":false,"entityFormat":1,"base":"ems-esp","discoveryPrefix":"homeassistant","discoveryType":0,"nestedFormat":1,"haEnabled":true,"mqttQos":0,"mqttRetain":false,"publishTimeHeartbeat":60,"publishTimeBoiler":10,"publishTimeThermostat":10,"publishTimeSolar":10,"publishTimeMixer":10,"publishTimeWater":10,"publishTimeOther":60,"publishTimeSensor":10,"publishSingle":false,"publish2command":false,"sendResponse":true},"syslog":{"enabled":false},"sensor":{"temperatureSensors":3,"temperatureSensorReads":2238,"temperatureSensorFails":0},"api":{"APICalls":0,"APIFails":0},"bus":{"busStatus":"connected","busProtocol":"HT3","busTelegramsReceived":3611,"busReads":1061,"busWrites":13,"busIncompleteTelegrams":0,"busReadsFailed":0,"busWritesFailed":0,"busRxLineQuality":100,"busTxLineQuality":100},"settings":{"boardProfile":"E32V2","locale":"en","txMode":3,"emsBusID":11,"showerTimer":false,"showerMinDuration":180,"showerAlert":false,"hideLed":false,"noTokenApi":false,"readonlyMode":false,"fahrenheit":false,"dallasParasite":false,"boolFormat":1,"boolDashboard":1,"enumFormat":1,"analogEnabled":false,"telnetEnabled":true,"maxWebLogBuffer":100,"modbusEnabled":false,"forceHeatingOff":false,"developerMode":true},"devices":[{"type":"boiler","name":"Condens, Logamax/Logomatic, Cerapur Top, Greenstar, Generic HT3","deviceID":"0x08","productID":95,"brand":"","version":"12.16","entities":66,"handlersReceived":"0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x04","handlersFetched":"0x14 0x16 0x33","handlersPending":"0xBF 0xC2 0x1A 0x35 0x26 0x2A 0x2E 0x3B","handlersIgnored":"0x23"},{"type":"thermostat","name":"CR50/RC200, CW100, CR120, CR50","deviceID":"0x18","productID":157,"brand":"","version":"43.02","entities":51,"handlersReceived":"0x06 0xA2 0x031D","handlersFetched":"0x02A5 0x02B9 0x02AF 0x029B 0x0291 0x02F5 0x023A 0x0241","handlersPending":"0xA3 0x12 0x13 0x0471 0x02A6 0x02BA 0x02B0 0x029C 0x0472 0x02A7 0x02BB 0x02B1 0x029D 0x0473 0x02A8 0x02BC 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CC 0x0467 0x02CE 0x0468 0x0292 0x02D0 0x0469 0x0293 0x02D2 0x046A 0x0294 0x02F6 0x031B 0x031E 0x0267 0xBB 0x023E 0x0269 0x016E","handlersIgnored":"0x043F 0xBF"},{"type":"controller","name":"HT3","deviceID":"0x09","productID":95,"brand":"","version":"12.16","entities":0},{"type":"temperaturesensor","name":"temperaturesensor","entities":3},{"type":"custom","name":"custom","entities":2}]}

```

### TO REPRODUCE

CR50 thermostat has "holiday" mode that is nowhere seen

### EXPECTED BEHAVIOUR

This is implemented in very simple way in this thermostat. One goes through the menu and sets holiday mode to ON and it asks you to set starting date and ending date of the mode. When the start date is reached the thermostat indicates that it is in holiday mode.

I believe that the dates are set using R/W telegram 043F in following format:

To turn mode off, set: 11 01 01 11 01 01
To turn mode on, set: 19 02 0F 19 02 10 (which means start on 24/02/15 end on 24/02/16)

Then I believe that there is also indication when it is actually ON (that is, between the above set dates) in the telegram 02A5 offset 18:

When the holiday mode is off the byte is: 00
When the holiday mode is on the byte is: 04

### SCREENSHOTS

[image]

[image]

[image]

### ADDITIONAL CONTEXT

From reading, it seems similar to RC300 (only start/end date, not temperature). Happy to do more testing for you if necessary.

**(Please remember to close the issue when it has been addressed)**

VlastiBroucek, 2025-02-15:
Looking into it bit further, it seems that also offset 10 of telegram 2A5 changes from 01 to 00 when ofset 18 goes to 04

00 E6 21 25 00 00 25 26 02 4C 01 01 03 02 4C 00 48 00 00 11 01 04 09 01 00 00 07 00 00 00 00 00 00 36 00 3C 01 FF 00 00 
00 E6 21 25 00 00 25 26 02 4B 01 01 03 02 4B 00 49 00 00 11 01 04 09 01 00 00 07 00 00 00 00 00 00 36 00 3C 01 FF 00 00
00 E6 21 25 00 00 25 26 02 4A 01 01 03 02 4A 00 4A 00 00 11 01 04 09 01 00 00 07 00 00 00 00 00 00 36 00 3C 01 FF 00 00
00 E6 21 25 00 00 25 26 02 48 **01** 01 03 02 48 00 4C 00 **00** 11 01 04 09 01 00 00 07 00 00 00 00 00 00 36 00 3C 01 FF 00 00
00 E6 21 25 00 00 25 25 05 A0 **00** 01 03 02 48 00 4C 00 **04** 11 01 04 09 01 00 00 07 00 00 00 00 00 00 36 00 3C 01 FF 00 00
00 E6 21 25 00 00 25 25 05 A0 **00** 01 03 02 47 00 4D 00 **04** 11 01 04 09 01 00 00 07 00 00 00 00 00 00 36 00 3C 01 FF 00 00
00 E6 21 25 00 00 25 26 02 46 **01** 01 03 02 46 00 4E 00 **00** 11 01 04 09 01 00 00 07 00 00 00 00 00 00 36 00 3C 01 FF 00 00
00 E6 21 25 00 00 25 26 02 44 01 01 03 02 44 00 50 00 00 11 01 04 08 FF 00 00 07 00 00 00 00 00 00 36 00 3C 01 FF 00 00
00 E6 21 25 00 00 25 26 02 42 01 01 03 02 42 00 52 00 00 11 01 04 08 FF 00 00 07 00 00 00 00 00 00 36 00 3C 01 FF 00 00
00 E7 21 25 00 00 25 26 02 3F 01 01 03 02 3F 00 55 00 00 11 01 04 09 06 00 00 07 00 00 00 00 00 00 36 00 3C 01 FF 00 00
00 E8 21 25 00 00 25 26 02 3A **01** 01 03 02 3A 00 5A 00 **00** 11 01 04 09 12 00 00 07 00 00 00 00 00 00 36 00 3C 01 FF 00 00
00 E8 21 25 00 00 25 25 05 A0 **00** 01 03 02 38 00 5C 00 **04** 11 01 04 09 14 00
00 E8 21 25 00 00 25 26 02 37 **01** 01 03 02 37 00 5D 00 **00** 11 01 04 09 14 00

VlastiBroucek, 2025-02-16:
**Further information**
When in holiday mode the desired temperature is set to "eco" (in the case of this particular thermostat it is called "setback") temperature.

TBH honest, I was hoping it will switch the boiler to "holiday mode". The boiler has button for it

[image]

I'll see if there will be anything visible on the bus if I use the button on the boiler and if yes, if it then can be changed from ems-esp. Any idea how to best capture what will be happening on the bus when I do this? Syslog and eagle eyes or should it be enough to watch boiler?

VlastiBroucek, 2025-02-16:
> Looking into it bit further, it seems that also offset 10 of telegram 2A5 changes from 01 to 00 when ofset 18 goes to 04
>
> 00 E6 21 25 00 00 25 26 02 4C 01 01 03 02 4C 00 48 00 00 11 01 04 09 01 00 [...]

I think I was too fast with this one - there are completely different numbers at offset 10 today... I now think only offset 18 changes.

MichaelDvP, 2025-02-16:
Hmm, 0x2A5 is a hc1 specific telegram, unclear if 0x43F is only hc1 or thermostat specific. But CR50 onlysupports one hc, so it do not matter much.
Naming is once again nasty with Bosch brands. Some thermostats call it holiday, some vacation, and some have holiday and vacaion modes, where holiday means "at home (warm)" and vacation means "away from home (cold)". 
You've mentioned that it gores to "eco", so i stick to vacationmode.
Please check if https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest works.

VlastiBroucek, 2025-02-16:
> Hmm, 0x2A5 is a hc1 specific telegram, unclear if 0x43F is only hc1 or thermostat specific. But CR50 onlysupports one hc, so it do not matter much. Naming is once again nasty with Bosch brands. Some [...]

Hi @MichaelDvP ,

**this works perfectly!** You guys rock!

It's not exactly what I was originally hoping to achieve when submitting this - turning the boiler to "holiday mode" as defined in the boiler manual, i.e. turning it nearly completely off, but since it does not seem to be working on the boiler anyway, I guess this will have to do.

VlastiBroucek, 2025-02-16:
BTW, I would not blame Bosch for the confusion in naming. I would blame variants of English used all around the world. E.g. here in Australia, we have public holiday (day of some significance which is work free day - we go even so far, that if it walls on weekend, we get extra free day(s) for it during working days), but we also go on holiday(s) (you could hardly ever hear vacation here - probably only from those who learnt American English) when we travel.

MichaelDvP, 2025-02-16:
> urning the boiler to "holiday mode" as defined in the boiler manual, 

Try to log the button action, if there is a ems telegram publsihed, we can try to implement. The telegram 0x2A5 is just a status monitor and not writeable.

VlastiBroucek, 2025-02-18:
> > urning the boiler to "holiday mode" as defined in the boiler manual,
> 
> Try to log the button action, if there is a ems telegram published, we can try to implement. The telegram 0x2A5 is just a status monitor and not writeable.

Tried it - the button actually goes green after holding it 5 seconds, indicating "holiday mode". However, nothing actually changes, heating continues, hot water continues too. Seems to me, that CR50 has precedence. Another hold for 5 seconds turns the green light off.

Also, I did not manage to see anything that would indicate changing the button on the bus.

So, I am thinking, that I'll need to use HA automation to turn off DHW and lower the "eco" temperature when the "Vacation Status" goes TRUE and revert to previous setting when it goes back to FALSE.

VlastiBroucek, 2025-02-18:
Hi @proddy and @MichaelDvP ,

are you interested in people regularly submitting "improvements" or "corrections" about various devices as they find them?

For example, while looking at this issue, I noticed that ems-esp suggested values for "dhw operating mode" in my thermostat (CR50) are:

- off
- normal
- comfort
- auto
- own prog

However, my CR50 has only:

- ON
- AUTO 
- OFF

[image]

I am just not sure if you are keen on getting too many "customisations" into the system...

proddy, 2025-02-18:
> Hi [@proddy](https://github.com/proddy) and [@MichaelDvP](https://github.com/MichaelDvP) ,
>
> are you interested in people regularly submitting "improvements" or "corrections" about various devices as [...]

yes! I would consider this a bug and the EMS-ESP code is able to manage these different variations. We must however be careful not to introduce breaking changes, as we've found in the past there are multiple flavors for the same device depending on model numbers, manufacturing dates etc. So always good to double check with other users before merging any code changes.

VlastiBroucek, 2025-02-18:
> yes! I would consider this a bug and the EMS-ESP code is able to manage these different variations. We must however be careful not to introduce breaking changes, as we've found in the past there are [...]

So, do you want me to make the above a different issue/discussion?

Also, did anyone ever figured out which telegrams CR50 uses to store times for what they call "heating" and "setback" times? I did some searches here and did not find anything useful for this particular thermostat. Not asking to implement those, just curious...

[image]

closed 2025-02-19 by VlastiBroucek