emsesp/EMS-ESP32 #2409: CR50 thermostat - timers decoded

VlastiBroucek, 2025-02-19:
### DESCRIPTION

I believe I have decoded a telegram used for storing heating (also used for DHW in AUTO) timers in CR50 Thermostat.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [X] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [X] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
Thermostat: CR50/RC200, CW100, CR120, CR50 (DeviceID:0x18, ProductID:157, Version:43.02) (53)
 Received telegram type IDs: 0x06 0xA2 0x31D
 Fetched telegram type IDs: 0x2A5 0x2B9 0x2AF 0x29B 0x291 0x2F5 0x23A 0x241 0x43F
 Pending telegram type IDs: 0xA3 0x12 0x13 0x471 0x2A6 0x2BA 0x2B0 0x29C 0x472 0x2A7 0x2BB 0x2B1 0x29D 0x473 0x2A8 0x2BC 0x2B2 0x29E 0x474 0x2A9 0x2BD 0x2B3 0x29F 0x475 0x2AA 0x2BE 0x2B4 0x2A0 0x476 0x2AB 0x2BF 0x2B5 0x2A1 0x477 0x2AC 0x2C0 0x2B6 0x2A2 0x478 0x2CC 0x467 0x2CE 0x468 0x292 0x2D0 0x469 0x293 0x2D2 0x46A 0x294 0x2F6 0x31B 0x31E 0x267 0xBB 0x23E 0x16E
 Ignored telegram type IDs: 0xBF 0x1C 0x04 0x16 0x18 0x19 0x34 0x33 0x14 0x15 0x29 0x1A 0x05 0x0A 0x17 0x1D 0x23 0x25 0x26 0x27 0x2A 0x2C3

```

### DETAILS

I believe I have decoded a telegram used for storing heating (also used for DHW in AUTO) timers in CR50 Thermostat.

I believe they are all saved in one long telegram only (0x02C3). 

`
0x02C3 (offset 00) 03 14 01 48 03 FF 01 FF 03 FF 01 FF means MO, H, 05:00, S, 18:00
0x02C3 (offset 12) 03 14 01 48 03 FF 01 FF 03 FF 01 FF means TU, H, 05:00, S, 18:00
0x02C3 (offset 24) 03 14 01 48 03 FF 01 FF 03 FF 01 FF means WE, H, 05:00, S, 18:00
0x02C3 (offset 36) 03 14 01 48 03 FF 01 FF 03 FF 01 FF means TH, H, 05:00, S, 18:00
0x02C3 (offset 48) 03 14 01 48 03 FF 01 FF 03 FF 01 FF means FR, H, 05:00, S, 18:00
0x02C3 (offset 60) 03 14 01 48 03 FF 01 FF 03 FF 01 FF means SA, H, 05:00, S, 18:00
0x02C3 (offset 72) 03 14 01 48 03 FF 01 FF 03 FF 01 FF means SU, H, 05:00, S, 18:00

03 - H (heating temp starts - temp selected elsewhere, not sure why 03 here - emsesp calls it comfort)
14 - time in count of 15 mins increments from midnight (i.e. 0x14x15=300 min, so 05:00) for first heating period of the day
01 - S (setback temp starts - temp selected elsewhere, not sure why 01 here - emsesp calls it eco)
48 - time in count of 15 mins increments from midnight (i.e. 0x48x15=1080 min, so 18:00) for first setback period of the day
03 - next heating starts
FF - not set
01 - next setback starts
FF - not set
03 - next heating
FF - not set
01 - next setback
FF - not set
`

### SCREENSHOTS

[image]

### ADDITIONAL CONTEXT

_Add any other context about the issue here._

**(Please remember to close the issue when it has been addressed)**

Sbried, 2025-02-19:
I added this plus some findings from my side to [https://docs.emsesp.org/ems-library/](url).
It will show up once @proddy approved the PR.

VlastiBroucek, 2025-02-19:
> I added this plus some findings from my side to [https://docs.emsesp.org/ems-library/](url).
> It will show up once @proddy approved the PR.
> 
> For DHW it should be 0x02FF. Maybe @VlastiBroucek can check?

CR50 doesn't have separate timers for DHW.

It also doesn't have anything in ofsetts 84-101

VlastiBroucek, 2025-02-20:
> I added this plus some findings from my side to [https://docs.emsesp.org/ems-library/](url). It will show up once [@proddy](https://github.com/proddy) approved the PR.
> 
> For DHW it should be 0x02FF. Maybe [@VlastiBroucek](https://github.com/VlastiBroucek) can check?

Hi @Sbried, thanks for doing this, but I am not sure if this:

[image]

is really correct. My CR50 Thermostat has read only values of 03 for odd switch times and 01 for even switch times. Additionally the thermostat has only two temperature settings

- heating (comfort in EMS-ESP terminology) - code 03
- setback (eco in EMS-ESP terminology) - code 01

So, logically I think the yellow highlighted comment should probably be

1. ECO1
2. ECO2
3. COMFORT1
4. COMFORT2

but of course, I cannot confirm it.

Sbried, 2025-02-20:
I also can´t test it, it´s just a collection of several information.
I think that different control units may use different subsets of a telegram definition. That´s why you can´t give a name for your setting and I can give it a name.
I would suggest to keep the EMS-library as it is and change it when we find more information. At least it covers everything you reported plus additional thing that others may be able to confirm or correct.

MichaelDvP, 2025-02-20:
See #1594

Sbried, 2025-02-20:
OK, following the link(s) I see there are much more telegrams used for timers. I will collect all information in the EMS library and add also the hardware used with each of them.

VlastiBroucek, 2025-03-01:
Hi @Sbried,

I am looking at telegrams 0x0C0-0x0C2 for CR50 Thermostat. These seems to be holding a record of errors/maintenance messages received from boiler... I am just starting, and I have very little only, so before I start digging, I was curious if you know anything about it. So far it seems they are 200 bytes long each and having one message per every 20 bytes... Right now, I think I correctly see this in 0x0C0:
```
08 5F 00 12 11 48 30 38 03 FA 99 03 00 01 00 00 00 00 00 00
08 5F 00 12 11 48 30 38 03 FA 99 02 12 11 00 99 02 12 11 02
08 5F 00 12 05 48 30 38 03 FA 99 02 10 11 0B 99 02 10 11 11
08 5F 00 0A 05 20 43 36 00 14 98 0A 00 08 34 98 0A 00 08 3A 
08 5F 00 0A 05 20 42 31 00 1F 95 02 03 01 1F 95 02 03 01 30 
08 5F 00 0A 05 20 42 31 00 1F 95 02 03 01 0F 95 02 03 01 19
08 5F 00 0A 05 20 42 31 00 1F 95 02 03 01 0D 95 02 03 01 0E
08 5F 00 0A 05 20 42 31 00 20 95 02 02 01 37 95 02 03 01 0D 
08 5F 00 0A 05 20 42 31 00 1F 95 02 02 01 32 95 02 02 01 37 
08 5F 00 0A 05 20 42 31 00 20 95 02 02 01 24 95 02 02 01 32

...
```
Offsets 5-7 seems to be ASCII code of message then offset 8-9 WORD representation of integer, so:
```
H08 1018 - which decodes as maintenance time reached and I see it on thermostat's display
H08 1018
H08 1018
 C6 0020 - which seems to be Fan not running
 B1 0031 - which seems to be Code plug not detected
 B1 0031
 B1 0031
 B1 0031
 B1 0032
 B1 0031
 B1 0032
```

Now offset 11-13 is MM??DD:
```
03??01
02??17
02??17
10??08
...
```

It seems that after new error/message, it moves 20 bytes down - I can see it by comparing what I collected few days ago and again today after I received new maintenance message on the thermostat

Anyway, this is how far I've got... What is also confusing me is that the physical display on thermostat shows me only two past errors - 1/3/2025 H08 1018 and 17/02/2025 H08 1018 again. I remember seeing C6 something and B1 something there before, but I did reset at some stage...

Would love to hear from you. Perhaps we could open it as a separate discussion and see if someone else knows anything about it?

Sbried, 2025-03-01:
I also looked at that in the past. 
0x00C0 is a list of 1...10 errors and 0x00C1 is a list of 11..20 errors.
0x00C2 is like a local error list of 1...10 errors, different per device. In my case the local error list of my heat pump at 0x00C2 is the same as in 0x00C0 because I have only errors from the heat pump. And the local error list of my controller at 0x00C2 is empty, showing `00` at index 0, 20, 40...

I would classify the telegrams as follows:
0x00C0 and 0x00C1 is the global system error list, held by the controller
0x00C2 is a local error list held by each device

Each Error has 20 bytes:
Offset 0: source address (if 0x00, then it is no error)
Offset 1: module ID (then same as in 0x0002 offset 0)
Offset 2: always 0. Maybe a placeholder to be able to extend offset 1
Offset 3: Error reason. I found like 0x20=maintenance, 0x18=reset maintenance, 0x0A=safety shutdown and 0x0C=intended shutdown. Maybe there are others
Offset 4: Bit field. 

- Bit 0: logging the error yes/no
- Bit 1...4: display level: 0=don´t display, 2=expert, 4=installer, 8=user

Offset 5-7: error code ASCII
Offset 8-9: error cause
Offset 10: year of error start. This is somehow complicated. If you ignore bit 7, bits 0...6 hold the year + 2000
Offset 11: month of error start
Offset 12: hour of error start !! I would have expected to have day on offset 12
Offset 13: day of error start !! I would have expected to have hour on offset 13
Offset 14: minute of error start
Offset 15: year of error end. Same as offset 10
Offset 16: month of error end
Offset 17: hour of error end !! I would have expected to have day on offset 17
Offset 18: day of error end !! I would have expected to have hour on offset 18
Offset 19: minute of error end

Please check if all this is somehow plausible.

VlastiBroucek, 2025-03-02:
@Sbried, I think that your explanation is plausible and very correct (including somehow mixed hours, minutes, days etc. - only Bosch people will know).

I went through my 0x0C0 and 0x0C2 and here is what I believe I see there and what reflects the reality

### 0x0C0
```
08 5F 00 12 11 48 30 38 03 FA 99 03 00 01 00 00 00 00 00 00 
08 5F 00 12 11 48 30 38 03 FA 99 02 12 11 00 99 02 12 11 02 
08 5F 00 12 05 48 30 38 03 FA 99 02 10 11 0B 99 02 10 11 11 
08 5F 00 0A 05 20 43 36 00 14 98 0A 00 08 34 98 0A 00 08 3A 
08 5F 00 0A 05 20 42 31 00 1F 95 02 03 01 1F 95 02 03 01 30 
08 5F 00 0A 05 20 42 31 00 1F 95 02 03 01 0F 95 02 03 01 19 
08 5F 00 0A 05 20 42 31 00 1F 95 02 03 01 0D 95 02 03 01 0E 
08 5F 00 0A 05 20 42 31 00 20 95 02 02 01 37 95 02 03 01 0D 
08 5F 00 0A 05 20 42 31 00 1F 95 02 02 01 32 95 02 02 01 37 
08 5F 00 0A 05 20 42 31 00 20 95 02 02 01 24 95 02 02 01 32

08 - boiler
5F - boiler info (from 0x02)
00
12 - 12: maintenance
11 - 00010001: login=yes, 8=user
48 30 38 - H08
03 FA - 1018
99 - 10011001: 25 year
03 - 03: March
00 - 00
01 - 01 (I had the maintenance date set to 01/03/2025 while waiting for it after it jumped on in February)
00 - 00
00 - seems to indicate current???
00
00
00
00

Telegram 0xC1
08 - boiler
5F - boiler info (from 0x02)
00
12 - 12: maintenance
11 - 00010001: login=yes, 8=user
48 30 38 - H08
03 FA - 1018
99 - 10011001: 25
02 - 02: February
12 - 18 hours
11 - 17 day
00 - 00 minutes
99 - 1001 1001: 25 
02 - 02: February 
12 - 18 hours
11 - 17 day
02 - 02 minutes

08 - boiler
5F - boiler info (from 0x02)
00
12 - 0x12: maintenance
05 - 00000101: login=yes, 2=expert (I remember changing it to show to user after this)
48 30 38 - H08
03 FA - 1018
99 - 10011001: 25
02 - 02: February
10 - 16 hours 
11 - 17 day
0B - 11 minutes
99 - 10011001: 25
02 - 02: February 
10 - 16 hours
11 - 17 day
11 - 17 minutes

08 - boiler
5F - boiler info (from 0x02)
00
0A - 0x0A - safety shutdown?
05 - 00000101: login=yes, 2=expert 
20 43 36 - C6: Fan not running
00 14 - 0024
98 - 10011000: 24
0A - 10 October
00 - 00 hours
08 - 08 day
34 - 52 min
98 - 10011000: 24
0A - 10 October
00 - 00 hours
08 - 08 day
3A - 58 min

08 - boiler
5F - boiler info (from 0x02)
00
0A - 0x0A - safety shutdown?
05 - 00000101: login=yes, 2=expert 
20 42 31 - B1: Code plug not detected
00 1F - 0031
95 - 10010101: 21 year
02 - 02: February
03 - 03 hours
01 - 01 day
1F - 31 minutes
95 - 10010101: 21 year
02 - 02: February
03 - 03 hours
01 - 01 day
30 - 48 minutes
```
The last one seems to be from the time of installation

### 0x0C2
```
18 9D 00 10 05 41 36 31 04 0E 00 00 00 3F 00 95 01 00 1F 00
00 9D 00 00 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 9D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 9D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 9D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 9D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 9D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 9D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 9D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 9D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

18  - thermostat
9D - thermostat info from (0x02)
00 
10 - 0x10 no idea what it means
05 - 00000101: login=yes, 2=expert
41 36 31 - A61 BUS error
04 0E - 1038
00 - 00000000: year not set?
00 - 00 Month
00 - 00 hours
3F - 63 day???
00 - 00 minutes
95 - 10010101: 21 year
01 - 01 January
00 - 00 Hours
1F - 31 day
00 - 00 minutes
```
This seems to be first power on for the thermostat???

Sbried, 2025-03-02:
Good to hear. Maybe we don´t know all details but there is an overall understanding of the telegram.
Regarding your last 0x00C2 telegram from the thermostat: Maybe the start date is weird because the date was not set at the time of first start up. In theory this date entry may be relative time since power up of the device. So, your day "63" can be something like 63 minutes since power up...
I will document all this in the EMS library [https://docs.emsesp.org/ems-library/](url)