emsesp/EMS-ESP32 #2410: CR50 thermostat - DHW Operating Mode incorrect

VlastiBroucek, 2025-02-19:
### DESCRIPTION

EMS-ESP showing DHW modes that do not exist on this thermostat

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [X] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [X] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [] Provide the System information in the area below, taken from `http://<IP>/api/system`

```
Thermostat: CR50/RC200, CW100, CR120, CR50 (DeviceID:0x18, ProductID:157, Version:43.02) (53)
 Received telegram type IDs: 0x06 0xA2 0x31D
 Fetched telegram type IDs: 0x2A5 0x2B9 0x2AF 0x29B 0x291 0x2F5 0x23A 0x241 0x43F
 Pending telegram type IDs: 0xA3 0x12 0x13 0x471 0x2A6 0x2BA 0x2B0 0x29C 0x472 0x2A7 0x2BB 0x2B1 0x29D 0x473 0x2A8 0x2BC 0x2B2 0x29E 0x474 0x2A9 0x2BD 0x2B3 0x29F 0x475 0x2AA 0x2BE 0x2B4 0x2A0 0x476 0x2AB 0x2BF 0x2B5 0x2A1 0x477 0x2AC 0x2C0 0x2B6 0x2A2 0x478 0x2CC 0x467 0x2CE 0x468 0x292 0x2D0 0x469 0x293 0x2D2 0x46A 0x294 0x2F6 0x31B 0x31E 0x267 0xBB 0x23E 0x16E
 Ignored telegram type IDs: 0xBF 0x1C 0x04 0x16 0x18 0x19 0x34 0x33 0x14 0x15 0x29 0x1A 0x05 0x0A 0x17 0x1D 0x23 0x25 0x26 0x27 0x2A 0x2C3
```

### TO REPRODUCE

EMS-ESP suggested values for "dhw operating mode" are:
- off
- normal
- comfort
- auto
- own prog

### EXPECTED BEHAVIOUR

However, my CR50 has only:
- ON
- AUTO
- OFF

OFF sets 0x02F5 offset 2 to 00 nad 0x031D offset 2 to 03
ON sets 0x02F5 offset 2 to 02 nad 0x031D offset 2 to 05
AUTO sets 0x02F5 offset 2 to 03 nad 0x031D offset 2 to 08

when setting thermostat to ON, EMS-ESP currently maps it to comfort, OFF and AUTO map correctly to off and auto.

### SCREENSHOTS

[image]

### ADDITIONAL CONTEXT

_Add any other context about the issue here._

**(Please remember to close the issue when it has been addressed)**

proddy, 2025-02-21:
Michael added, in 3.7.2-dev.21

VlastiBroucek, 2025-02-22:
Hi guys, thanks for working on this, but unfortunately it does not help and makes things even worse. I have done some further reading of telegrams, and this is what I am getting with DEV21 (full telegrams this time):

1. Thermostat AUTO and active HC1 program is COMFORT
[emsesp] thermostat(0x18) W me(0x0B), RC300WWmode(0x02F5), data: 01 00 **03** 01 00 00 08 01 00 08 04 00
[emsesp] thermostat(0x18) W me(0x0B), RC300WWmode2(0x031D), data: 00 00 **0A** 07
EMS-ESP shows OFF

2. Thermostat AUTO and active HC1 program is ECO
[emsesp] thermostat(0x18) W me(0x0B), RC300WWmode(0x02F5), data: 01 00 **03** 01 00 00 08 01 00 08 04 00
[emsesp] thermostat(0x18) W me(0x0B), RC300WWmode2(0x031D), data: 00 00 **0A** 07
EMS-ESP shows OFF

3. Thermostat OFF
[emsesp] thermostat(0x18) W me(0x0B), RC300WWmode(0x02F5), data: 01 00 **00** 01 00 00 08 01 00 08 04 00
[emsesp] thermostat(0x18) W me(0x0B), RC300WWmode2(0x031D), data: 00 00 **03** 07
EMS-ESP shows OFF

4. Thermostat ON
[emsesp] thermostat(0x18) W me(0x0B), RC300WWmode(0x02F5), data: 01 00 **02** 01 00 00 08 01 00 08 04 00
[emsesp] thermostat(0x18) W me(0x0B), RC300WWmode2(0x031D), data: 00 00 **05** 07
EMS-ESP shows NORMAL

The cases 1. and 2. should be showing as AUTO, but it shows off

[image]

I am not sure of 0A value in 0x031D in the second case - it seems that if I change from any other program to AUTO (using thermostat!) while HC1 program is ECO, then the value is set to 08 and while HC1program is COMFORT to 0A, but I am struggling to replicate this consistently. It's like if something else is forcing that 0A into it. I even tried watching 0x031D during the change of the program from COMFORT to ECO and it remained at 0A. Perhaps because the CR50 has only one DHW mode?

I am obviously still learning about this, so will appreciate any advice, solution.

Cheers

MichaelDvP, 2025-02-22:
You are comparing HC1 modes with DHW telegrams. What do you want to express with this?

VlastiBroucek, 2025-02-22:
Hi @MichaelDvP,

> You are comparing HC1 modes with DHW telegrams.

Yes, I know. I am talking about **DHW mode setting** but it's bit more complex so:

1. I am trying to give you as much info as possible
2.  Sorry, when I said above "Thermostat AUTO and active HC1 program is COMFORT" I should have said "**Thermostat DHW MODE AUTO** and active HC1 program is COMFORT" etc
4. Since CR50 does not have its own program/timers for **DHW mode,** it is using HC1 program/timers in AUTO (see picture from the manual in the original post).> What do you want to express with this?

What do I want to achieve:

1. Most importantly, I would like EMS-ESP to show same values in "dhw operating mode" as my thermostat shows for **DHW mode** - i.e. ON, OFF or AUTO. There is no other info shown - well there is - when in AUTO, it shows HC1 programming chart, so I know if I am in "heating/comfort" or "setback/eco" period. And that's the reason why I am getting your HC1 info too!  (see picture below)

[image]

2.  I would also like to be able to use EMS-ESP (or HA via EMS-ESP) to set ON, OFF or AUTO for DHW mode. Right now, I see five different values "off, normal, comfort, auto, own prog" which does not reflect what thermostat has.  Ideally, I'd like to see only ON, OFF, AUTO, but I accept you cannot do too many variants, so I think I'd be happy if there was this mapping, and I would avoid using other two options:
```
Thermostat | EMS-ESP
------------------------ 
AUTO       | auto
ON         | normal
OFF        | off
```
I think that should be doable as it appears that "normal" means always on...

I still think that 0x31D offset 2 is somehow related to HC1 (in the case of this particular thermostat) and indicates which stage of AUTO it is in. It has again changed to 0A this morning because (I think) right now, I am in the "comfort" zone of the AUTO mode.

Sorry if this is somehow convoluted, happy to provide more details and or captures.

Sbried, 2025-02-23:
Telegram **0x031D** is for DHW system 1. It has at offset 0 DHW extra active (false=0, true=1), at offset 1 holiday mode active (false=0, true=1), at **offset 2** the status of the DHW setpoint with 10 stati (?=1, extra=2, **manual off=3**, manual low=4, **manual high=5**, holiday off=6, holiday low=7, **clock off=8**, clock  low=9, **clock high=10**), at offset 3 the status of the DHW circulation pump with 7 stati (?=1, extra=2, manual off=3, manual on =4, holiday off=5, clock off=6, clock on=7), at offset 4-5 the remaining time in DHW extra mode
To show the real DHW status in your system it would additionally be needed to look at your HC1 status when 0x02F5 offset 2 is 3:
**0x02F5** is for the parameters of DHW and is writeable. It shows at offset 0 the DHW configuration (none=0, dhw appliance=1, dhw module=2, fresh water station=4), at offset 1 if a ciculation pump is installed (yes=-1, no=0), at **offset 2** the DHW operation mode (**always off=0**, always on low=1, **always on high=2**, **HC program=3**, own program=4, always on eco=5).
Maybe by writing a 0 or 2 to 0x02F5 offset 2 would turn your DHW off or on and you can achieve what you wanted at (2.). You may try this. Writing a 3 should then set your mode to auto.

In general the manual of the RC50 shows more control options like defining a DHW timer program. I think in combination with your boiler only a subset of the RC50 capabilities are used.

MichaelDvP, 2025-02-23:
@VlastiBroucek  Please try if https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest works.

VlastiBroucek, 2025-02-23:
> @VlastiBroucek  Please try if https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest works.

Will try tomorrow morning... It's too late here.

Can I have a copy of CR50 manual you refer to? My manual doesn't have any other options for DHW. Maybe we are talking about different CR50 thermostats?

Sbried, 2025-02-23:
I looked at this manual [https://www.manualslib.com/manual/1409472/Bosch-Cr-50.html?page=5#manual](url).
But you´re right: the timer program in chapter 4.3 is about the heating, not dhw.

VlastiBroucek, 2025-02-23:
Hi all - thanks for fantastic work again

@MichaelDvP  - the fix above seems to be spot on, everything appears and works correctly

@Sbried - your description of 0x02F5 and 0x031D seems to be spot on too. All I can add is that in the case of my particular thermostat (that is CR50 as sold in Australia), the 0x031D offset 2 when in AUTO mode seems to be changing as follows:

- it goes from 08 to 0A exactly 30 minutes before HC1 timer goes from setback/eco to heating/comfort
- it goes from 0A to 08 exactly 30 minutes after HC1 timer goes from heating/comfort to setback/eco

@Sbried - I am attaching PDF versions of CR50 manuals (both customer and installer) - these are exactly same as hard copies I got with the physical thermostat. Let me know if you would be interested in the same for the boiler as it can hopefully help with your telegram research.

[cr50_controller_installation_instructions.pdf](https://github.com/user-attachments/files/18933339/cr50_controller_installation_instructions.pdf)
[cr50_operating_instructions.pdf](https://github.com/user-attachments/files/18933340/cr50_operating_instructions.pdf)

commit 77ff6104 (2025-02-24): Merge pull request #2422 from MichaelDvP/dev

second fix #2410
commit a8eb06be (2025-02-24): second fix #2410
closed 2025-02-24 by proddy