emsesp/EMS-ESP32 #2411: Bosch CS6800i - Incorrect display "Electric aux heater"

HEL400, 2025-02-19:
-->  **Bosch CS6800i**

When the electric auxiliary heater is ON, the state displayed in EMS-ESP32 is OFF. 
(the incorrect state will also be transmitted in MQTT)

Here are some screenshots of the Bosch- and the EMS-Display
NOTE: My system runs with German language (I'm from Austria), therefore I wrote the correct English device entities in red letters!

1) General state (Bosch and EMS):

[image]

[image]

2) Detailes state of step 1 (Bosch and EMS):

[image]

[image]

proddy, 2025-02-22:
Can attach here the System Info so we can deep dive into this issue

HEL400, 2025-02-23:
Oh sorry, what exactly do you need?
Here are some additional infos:

Heatpump:
Bosch CS6800i/WLW176i
SW: com.bosch.tt.rhp:hyd.5.aw.mt1
SW-Version: 7.10.0-7cfe7e9c

BBQ:
Gateway S3 (Standard WiFi Edition V3)
SW-Version:  3.7.1

proddy, 2025-02-23:
https://docs.emsesp.org/Support/

HEL400, 2025-02-23:
Hi proddy,
I apologize for my blindness   :-)   
here are the missing infos

[emsesp_system_info.json](https://github.com/user-attachments/files/18929835/emsesp_system_info.json)

proddy, 2025-02-24:
We have the Aux Heater Status as a boolean (auxHeaterStatus) - can you do a read from `8 488`. You can do this from Status->System Log, blue arrow (make sure Developer Mode is enabled in Application Settings).

for heater step (elHeatStep1) could you do a read on `8 49D`.

look for the response telegrams and post them back here

HEL400, 2025-02-26:
Hi Proddy,
attached is the system log in developer mode.

[log.txt](https://github.com/user-attachments/files/18979483/log.txt)

Sbried, 2025-02-26:
Actuallay 0x049D is a telegram for testing. No idea if here the monitor values will be shown. To monitor the aux heater status 0x048D should be better. Byte 3 bits 0,1 and 2 should show the steps of the additional heater while byte 8 should show on/off.
But to check it should be made sure that the additional heater is really on.

HEL400, 2025-02-27:
Well, it's hard to test that because the auxiliary heater can't be switched on manually.  
BUT since it definitely shows wrong with "488" and "49D", the change to query the "048D" values would probably be better, wouldn't it?

proddy, 2025-02-27:
> Well, it's hard to test that because the auxiliary heater can't be switched on manually. BUT since it definitely shows wrong with "488" and "49D", the change to query the "048D" values would probably be better, wouldn't it?

could you try it out using a custom entity (type bool, offset 8) for on/off and offset 3 uint8_t so we can extrapolate the enum. Only you can test this and validate if it matches the heatpump service screens.

NoPlayBack, 2025-02-28:
will join this thread here... have same issue with that signal, as stated in the forum already. Will be switching to 3.7.1 this weekend and then i can start to search for that information. But as stated from HEL400 already... it is hard to manipulate the heatpump in such a way that it activates the aux-heater at these temperatures...

HEL400, 2025-02-28:
I think I have an idea to start the aux heater in an easy way - I'll do this today or tomorrow

HEL400, 2025-03-01:
Well, here is the system log with the aux heater ACTIVE. 
The heater is on at the beginning of the protocol, then at the end it is off again.
Bosch Display: Aux heater status = "Warmwasser" (tapwater), step 1 = ON and step 2+3 = OFF

[log(1).txt](https://github.com/user-attachments/files/19038165/log.1.txt)

Sbried, 2025-03-01:
Fits good :-)

005+02:53:25.684 TRACE 4072: [emsesp] boiler(0x08) -> me(0x0B), W, HpPower(0x048D), data: 40 00 10 **11** 64 00 21 03 **03** 1E 00 1B 00 00 00 00 39 64 64 3B 20 26 63 E9 2D
005+02:54:24.505 TRACE 4475: [emsesp] boiler(0x08) -> me(0x0B), W, HpPower(0x048D), data: 40 00 10 **11** 64 00 21 03 **03** 1E 00 1B 00 00 00 00 39 64 64 3B 20 27 64 E9 2D
005+02:55:08.640 TRACE 4820: [emsesp] boiler(0x08) -> all(0x00), B, HpPower(0x048D), data:         **10** 64 00 00 03 **00** 00 00 1C 00 00 00 00 39 64 64 3B 20 28 12 (offset 3)
005+02:55:23.230 TRACE 4890: [emsesp] boiler(0x08) -> me(0x0B), W, HpPower(0x048D), data: 40 00 10 **10** 64 00 00 03 **00** 00 00 19 00 00 00 00 35 5B 64 38 20 28 12 E9 2D

Until trace 4072 byte 3 is 11, means aux heater step3=0, 2=0, 1=1 => runs on step 1.
And byte 8 is 03 => dhw. Byte 8 = 01 would means aux heater for heating
From trace 4820 on byte 3 is 10, means aux heater step3=step2=step1=0.
And also byte 8 is 00, means aux heater =off

HEL400, 2025-03-01:
... and looks perfect   :-)
It seems that's the solution

proddy, 2025-03-05:
if we want to implement this into EMS-ESP I would need an idea of the naming etc. Screen shots of Custom Entities is a great help.

commit 7eb15652 (2025-03-07): Merge pull request #2449 from MichaelDvP/dev

fix #2446, pc0Flow for all boilers, fix #2411
commit c4f6f01f (2025-03-07): fix #2446, pc0Flow for all boilers, fix #2411
closed 2025-03-07 by proddy