emsesp/EMS-ESP32 #2427: IMPROVEMENT: Please can HELP topics open in new window/tab?

VlastiBroucek, 2025-02-26:
### DESCRIPTION

Please make three links in Help to open in the new window/tab

[image]

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [X] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [X] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the System information in the area below, taken from `http://<IP>/api/system`

### TO REPRODUCE

Clicking on the three topics in the Help page open them in the same window, link while (c) link at the bottom opens in new window/tab

If someone is looking for additional help while setting something, the last thing one needs is to be taken away from ems-esp window...

### EXPECTED BEHAVIOUR

So, I think the better would be to open them in new window/tab using ``target="_blank"``

Additionally, ``rel="noopener noreferrer"`` should also be used whenever ``target="_blank"`` is used - i.e. the (c) link at the bottom.

From the security/privacy point of view, the links that open in new window should always look like this
``<a href="https://example.com" target="_blank" rel="noopener noreferrer">Visit Example</a>``

**(Please remember to close the issue when it has been addressed)**

proddy, 2025-02-26:
good one. I'll add.

proddy, 2025-02-26:
added `target="_blank" rel="noreferrer"` (which includes noopener)

closed 2025-02-26 by proddy