emsesp/EMS-ESP32 #2437: EMS ESP not fetching needed sensor automatically

RobPee2023, 2025-03-02:
### DESCRIPTION

Environment
Device: CR10+ Thermostat
Integration: EMS-ESP
EMS-ESP Version: 3.7.1

Issue Description
I am using EMS-ESP to integrate a CR10+ thermostat into Home Assistant, managing temperature settings across different modes (Manual/Auto). However, I am encountering issues with how the ESP EMS is retrieving some sensors. Whereas for homeassistant (or just through the EMS interface) I can't change the temperature currently. By default there are 14 entities being fetched.
But for a proper working these 2 should be fetched as well:
**Thermostat/mode**
**Thermostat/manualtemp**

When fetching them automatically through Telnet/console it is working fine, and both entities are being fetched and added in the EMS-ESP device information. From this moment it is possible to adjust the temperature

_EMS-ESP:# call thermostat mode manual
000+00:08:21.026 I 15: [command] Called command thermostat/mode (Modus) with val                                 ue manual
EMS-ESP:# call thermostat manualtemp 17.5
000+00:08:11.516 I 14: [command] Called command thermostat/manualtemp (Temperatu                                 ur handmatig) with value 17.5_

The mapping is as following:
When in automatic mode it uses the Thermostat/manualtemp
When in manual mode it uses directly the thermostat/seltemp

So for a proper way fetching (reading) these entities every x-time would be great

### TO REPRODUCE

Restart the EMS ESP

### EXPECTED BEHAVIOUR

An additional setting at the device entities for auto fetching in the adjust entities menu would be a great solution or upon startup do an automatic fetch so these entities are automatically mapped to the EMS ESP

[image]

### SCREENSHOTS

_If applicable, add screenshots to help explain your issue._

### ADDITIONAL CONTEXT

_Add any other context about the issue here._

**(Please remember to close the issue when it has been addressed)**

proddy, 2025-03-02:
Could you attach the System Info (https://docs.emsesp.org/Support/) as it will show us which telegrams are fetched/received etc. 

The auto-fetching per entity via the customizations page is not possible as we can only automate the fetching of telegrams and the entities are hidden across the telegrams and differ per EMS models.

RobPee2023, 2025-03-02:
Hereby:

{
  "system": {
    "version": "3.7.1",
    "uptime": "000+00:03:42.444",
    "uptimeSec": 222,
    "platform": "ESP32",
    "cpuType": "ESP32-D0WD-V3",
    "arduino": "Tasmota Arduino v2.0.17",
    "sdk": "4.4.8.240628",
    "freeMem": 158,
    "maxAlloc": 99,
    "freeCaps": 108,
    "usedApp": 1740,
    "freeApp": 244,
    "partition": "app0",
    "resetReason": "Power on reset / APP CPU reset by PRO CPU",
    "psram": false,
    "model": ""
  },
  "network": {
    "network": "WiFi",
    "hostname": "EMS-ESP",
    "RSSI": -57,
    "WIFIReconnects": 0,
    "TxPowerSetting": 0,
    "staticIP": false,
    "lowBandwidth": false,
    "disableSleep": true,
    "enableMDNS": true,
    "enableCORS": false,
    "APProvisionMode": "disconnected",
    "APSecurity": "wpa2",
    "APSSID": "ems-esp"
  },
  "ntp": {
    "NTPStatus": "disconnected",
    "enabled": false,
    "server": "time.google.com",
    "tzLabel": "Europe/Amsterdam"
  },
  "mqtt": {
    "MQTTStatus": "connected",
    "MQTTPublishes": 150,
    "MQTTQueued": 0,
    "MQTTPublishFails": 0,
    "MQTTReconnects": 0,
    "enabled": true,
    "clientID": "EMS-ESP",
    "keepAlive": 60,
    "cleanSession": false,
    "entityFormat": 0,
    "base": "ems-esp",
    "discoveryPrefix": "homeassistant",
    "discoveryType": 0,
    "nestedFormat": 1,
    "haEnabled": true,
    "mqttQos": 0,
    "mqttRetain": false,
    "publishTimeHeartbeat": 60,
    "publishTimeBoiler": 10,
    "publishTimeThermostat": 10,
    "publishTimeSolar": 10,
    "publishTimeMixer": 10,
    "publishTimeWater": 10,
    "publishTimeOther": 60,
    "publishTimeSensor": 10,
    "publishSingle": false,
    "publish2command": false,
    "sendResponse": false
  },
  "syslog": {
    "enabled": false
  },
  "sensor": {

  },
  "api": {
    "APICalls": 0,
    "APIFails": 0
  },
  "bus": {
    "busStatus": "connected",
    "busProtocol": "HT3",
    "busTelegramsReceived": 332,
    "busReads": 129,
    "busWrites": 0,
    "busIncompleteTelegrams": 0,
    "busReadsFailed": 0,
    "busWritesFailed": 0,
    "busRxLineQuality": 100,
    "busTxLineQuality": 100
  },
  "settings": {
    "boardProfile": "CUSTOM",
    "locale": "nl",
    "txMode": 2,
    "emsBusID": 11,
    "showerTimer": false,
    "showerMinDuration": 180,
    "showerAlert": false,
    "phyType": 0,
    "rxGPIO": 23,
    "txGPIO": 22,
    "dallasGPIO": 0,
    "pbuttonGPIO": 0,
    "ledGPIO": 2,
    "hideLed": false,
    "noTokenApi": false,
    "readonlyMode": false,
    "fahrenheit": false,
    "dallasParasite": false,
    "boolFormat": 1,
    "boolDashboard": 1,
    "enumFormat": 1,
    "analogEnabled": false,
    "telnetEnabled": true,
    "maxWebLogBuffer": 25,
    "webLogBuffer": 25,
    "modbusEnabled": false,
    "forceHeatingOff": false,
    "developerMode": false
  },
  "devices": [
    {
      "type": "boiler",
      "name": "Condens, Logamax/Logomatic, Cerapur Top, Greenstar, Generic HT3",
      "deviceID": "0x08",
      "productID": 95,
      "brand": "",
      "version": "23.12",
      "entities": 66,
      "handlersReceived": "0x10 0x11 0x15 0x1C 0x18 0x19 0x34 0x04",
      "handlersFetched": "0x14 0x16 0x33",
      "handlersPending": "0xBF 0xC2 0x1A 0x35 0x2A 0x2E 0x3B"
    },
    {
      "type": "thermostat",
      "name": "CR10/RC100, CR10, Moduline 1000/1010",
      "deviceID": "0x18",
      "productID": 165,
      "brand": "",
      "version": "12.04",
      "entities": 14,
      "handlersReceived": "0xA2",
      "handlersFetched": "0x02A5 0x029B 0x0291",
      "handlersPending": "0xA3 0x06 0x12 0x13 0x02A6 0x02BA 0x02B0 0x029C 0x0472 0x02A7 0x02BB 0x02B1 0x029D 0x0473 0x02A8 0x02BC 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031D 0x031E 0x0267",
      "handlersIgnored": "0xBF 0x16 0x33"
    },
    {
      "type": "controller",
      "name": "HT3",
      "deviceID": "0x09",
      "productID": 95,
      "brand": "",
      "version": "23.12",
      "entities": 0
    }
  ]
}

MichaelDvP, 2025-03-02:
Please update to the latest dev. If the issue still exists (after a few minutes), please attach a full log. Both entities are in telegram 0x29B which is already fetched every minute.

RobPee2023, 2025-03-02:
Hi Michael,

Updated to the newest dev version (3.7.2-dev.23)
I've attached a full logging of the device running and enabled the logging to all for that output.

FYI out of all the possible entities (77) only 13 are showing up (previously before the dev update 14, can't recall which one isn't popping up currently).

`EMS-ESP:# log all
Log level: ALL
000+00:06:33.531 T 2: [emsesp] boiler(0x08) B all(0x00), UBAMonitorFast(0x18), data: 00 00 F9 00 00 00 02 00 C0 01 1D 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+00:06:33.740 T 3: [emsesp] boiler(0x08) B all(0x00), UBAMonitorWW(0x34), data: 32 01 1D 80 00 81 00 00 01 00 00 CE 65 00 B2 84 00
000+00:06:41.903 T 4: [emsesp] thermostat(0x18) B all(0x00), RC300Monitor(0x02A5), data: 00 C9 21 26 00 00 26 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 07 DD
000+00:06:42.101 T 5: [emsesp] thermostat(0x18) B all(0x00), RC300Monitor(0x02A5), data: 00 00 07 (offset 24)
000+00:06:43.530 T 6: [emsesp] boiler(0x08) B all(0x00), UBAMonitorFast(0x18), data: 00 00 F9 00 00 00 02 00 C0 01 1D 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+00:06:43.740 T 7: [emsesp] boiler(0x08) B all(0x00), UBAMonitorWW(0x34), data: 32 01 1D 80 00 81 00 00 01 00 00 CE 65 00 B2 84 00
000+00:06:53.529 T 8: [emsesp] boiler(0x08) B all(0x00), UBAMonitorFast(0x18), data: 00 00 F9 00 00 00 02 00 C0 01 1D 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
000+00:06:53.737 T 9: [emsesp] boiler(0x08) B all(0x00), UBAMonitorWW(0x34), data: 32 01 1D 80 00 81 00 00 01 00 00 CE 65 00 B2 84 00
000+00:07:00.212 T 10: [emsesp] me(0x0B) R boiler(0x08), UBATotalUptime(0x14), length: 27
000+00:07:00.254 T 11: [emsesp] boiler(0x08) W me(0x0B), UBATotalUptime(0x14), data: 3C 16 18
000+00:07:00.371 T 12: [emsesp] me(0x0B) R boiler(0x08), UBAParameters(0x16), length: 27
000+00:07:00.420 T 13: [emsesp] boiler(0x08) W me(0x0B), UBAParameters(0x16), data: FF 40 4C 00 00 FB 03 01 03 64 0A 02
000+00:07:00.544 T 14: [emsesp] me(0x0B) R boiler(0x08), UBAParameterWW(0x33), length: 27
000+00:07:00.577 T 15: [emsesp] boiler(0x08) W me(0x0B), UBAParameterWW(0x33), data: 08 FF 32 00 00 00 00 02 4B 00 FF FF
000+00:07:00.696 T 16: [emsesp] me(0x0B) R boiler(0x08), UBASettingsWW(0x26), length: 27
000+00:07:00.718 T 17: [emsesp] boiler(0x08) W me(0x0B), UBASettingsWW(0x26), data: <empty>
000+00:07:00.871 T 18: [emsesp] me(0x0B) R thermostat(0x18), RC300Monitor(0x02A5), length: 25
000+00:07:00.984 T 19: [emsesp] thermostat(0x18) W me(0x0B), RC300Monitor(0x02A5), data: 00 C9 21 26 00 00 26 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 07 DD 00
000+00:07:00.984 T 20: [emsesp] me(0x0B) R thermostat(0x18), RC300Monitor(0x02A5), length: 25 (offset 25)
000+00:07:01.035 T 21: [emsesp] thermostat(0x18) W me(0x0B), RC300Monitor(0x02A5), data: 07 00 07 (offset 24)
000+00:07:01.168 T 22: [emsesp] me(0x0B) R thermostat(0x18), RC300Set(0x02B9), length: 25
000+00:07:01.195 T 23: [emsesp] thermostat(0x18) W me(0x0B), RC300Set(0x02B9), data: <empty>
000+00:07:01.324 T 24: [emsesp] me(0x0B) R thermostat(0x18), RC300Summer(0x02AF), length: 25
000+00:07:01.355 T 25: [emsesp] thermostat(0x18) W me(0x0B), RC300Summer(0x02AF), data: <empty>
000+00:07:01.512 T 26: [emsesp] me(0x0B) R thermostat(0x18), RC300Curves(0x029B), length: 25
000+00:07:01.561 T 27: [emsesp] thermostat(0x18) W me(0x0B), RC300Curves(0x029B), data: 04 01 00 00 00 01 00 00 40
000+00:07:01.699 T 28: [emsesp] me(0x0B) R thermostat(0x18), RC300Summer2(0x0471), length: 25
000+00:07:01.731 T 29: [emsesp] thermostat(0x18) W me(0x0B), RC300Summer2(0x0471), data: <empty>
000+00:07:01.855 T 30: [emsesp] me(0x0B) R thermostat(0x18), RC300Set2(0x02CC), length: 25
000+00:07:01.882 T 31: [emsesp] thermostat(0x18) W me(0x0B), RC300Set2(0x02CC), data: <empty>
000+00:07:01.996 T 32: [emsesp] me(0x0B) R thermostat(0x18), HPSet(0x0467), length: 25
000+00:07:02.043 T 33: [emsesp] thermostat(0x18) W me(0x0B), HPSet(0x0467), data: <empty>
000+00:07:02.199 T 34: [emsesp] me(0x0B) R thermostat(0x18), HPMode(0x0291), length: 25
000+00:07:02.229 T 35: [emsesp] thermostat(0x18) W me(0x0B), HPMode(0x0291), data: 01
000+00:07:02.355 T 36: [emsesp] me(0x0B) R thermostat(0x18), HPMode(0x0292), length: 25
000+00:07:02.386 T 37: [emsesp] thermostat(0x18) W me(0x0B), HPMode(0x0292), data: <empty>
000+00:07:02.512 T 38: [emsesp] me(0x0B) R thermostat(0x18), HPMode(0x0293), length: 25
000+00:07:02.543 T 39: [emsesp] thermostat(0x18) W me(0x0B), HPMode(0x0293), data: <empty>
000+00:07:02.699 T 40: [emsesp] me(0x0B) R thermostat(0x18), HPMode(0x0294), length: 25
000+00:07:02.730 T 41: [emsesp] thermostat(0x18) W me(0x0B), HPMode(0x0294), data: <empty>
000+00:07:02.855 T 42: [emsesp] me(0x0B) R thermostat(0x18), RC300WWmode(0x02F5), length: 25
000+00:07:02.882 T 43: [emsesp] thermostat(0x18) W me(0x0B), RC300WWmode(0x02F5), data: <empty>
000+00:07:03.011 T 44: [emsesp] me(0x0B) R thermostat(0x18), RC300WW2mode(0x02F6), length: 25
000+00:07:03.041 T 45: [emsesp] thermostat(0x18) W me(0x0B), RC300WW2mode(0x02F6), data: <empty>
000+00:07:03.152 T 46: [emsesp] me(0x0B) R thermostat(0x18), RC300WWtemp(0x031B), length: 25
000+00:07:03.193 T 47: [emsesp] thermostat(0x18) W me(0x0B), RC300WWtemp(0x031B), data: <empty>
000+00:07:03.355 T 48: [emsesp] me(0x0B) R thermostat(0x18), RC300OutdoorTemp(0x023A), length: 25
000+00:07:03.385 T 49: [emsesp] thermostat(0x18) W me(0x0B), RC300OutdoorTemp(0x023A), data: <empty>
000+00:07:03.511 T 50: [emsesp] me(0x0B) R thermostat(0x18), RC300Settings(0x0241), length: 25
[... cut ...]
000+00:08:04.411 T 123: [emsesp] me(0x0B) R thermostat(0x18), PVSettings(0x023E), length: 25
000+00:08:04.440 T 124: [emsesp] thermostat(0x18) W me(0x0B), PVSettings(0x023E), data: <empty>
000+00:08:04.567 T 125: [emsesp] me(0x0B) R thermostat(0x18), CRHolidays(0x043F), length: 25
000+00:08:04.600 T 126: [emsesp] thermostat(0x18) W me(0x0B), CRHolidays(0x043F), data: <empty>
000+00:08:04.739 T 127: [emsesp] me(0x0B) R thermostat(0x18), Absent(0x016E), length: 25
000+00:08:04.783 T 128: [emsesp] thermostat(0x18) W me(0x0B), Absent(0x016E), data: <empty>
`

RobPee2023, 2025-03-02:
Hi Michael, hereby one attached of 10 minutes directly from the ESP EMS web interface. Previous one was copy paste through putty

[10min_ESPEMS_Log.txt](https://github.com/user-attachments/files/19044508/10min_ESPEMS_Log.txt)

MichaelDvP, 2025-03-03:
Thanks, the 0x29B is missing, don't know why, the CR thermostats are sometimes different. 
Please post the response from a `read 18 29B`

RobPee2023, 2025-03-03:
Here it is:

```
EMS-ESP:# read 18 29B
000+13:03:02.909 N 2: [emsesp] thermostat(0x18) W me(0x0B), RC300Curves(0x029B),                                 data: 04 01 00 00 00 01 00 00 40
```

MichaelDvP, 2025-03-03:
Ups, typo, sorry, i mean `read 18 2B9`

RobPee2023, 2025-03-03:
No worries:
```
EMS-ESP:# read 18 2b9
000+13:15:05.107 N 2: [emsesp] thermostat(0x18) W me(0x0B), RC300Set(0x02B9), data: <empty>
```

MichaelDvP, 2025-03-03:
Oh, we've seen similar for the CR11. Do the thermostat have different modes? Or is it fixed to manual?
If you set manualtemp, does it change the seltemp reading (and the real thermostat temperature)?

RobPee2023, 2025-03-03:
Behaviour looks to be as below:
There is an option to write manualo/auto to hc1/mode
when it is in manual i can write the hc1/seltemp
when it is in automatic i need to write to the hc1/manualtemp since the seltemp is constantly being overwritten by the manualtemp

MichaelDvP, 2025-03-03:
> There is an option to write manualo/auto to hc1/mode

yes, in ems-esp, but i think not in the thermostat, so this option is wrong. Please check if 
https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest
works.

RobPee2023, 2025-03-03:
Yes indeed I'm doing it through the console/telnet connection since these tags arent fetched by default. 
I'll update to the newest dev. version. keep you posted about it!

RobPee2023, 2025-03-03:
Hi Michael,

Just updated to the latest dev. Currently only 7 items are fetched.
Note: currently the hc1/Seltemp isn't settable, still need to use the console/telnet connection to write to hc1/manualtemp to adjust it
Or am I wrong that the Auto/Man (mode setting) should be on manual forever? (from the beginning I guess it has been on auto mode)

RobPee2023, 2025-03-03:
Currently the only way to show up the faceplate in HomeAssistant is when I've gone into console/telnet and execute these 2 commands:

call thermostat manualtemp xxx (doenst matter what value, just to get the sensor fetched once)
call thermostat mode manual/auto (doesn't matter, just to get the sensor fetched)

[image]

MichaelDvP, 2025-03-03:
Have you updated to the version from the link i've posted? The official dev is not changed, i'll do a PR if all is working  in my version.

RobPee2023, 2025-03-03:
I guess i didn't, i just flashed it and automatically the modus in in manual mode, Homeassistant faceplate is working fine.
As far as I see this topic can be closed,
Thanks for the efforts Michael!

RobPee2023, 2025-08-17:
**Thermostat: Junkers/Bosch CR10 EMS Plus**
Reason for reopening:
During the discussion in #2437, it was requested to verify if the CR10/CR11 has a mode other than MANUAL (which I didn't knew at the time).:
_"Oh, we've seen similar for the CR11. Do the thermostat have different modes? Or is it fixed to manual?
If you set manualtemp, does it change the seltemp reading (and the real thermostat temperature)?"_

in case a new case should be logged for this let me know and I'll do.

Current behavior shows that EMS‑ESP always reports MANUAL:
- when the setpoint ≥ 5°C
- when the thermostat knob is turned to OFF (lower than 5°C), even though the thermostat display clearly shows "OFF".
Behavior rules (per manual):

Any temperature ≥ 5°C → MANUAL (cannot set a setpoint physical or through EMS-ESP below 5°C)
Otherwise → OFF (display shows OFF) physical changing the thermostate.

Observed behavior:
Setting 10°C → EMS‑ESP correctly shows MANUAL
Setting OFF (display shows OFF, 0x02B9 offset 10 = 0x00) → EMS‑ESP still shows MANUAL
Debug logs:
Set 10°C:
`2025-08-17 14:51:33.655 T 53: [emsesp] thermostat(0x18) B all(0x00), RC300Set(0x02B9), data: 14 (offset 10)`

Set OFF:
`2025-08-17 14:52:13.015 T 83: [emsesp] thermostat(0x18) B all(0x00), RC300Set(0x02B9), data: 00 (offset 10)`

Suggested fix:
Currently, in thermostat.cpp:
```
} else if (model == EMSdevice::EMS_DEVICE_FLAG_CR11) {
    return HeatingCircuit::Mode::MANUAL;
}
```

This forces the thermostat mode to always be MANUAL. I suggest deriving the mode from the setpoint byte (telegram 0x02B9, offset 10):
```
if (_setpoint >= 5) {
    return HeatingCircuit::Mode::MANUAL;
} else {
    return HeatingCircuit::Mode::OFF;
}
```

Note: I haven’t found where _setpoint is defined in the code, so I haven’t cloned/tested this yet.

Additional notes:
- The thermostat does not send a dedicated “mode” value; OFF vs MANUAL must be inferred from the setpoint.
- Changing the thermostat mode to OFF should just set the temperature setpoint to 0°C. Any value between 0 and 5°C will be overwritten by the active setpoint.
- Implementing this change would allow EMS‑ESP to correctly detect when the knob is set to OFF.

RobPee2023, 2025-11-04:
Hi @proddy  & @MichaelDvP ,

I’ve just updated to the Dev version, this is exactly what I was hoping for. Thanks for adding this feature. Everything’s working perfectly so far. Great job, and appreciate the update!

From my side, this topic can be closed.

proddy, 2025-11-05:
@RobPee2023 thanks for the feedback, means a lot to us.