emsesp/EMS-ESP32 #2438: Lastcode missing?

Madaraxa, 2025-03-02:
I Updatet to 3.7.x and now 'lastcode' (last error Code) ist missing / without value.
If i read my api the following is the Response:
name":"lastcode","fullname":"last error code","circuit":"","type":"string","readable":true,"writeable":false,"visible":true}

So value is missing. Is this intended?

proddy, 2025-03-02:
Can you post your System Info, then we can determine which telegram (process_UBAErrorMessage) is being used

Madaraxa, 2025-03-03:
Hi. Is this enough? Or what Do you need? 

[image]

[image]

proddy, 2025-03-03:
You can download the info file from the webUI directly. see https://docs.emsesp.org/Support/

Madaraxa, 2025-03-03:
Ah sorry.

[emsesp_system_info.json](https://github.com/user-attachments/files/19049684/emsesp_system_info.json)

proddy, 2025-03-05:
Could you post back the results of reading the telegrams 0x10 and 0x11 from your boiler (0x08). Either via the Console's read command or via the System Log in the WebUI

Madaraxa, 2025-03-05:
I hope this was correct:
`ems-esp:$ read 0x08 0x10
083+14:06:21.990 N 4: [emsesp] boiler(0x08) -> me(0x0B), W, UBAErrorMessage1(0x10), data: 45 41 00 E3 95 0B 08 03 30 00 00 00
ems-esp:$ read 0x08 0x11
083+14:06:27.656 N 5: [emsesp] boiler(0x08) -> me(0x0B), W, UBAErrorMessage2(0x11), data: 45 41 00 E3 00 3F B6 DC 00 00 00 00
`
At the moment I put the commands in Putty the lastcode got filled. 
Now the result is:
{"name":"lastcode","fullname":"last error code","circuit":"","value":"EA(227) 03.11.2021 08:48 (0 min)","type":"string","readable":true,"writeable":false,"visible":true}

Madaraxa, 2025-03-06:
Im Not sure if thats correct behavior. If it is you can close this issue. But I got the feeling, that the value should be filled without using the web ui or the console.

proddy, 2025-03-06:
@MichaelDvP looks like we might need to 'fetch' UBAErrorMessage1 for Junkers?

proddy, 2025-03-06:
so @Madaraxa try a restart, and see if the value is fetched. it could have been a timing issue

Madaraxa, 2025-03-06:
Hi.
No, sorry. I restarted and this is the api Response
`{"name":"lastcode","fullname":"last error code","circuit":"","type":"string","readable":true,"writeable":false,"visible":true}
`
So no value...

Madaraxa, 2025-03-06:
Yes. Sorry. I'm new to Bug tracking in ems-esp

[log.txt](https://github.com/user-attachments/files/19114426/log.txt)

MichaelDvP, 2025-03-07:
Oh, difficult. Message 0x11 have only one message without valid date. ems-esp reads both telegrams and checks for the newest error. So we need always two reads with valid date to compare and publish.

Madaraxa, 2025-03-08:
Last dev Version seems to have fixed my Problem.
After Update and after another restart lastcode is filled.

closed 2025-03-08 by Madaraxa

proddy, 2025-03-08:
thanks. Nice work Michael.