emsesp/EMS-ESP32 #2440: remotehum command fails

proddy, 2025-03-02:
From a Discord thread, there was a report that the Remote Thermostat's humidity cannot be changed. It results in the error:

```
2025-03-01 11:08:36.748 W 3376: [command] Command remotehum: 44 failed (Error)
2025-03-01 11:08:36.748 E 3377: [mqtt] MQTT command failed with error Command remotehum: 44 failed (Error) (Error)
```

It's not easy to reproduce as we have no test data in the standalone unit tests.

MichaelDvP, 2025-03-03:
The remotehum command is only working for RC100H emulation (H=>humidity). What thermostat is emulated is  read from hcx/control. Not all master thermostats/systems allow to select RC100H.

proddy, 2025-03-03:
> The remotehum command is only working for RC100H emulation (H=>humidity). What thermostat is emulated is read from hcx/control. Not all master thermostats/systems allow to select RC100H.

Yes. I thought this was with the RC100H (as they reported). However, @mattreim reported it working, so perhaps it's not an issue after all.

But I would like to fix the error message, as now "message" is set to the humidity value sent to the command. I wanted to reproduce and fix this, but had no standalone test data.

MichaelDvP, 2025-03-03:
The reject (return false) is here:
https://github.com/emsesp/EMS-ESP32/blob/4d5a27f45ef4bb4882368e0f3115cf0b682fd4c3/src/devices/thermostat.cpp#L2055-L2056
For standalone you have to set RC310 with telegram 0x291 offset 3 to 3=RC100H for working humidity or a different value for rejecting humidity.

proddy, 2025-03-03:
thanks, I'll try and emulate it later. I was curious why the output json has the value in the message, even if the command fails.

https://github.com/emsesp/EMS-ESP32/blob/4d5a27f45ef4bb4882368e0f3115cf0b682fd4c3/src/core/command.cpp#L383

closed 2025-03-11 by proddy