emsesp/EMS-ESP32 #2448: Bosch Compress 7800i LW + Bosch Vent 5001C = NO energy data

MasterBytez, 2025-03-05:
I have installed the BBQKEES E32 V2, but I don't get any values ​​for the "energy consumed" of the ventilation system or for the "heat recovery". The ventilation system is correctly connected to the heat pump and also shows a few values. The energy data of the heat pump is all there. All the values ​​of the ventilation system are available in the Bosch HomeCom app, so you should be able to access them???

[image]

[image]

[image]

[image]
[image]

[image]
[image]

proddy, 2025-03-05:
Czn you attached your System Info?

MasterBytez, 2025-03-05:
That's enough?

[image]
[image]

[image]

proddy, 2025-03-06:
good enough. We ask for the System Info when dealing with issues. It would be in the Github template when you created the issue and also described In https://docs.emsesp.org/Support/ but many people miss it.

MasterBytez, 2025-03-08:
I'm sorry that I posted so hastily

{
  "system": {
    "version": "3.7.2-dev.25",
    "uptime": "000+04:06:37.448",
    "uptimeSec": 14797,
    "platform": "ESP32",
    "cpuType": "ESP32-D0WD-V3",
    "arduino": "ESP32 Arduino v2.0.17",
    "sdk": "v4.4.7-dirty",
    "freeMem": 151,
    "maxAlloc": 107,
    "freeCaps": 4169,
    "usedApp": 2239,
    "freeApp": 2433,
    "partition": "app0",
    "flash_chip_size": 16384,
    "resetReason": "Software reset CPU / Software reset CPU",
    "psram": true,
    "psramSize": 4093,
    "freePsram": 4052,
    "model": ""
  },
  "network": {
    "network": "Ethernet",
    "hostname": "ems-esp",
    "TxPowerSetting": 0,
    "staticIP": false,
    "lowBandwidth": false,
    "disableSleep": true,
    "enableMDNS": true,
    "enableCORS": false,
    "APProvisionMode": "never",
    "APSecurity": "wpa2",
    "APSSID": "ems-esp"
  },
  "ntp": {
    "NTPStatus": "disconnected",
    "enabled": false,
    "server": "time.google.com",
    "tzLabel": "Europe/Amsterdam"
  },
  "mqtt": {
    "MQTTStatus": "connected",
    "MQTTPublishes": 3713,
    "MQTTQueued": 0,
    "MQTTPublishFails": 0,
    "MQTTReconnects": 0,
    "enabled": true,
    "clientID": "esp32-62bbbc2c",
    "keepAlive": 60,
    "cleanSession": false,
    "entityFormat": 0,
    "base": "ems-esp",
    "discoveryPrefix": "homeassistant",
    "discoveryType": 0,
    "nestedFormat": 1,
    "haEnabled": true,
    "mqttQos": 0,
    "mqttRetain": false,
    "publishTimeHeartbeat": 60,
    "publishTimeBoiler": 10,
    "publishTimeThermostat": 10,
    "publishTimeSolar": 10,
    "publishTimeMixer": 10,
    "publishTimeWater": 10,
    "publishTimeOther": 60,
    "publishTimeSensor": 10,
    "publishSingle": false,
    "publish2command": false,
    "sendResponse": false
  },
  "syslog": {
    "enabled": false
  },
  "sensor": {
    "temperatureSensors": 0,
    "temperatureSensorReads": 0,
    "temperatureSensorFails": 0,
    "analogSensors": 0,
    "analogSensorReads": 0,
    "analogSensorFails": 0
  },
  "api": {
    "APICalls": 1,
    "APIFails": 0
  },
  "bus": {
    "busStatus": "connected",
    "busProtocol": "HT3",
    "busTelegramsReceived": 56563,
    "busReads": 14839,
    "busWrites": 0,
    "busIncompleteTelegrams": 0,
    "busReadsFailed": 24,
    "busWritesFailed": 0,
    "busRxLineQuality": 100,
    "busTxLineQuality": 100
  },
  "settings": {
    "boardProfile": "E32V2",
    "locale": "de",
    "txMode": 1,
    "emsBusID": 11,
    "showerTimer": false,
    "showerMinDuration": 180,
    "showerAlert": false,
    "hideLed": false,
    "noTokenApi": false,
    "readonlyMode": true,
    "fahrenheit": false,
    "dallasParasite": false,
    "boolFormat": 1,
    "boolDashboard": 1,
    "enumFormat": 1,
    "analogEnabled": true,
    "telnetEnabled": true,
    "maxWebLogBuffer": 50,
    "modbusEnabled": false,
    "forceHeatingOff": false,
    "developerMode": false
  },
  "devices": [
    {
      "type": "boiler",
      "name": "Bosch Compress 7800i LW Wärmepumpe",
      "deviceID": "0x08",
      "productID": 172,
      "brand": "Bosch",
      "version": "03.03",
      "entities": 157,
      "handlersReceived": "0xBF 0xC2 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F",
      "handlersFetched": "0x14 0xE6 0xEA 0x048D 0x048A 0x04A2 0x0485 0x0486 0x0492 0x0488 0x0484 0x048B 0x0491 0x0499 0x049C 0x049D 0x02CC 0x04A5 0x04AA 0x04A7 0x02D6",
      "handlersPending": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0x28 0x04AE 0x04AF",
      "handlersIgnored": "0xF7 0x04A0 0x0240 0x0291 0x029B 0x0489 0xE7 0x02E0 0x02EA 0x04AC 0x061E 0x0497 0x2E 0x3B"
    },
    {
      "type": "thermostat",
      "name": "Bosch Compress 7800i LW Steuerung",
      "deviceID": "0x10",
      "productID": 253,
      "brand": "",
      "version": "47.07",
      "entities": 77,
      "handlersReceived": "0x06 0x02BA 0x02BB 0x02BC 0x031D 0x0267",
      "handlersFetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02CC 0x0467 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240 0xBB 0x023E 0x016E",
      "handlersPending": "0xA3 0xA2 0x12 0x13 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x02F6 0x031B 0x031E 0x0269",
      "handlersIgnored": "0x0507 0x0508 0x0509 0x050A 0x059D 0xBF"
    },
    {
      "type": "gateway",
      "name": "Bosch K30RF/WiFi Modul",
      "deviceID": "0x48",
      "productID": 252,
      "brand": "Bosch",
      "version": "11.01",
      "entities": 0,
      "handlersIgnored": "0xF9 0x2040 0x0495 0x0494 0x3B 0x0590"
    },
    {
      "type": "ventilation",
      "name": "Bosch Vent 5001C 260",
      "deviceID": "0x51",
      "productID": 231,
      "brand": "",
      "version": "53.02",
      "entities": 10,
      "handlersReceived": "0x0585 0x0583 0x05D9 0x0587",
      "handlersFetched": "0x056B",
      "handlersIgnored": "0x0589 0x0575 0x0586 0x0588 0x058D 0x058E 0x058F 0x05CF 0x05E3 0xBF 0x0590"
    }
  ]
}

proddy, 2025-03-08:
You'll need to help out here. My German isn't good. Are you saying "energy consumed" and "heat consumed" are in EMS-ESP but not showing value or are they just not there and need to be added (as an enhancement)? And which device would provide this data? 

https://docs.emsesp.org/Troubleshooting/#im-missing-certain-data-from-an-ems-device