emsesp/EMS-ESP32 #2451: Improvement: Support nested conditional expressions in the scheduler

philwingfield, 2025-03-07:
### DESCRIPTION

The scheduler expression language current supports a single conditional express e.g. <cond> ? <expr1> : <expr2>
This is quite limited in scope and it would be nice to support nested ones to enable more complex functionality.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [x] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the System information in the area below, taken from `http://<IP>/api/system`

Sorry I can't paste the system info but see here for the docs where it describes the limitation
https://docs.emsesp.org/Configuring/#conditions

```

### TO REPRODUCE

Try to enter any nested conditional or multiple conditionals e.g. any of the following will fail

- "1<2?(3<4?5:6):7" should give 5
- "1>2?(3<4?5:6):7" should give 7
- "1<2?(3>4?5:6):7" should give 6
- "1<2?3:(4<5?6:7)" should give 3
- "1>2?3:(4<5?6:7)" should give 6
- "1>2?3:(4>5?6:7)" should give 7
- "(1>2?3:4)+(10>20?30:40)" should give 44

### EXPECTED BEHAVIOUR

The above list gives the expected results for these expressions

### SCREENSHOTS

N/A

### ADDITIONAL CONTEXT

I have a fix ready for this and will raise a PR shortly

MichaelDvP, 2025-03-07:
Nested works with some rules, multiple is not supported.
The condition is always executed first, the parser searches ? and : before calculating a formula.
`a<b ? c<d ? 1 : e<f ? 2 : 3 : 4` works
`(a<b) ? (c<d) ? 1 : (e<f) ? 2 : 3 : 4` works also
but not with `?` or `:` within the bracket

also `(a<b ? 2 : 4) + 10` does not work
you have to write `a < b ? 2 + 10 : 4 + 10`

proddy, 2025-03-08:
@philwingfield out of interest, what was your use case for adding this? I could add it as a real life example to the docs

philwingfield, 2025-03-10:
I wanted to try and implement a simple PID or probably just PI controller. To modulate the flow temp of my boiler to keep the return temp below 50c for best efficiency. I found the old implementation slightly too limiting for this.

I will post it somewhere if I ever get it working nicely!

closed 2025-03-22 by proddy