emsesp/EMS-ESP32 #2454: Gateway E32 (Ethernet / WiFi)  lost ethernet connection

SamA699, 2025-03-08:
### DESCRIPTION

I own an Gateway E32 (Ethernet + WiFi / Gateway E32) incl. EMS Servicekabel. Since 06/23  there were no problemes in combination connected over the service port of my Buderus GB172 and includes in homeassistant.

For about 1 week now the Gateway looses his connection to my fritzbox 7590 once a day.

To solve the problem i had to rejack the ethernet cabel in the gateway. Then it worked well for one day.  On the next day -  the same problem.

Is there any solution?

I add my EMS-system-info-file below:

[emsesp_system_info.txt](https://github.com/user-attachments/files/19143684/emsesp_system_info.txt)

proddy, 2025-03-08:
is EMS-ESP connected to your router via Ethernet or WiFi? Also please upgrade to the latest version 3.7.1, or even better 3.7.2. We're only helping out on the latest version and 3.6.5 is a year old.

SamA699, 2025-03-08:
Hello Prody, thanks for you quick response.
The EMS-ESP is connected to our router via Ethernet. I first made the update to 3.7.1. and for the above problem I downgrade to 3.6.5. ..... but the problem is in both versions.

proddy, 2025-03-08:
oh, that is strange. Is there anything in the Frtizbox logs showing the reconnect?

SamA699, 2025-03-08:
No, I just checked it twice ... there are no logs showing the reconnect in the Fritzbox systemlogs.

bbqkees, 2025-03-08:
As this is about the older E32 you should try to do a full erase of the memory via USB and flash it again with the 3.6.5 firmware.
https://bbqkees-electronics.nl/wiki/gateway/firmware-update-and-downgrade.html#uploading-the-firmware-via-ems-esp-flasher-flashtool
Uncheck 'Keep Settings' to to a full erase.

Before you start, you can download/export your device setting and customization from the EMS-ESP web interface.
After a successful flash you can upload these again.

For the older E32 with limited RAM memory it is advised to disable all services that are not used and turn off all entities you are not interested in to free up as much memory as possible.
https://docs.emsesp.org/Troubleshooting/#it-may-be-memory-related

If the above did not solve the issue reply to the order confirmation email from the E32 order or use the contact form on www.bbqkees.com to contact us and we will see how we can help.

SamA699, 2025-03-08:
Thank you very much for the info. I will try it and give feedback.

SamA699, 2025-03-13:
I did the full erase of the memory via USB and flashed it again with the 3.6.5-4mb firmware and turned off all entities I'm not interested. Since 1 day now all seems to work well.
Should I update to the 3.7.1-4mb firmware as well?

proddy, 2025-03-22:
I don't what is causing it. Version is 3.6.5 is a year old, so always take the latest. Also make sure the ssid is empty if you're using only Ethernet to avoid conflicts. And I see you have a static IP and have enabled ipv6, so maybe something is there. I would look into the Fritzbox logs and see why its loosing the connection. I don't think its an EMS-ESP issue but can't rule it out.

closed 2025-04-10 by proddy