emsesp/EMS-ESP32 #2457: String comparision in schedule seems to break with uppercase letters

Mortim-zz, 2025-03-09:
### DESCRIPTION

I wanted to create a schedule which checks for a specific error code. The string is `A21(816)`. 
The string comparison does not work for the schedule condition, according to the documentation I would have expected the string `thermostat/errorcode == "A21(816)"` to work. I tried some different values with escaping, quotes, lowercase etc. and sent the results as message, none matched. I tried other comparisons with a custom RAM entity, and the comparison does not work when there is an uppercase in the entity. Lowercase works fine.
```
thermostat/errorcode                  -> A21(816)
thermostat/errorcode == "A21(816)"    -> 0
-------------------
custom/testval                  -> A
custom/testval == "A"           -> 0
-------------------
custom/testval                  -> a
custom/testval == "a"           -> 1
-------------------
custom/testval                  -> a21(816)
custom/testval == "a21(816)"    -> 1
```

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [X] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [X] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [X] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [X] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
{"system":{"version":"3.7.1","uptime":"000+01:07:42.873","uptimeSec":4062,"platform":"ESP32","cpuType":"ESP32-D0WD-V3","arduino":"Tasmota Arduino v2.0.17","sdk":"4.4.8.240628","freeMem":160,"maxAlloc":103,"freeCaps":4181,"usedApp":1740,"freeApp":2932,"partition":"app1","resetReason":"Software reset CPU / Software reset CPU","psram":false,"model":""},"network":{"network":"Ethernet","hostname":"ems-esp","TxPowerSetting":0,"staticIP":false,"lowBandwidth":false,"disableSleep":true,"enableMDNS":true,"enableCORS":false,"APProvisionMode":"disconnected","APSecurity":"wpa2","APSSID":"ems-esp"},"ntp":{"NTPStatus":"connected","enabled":true,"server":"192.168.178.1","tzLabel":"Europe/Amsterdam"},"mqtt":{"MQTTStatus":"connected","MQTTPublishes":1173,"MQTTQueued":0,"MQTTPublishFails":0,"MQTTReconnects":0,"enabled":true,"clientID":"esp32-f022c930","keepAlive":60,"cleanSession":false,"entityFormat":1,"base":"ems-esp","discoveryPrefix":"homeassistant","discoveryType":0,"nestedFormat":1,"haEnabled":true,"mqttQos":0,"mqttRetain":false,"publishTimeHeartbeat":60,"publishTimeBoiler":10,"publishTimeThermostat":10,"publishTimeSolar":10,"publishTimeMixer":10,"publishTimeWater":10,"publishTimeOther":60,"publishTimeSensor":10,"publishSingle":false,"publish2command":false,"sendResponse":false},"syslog":{"enabled":false},"sensor":{"temperatureSensors":0,"temperatureSensorReads":0,"temperatureSensorFails":0,"analogSensors":0,"analogSensorReads":0,"analogSensorFails":0},"api":{"APICalls":0,"APIFails":0},"bus":{"busStatus":"connected","busProtocol":"Buderus","busTelegramsReceived":3879,"busReads":975,"busWrites":0,"busIncompleteTelegrams":0,"busReadsFailed":0,"busWritesFailed":0,"busRxLineQuality":100,"busTxLineQuality":100},"settings":{"boardProfile":"E32V2","locale":"en","txMode":1,"emsBusID":11,"showerTimer":false,"showerMinDuration":180,"showerAlert":false,"hideLed":false,"noTokenApi":false,"readonlyMode":false,"fahrenheit":false,"dallasParasite":false,"boolFormat":1,"boolDashboard":1,"enumFormat":1,"analogEnabled":true,"telnetEnabled":true,"maxWebLogBuffer":25,"webLogBuffer":25,"modbusEnabled":false,"forceHeatingOff":false,"developerMode":false},"devices":[{"type":"boiler","name":"EASY CONNECT/Ecomline Excellent","deviceID":"0x08","productID":206,"brand":"Nefit","version":"02.01","entities":26,"handlersReceived":"0x10 0x11 0x1C 0x18 0x19 0x34 0x04","handlersFetched":"0x14 0x16 0x33","handlersPending":"0xBF 0xC2 0x15 0x1A 0x35 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x28 0x2E 0x3B"},{"type":"thermostat","name":"RC35","deviceID":"0x10","productID":86,"brand":"","version":"01.08","entities":61,"handlersReceived":"0xA3 0x06 0xA2 0x12 0x13 0x42 0x48 0x47 0x52 0x51 0x5C 0x5B 0x38 0x39","handlersFetched":"0x3E 0x3D 0x3F 0xA5 0x37","handlersPending":"0x49 0x4C 0x53 0x56 0x5D 0x60"},{"type":"thermostat","name":"ES72, RC20","deviceID":"0x18","productID":113,"brand":"","version":"02.01","entities":1,"handlersReceived":"0xAF","handlersPending":"0xA3 0x06 0xA2 0x12 0x13"},{"type":"controller","name":"EASY CONNECT/Ecomline","deviceID":"0x09","productID":206,"brand":"Nefit","version":"02.01","entities":0,"handlersIgnored":"0x29"},{"type":"scheduler","name":"scheduler","entities":5},{"type":"custom","name":"custom","entities":2}]}

```

### TO REPRODUCE

Create a schedule which should fire on a string comparison of an entity with an uppercase A in the entity value. A custom entity in RAM is sufficient.
The schedule is not executed when the entity matches.
It seems the entity values are not converted to lowercase contrary to the docs.

### EXPECTED BEHAVIOUR

The schedule should be executed when the compared strings match.

proddy, 2025-03-10:
Thanks. We'll add some extra test cases and verify this in 3.7.2.

MichaelDvP, 2025-03-10:
Yes, we convert the condition string to lowercase, but forget to convert the api responses of embedded commands. Fix in https://github.com/MichaelDvP/EMS-ESP32/commit/d769999f1074a0369bd75fd00f7443121c9f2bbc

Mortim-zz, 2025-03-10:
Thanks for looking into this so fast!
Is there a reason all strings are converted to lowercase first and not only for string comparisons? 
I now realized this also influences raw values one might want to put in messages, ram values etc. It looks like everything coming from a schedule will be lowercase.
This might break things if there is an auth token or password (e.g. as in https://github.com/emsesp/EMS-ESP32/issues/1806#issuecomment-2261308537 ) with uppercase in there, if I'm not overlooking something.

MichaelDvP, 2025-03-10:
Yes, you are right, the tokens and passwords will fail with capitals. 
The conditions were introduced as simple compare with commands. All commands are always lower case, so it was simple to convert all to lowercase.

MichaelDvP, 2025-03-10:
@Mortim-zz Please try this build https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest
I changed to use lower case only for commands and compares, 
E.g. for the error code `Thermostat/errorCode == A21(816)` and `thermostat/errocode == a21(816)` should both work.
But the api is read mixed case, E.g. command: `custom/testval`: value `thermostat/errocode` will set testval to `A21(816)`

Please check. My schedules are all working, including the pushover http command, but i have not checked all possible combinations.

Mortim-zz, 2025-03-10:
I tried this version and it mostly works, thank you.
Only strings with uppercase letters without quotes seem break something now, they create an error.
If I execute an immediate schedule like below, I just get an error in the log:
```
2025-03-10 21:36:23.109 W 24: [command] Command message failed (Error)
2025-03-10 21:36:23.109 W 25: [emsesp] Schedule : Command message failed (Error)
```

[image]

MichaelDvP, 2025-03-11:
Thanks for testing, the mixed case strings should work now in my dev27.

Mortim-zz, 2025-03-11:
@MichaelDvP Yes, that works now, great!
I found a few more cases where something goes wrong, sorry:
- If there is a forward slash anywhere in the condition and this is not part of a command this gives an error. Also when the slash is in the result of commands in the condition.
- Parentheses in the result of a command get removed
- Commands inside quoted strings are still resolved to their result (I think this is not supposed to happen)

While playing around with different strings in schedules it also rebooted once and deleted all schedules, but I can't reproduce this.

Mortim-zz, 2025-03-11:
Btw, is there a way to compare to a value that is not always there? In my case I want to compare to the error code of the thermostat which is only there during the error condition. It functions fine, but the log is flooded with error messages every 10 seconds as long as there is no error.

proddy, 2025-03-11:
> Btw, is there a way to compare to a value that is not always there? In my case I want to compare to the error code of the thermostat which is only there during the error condition. It functions fine, but the log is flooded with error messages every 10 seconds as long as there is no error.

we could check the return code of the call to `Command::process()` in `ShuntingYard::commands()` and false the outcome of the command?

proddy, 2025-03-12:
Thanks. I've added some tests to the standalone framework so we can test without building & uploading the ESP firmware. It helps troubleshoot.

MichaelDvP, 2025-03-12:
While testing some strings in scheduler i've found some issues. 
`Test 1+2+3` works, but a string after the formula like `Test (1+2+3) times` does not work.
This is an issue in shuntingyard when having more parameters than operations. I think we should add a string addition. Then it should be `Test+(1+2+3)+times` to calculate to result: `Test6times`, that's easier to handle in the shuntingYard.

closed 2025-03-15 by proddy