emsesp/EMS-ESP32 #2497: GUI - Status - wrong Download Links

gh47110815, 2025-03-28:
### DESCRIPTION

Running ems-esp 3.7.2 and visiting the web page -> "Status" the UI offers me two Versions Develop and Stable together with a Download Link for each.

Pressing the "Download" link for the stable release (3.7.2): ok - >downloads stable release image 3.7.2 (EMS-ESP-3_7_2-ESP32-16MB.bin)
Pressing the  "Download" link for the dev release (3.7.3-dev3): ISSUE -> also downloads the stable release image 3.7.2 (EMS-ESP-3_7_2-ESP32-16MB.bin, https://github.com/emsesp/EMS-ESP32/releases/download/v3.7.2/EMS-ESP-3_7_2-ESP32-16MB.bin
### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
System:
 Version: 3.7.2
 Platform: ESP32 (ESP32-D0WD-V3)
 Model:
 Partition: app0
 Language: de
 Board profile: E32V2
 Uptime: 001+06:40:05.147
 Base MAC Address: xx:xx:xx:xx:xx:xx
 SDK version: 4.4.8.240628
 CPU frequency: 240 MHz
 Free heap/Max alloc: 165 KB / 107 KB
 App used/free: 1650 KB / 4046 KB
 FS used/free: 36 KB / 2012 KB
 Flash size: 16384 KB
 PSRAM: not available

```

### TO REPRODUCE
Visit 
http://<esp>/status/version
if both stable and dev releases are available for download.
Download both types and check, if dev-release is really the dev-binary image
### EXPECTED BEHAVIOUR
Downloads different images

_A clear and concise description of what you expected to happen._

### SCREENSHOTS

_If applicable, add screenshots to help explain your issue._

### ADDITIONAL CONTEXT

_Add any other context about the issue here._

**(Please remember to close the issue when it has been addressed)**

proddy, 2025-03-28:
oh, that's not good! thanks for reporting, I'll take a look

proddy, 2025-03-29:
All good ideas, thanks. I've thought a lot about this and it's tricky. I assumed if someone is on the dev branch they wouldn't want to downgrade to stable as we don't have this coded yet in the back-end code. Maybe as @gh47110815 pointed out we're making this too complicated. Let me think about it...

MichaelDvP, 2025-03-30:
> I assumed if someone is on the dev branch they wouldn't want to downgrade to stable as we don't have this coded yet in the back-end code. 

Ok for downgrades, but the upgrade to next stable is also not provided. Assume someone uses dev (e.g. 3.7.2-dev.xx) because of a fixed bug and after release of stable 3.7.2 he has only the button for 3.7.3-dev.0, but no upgrade button for new stable.

proddy, 2025-03-30:
I'm working on the changes now

closed 2025-03-31 by proddy

gh47110815, 2025-04-04:
Hi, just for your info: At my side, Download-Dev-Images still does not work directly from the /status/version page.
What I did.....after you tagged this issue as fixed, I installed latest Dev-Release 5. At this time, it was the latest dev-release, so the download button for a newer dev-release did not occur. Today I realized, that there seems to be a new dev-6. But, the download button for this dev-6 (currently running dev-5) still used the link to the stable release 3.7.2 (in my case: https://github.com/emsesp/EMS-ESP32/releases/download/v3.7.2/EMS-ESP-3_7_2-ESP32-16MB.bin). Tested with  safari on a mac WITHOUT clearing the cache AND Firefox on a mac WITH clearing the cache. Maybe a problem on my side, if it's working on your side ;-) ??????

gh47110815, 2025-04-21:
Nice: I was running 3.7.3-dev_7 ... today dev_8 was shown as available, and ..... Download of exactly this dev_8 worked as expected. Thanks a lot !

commit 0f224460 (2025-03-31): fixes #2497 - showFirmwareDialog() was missing argument