emsesp/EMS-ESP32 #2500: Buderus WLW 176i + FS/2 FS20/2 can't change some dhw values

lex777777, 2025-03-29:
### DESCRIPTION

Can't set dhw comfort stop temp below 64°C or change dhw comfort mode

### REQUESTED INFORMATION

- [X ] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [X ] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ X] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [X ] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
  {"system":{"version":"3.7.3-dev.3","uptime":"000+00:13:12.677","uptimeSec":792,"platform":"ESP32","cpuType":"ESP32-D0WD-V3","arduino":"ESP32 Arduino v2.0.17","sdk":"v4.4.7-dirty","freeMem":115,"maxAlloc":53,"freeCaps":4133,"usedApp":2243,"freeApp":2429,"partition":"app1","flash_chip_size":16384,"resetReason":"Software reset CPU / Software reset CPU","psram":true,"psramSize":4093,"freePsram":4052,"model":""},"network":{"network":"WiFi","hostname":"ems-esp","RSSI":-58,"WIFIReconnects":0,"TxPowerSetting":0,"staticIP":false,"lowBandwidth":false,"disableSleep":true,"enableMDNS":true,"enableCORS":false,"APProvisionMode":"disconnected","APSecurity":"wpa2","APSSID":"ems-esp"},"ntp":{"NTPStatus":"disconnected","enabled":false,"server":"time.google.com","tzLabel":"Europe/Amsterdam"},"mqtt":{"MQTTStatus":"disconnected","enabled":false,"clientID":"esp32-62bbbc2c","keepAlive":60,"cleanSession":false,"entityFormat":0,"base":"ems-esp","discoveryPrefix":"homeassistant","discoveryType":0,"nestedFormat":1,"haEnabled":false,"mqttQos":0,"mqttRetain":false,"publishTimeHeartbeat":60,"publishTimeBoiler":10,"publishTimeThermostat":10,"publishTimeSolar":10,"publishTimeMixer":10,"publishTimeWater":10,"publishTimeOther":60,"publishTimeSensor":10,"publishSingle":false,"publish2command":false,"sendResponse":false},"syslog":{"enabled":false},"sensor":{"temperatureSensors":0,"temperatureSensorReads":0,"temperatureSensorFails":0,"analogSensors":0,"analogSensorReads":0,"analogSensorFails":0},"api":{"APICalls":0,"APIFails":0},"bus":{"busStatus":"connected","busProtocol":"Buderus","busTelegramsReceived":3917,"busReads":972,"busWrites":17,"busIncompleteTelegrams":0,"busReadsFailed":0,"busWritesFailed":0,"busRxLineQuality":100,"busTxLineQuality":100},"settings":{"boardProfile":"E32V2","locale":"de","txMode":2,"emsBusID":11,"showerTimer":false,"showerMinDuration":180,"showerAlert":false,"hideLed":false,"noTokenApi":true,"readonlyMode":false,"fahrenheit":false,"dallasParasite":false,"boolFormat":1,"boolDashboard":1,"enumFormat":1,"analogEnabled":true,"telnetEnabled":true,"maxWebLogBuffer":50,"modbusEnabled":false,"forceHeatingOff":false,"developerMode":false},"devices":[{"type":"boiler","name":"XCU_THH/CS*800i, Logatherm WLW*","deviceID":"0x08","productID":8,"brand":"Buderus","version":"00.00","entities":164,"handlersReceived":"0xBF 0xD1 0xE3 0xE4 0xE5 0xE9 0x0494 0x0495 0x048F","handlersFetched":"0x14 0xE6 0xEA 0x048D 0x04A2 0x0485 0x0486 0x0492 0x0488 0x0484 0x048B 0x0491 0x0499 0x049C 0x049D 0x04AE 0x04AF 0x02CC 0x02D6","handlersPending":"0x10 0x11 0xC2 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0x28 0x048A 0x04A5 0x04AA 0x04A7","handlersIgnored":"0xE7 0x02E0 0x02EA 0x061E 0x0240 0x0291 0x0292 0x0293 0x0294 0x029B 0x02A5 0x04A8 0x04AC 0x02F5 0x04CA 0x04CB 0x0497 0xF9 0x04A0"},{"type":"thermostat","name":"HMI800.2/Rego 3000, UI800, Logamatic BC400","deviceID":"0x10","productID":253,"brand":"","version":"47.07","entities":76,"handlersReceived":"0x06 0x02BA 0x02BB 0x02BC 0x031D 0x0267","handlersFetched":"0x02A5 0x02B9 0x02AF 0x029B 0x0471 0x02CC 0x0467 0x0291 0x0292 0x0293 0x0294 0x02F5 0x023A 0x0240 0xBB 0x023E 0x016E","handlersPending":"0xA3 0xA2 0x12 0x13 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x02F6 0x031B 0x031E 0x0269","handlersIgnored":"0x0507 0x0508 0x0509 0x050A 0x059D 0xF7 0xBF"},{"type":"gateway","name":"MX300/WiFi module","deviceID":"0x48","productID":252,"brand":"Buderus","version":"11.01","entities":0,"handlersIgnored":"0x2040 0xF9"},{"type":"water","name":"MS100/SM100, MS100","deviceID":"0x2A","productID":163,"brand":"","version":"27.09","entities":18,"handlersReceived":"0x07D6 0x07AA 0x07AD","handlersFetched":"0x07A5 0x07A6 0x07AE 0x07E0","handlersPending":"0x07AB 0x07AC","handlersIgnored":"0x07A9 0x07B1 0xBF"},{"type":"custom","name":"custom","entities":1}]}

```

### TO REPRODUCE

Try to set dhw comfort stop temp to 60°C:
```000+00:17:58.352 I 4967: [command] Called command boiler/comfstop (comfort stop temp) with value 60 and id 9 on device 0x08
000+00:17:58.381 T 4968: [emsesp] me(0x0B) W boiler(0x08), HPDhwSettings(0x0499), data: 3C (offset 8)
000+00:17:58.500 T 4969: [emsesp] boiler(0x08) B all(0x00), UBAMonitorWWPlus(0xE9), data: 3C (offset 23)
000+00:17:58.672 T 4970: [emsesp] boiler(0x08) B all(0x00), ?(0x0497), data: 02 58 (offset 6)
000+00:17:58.843 T 4971: [emsesp] boiler(0x08) B all(0x00), HPDhwSettings(0x0499), data: 3C (offset 8)
000+00:17:59.069 T 4972: [emsesp] boiler(0x08) B all(0x00), ValueConfig(0xF9), data: EA 00 00 06 07 00 00 00 28 00 00 00 35 00 00 00 37 00 00 00 28
000+00:17:59.298 T 4973: [emsesp] me(0x0B) R boiler(0x08), HPDhwSettings(0x0499), length: 1 (offset 8)
000+00:17:59.312 T 4974: [emsesp] boiler(0x08) W me(0x0B), HPDhwSettings(0x0499), data: 3C (offset 8)
000+00:17:59.802 T 4975: [emsesp] thermostat(0x10) W boiler(0x08), HPDhwSettings(0x0499), data: 40 (offset 8)
000+00:17:59.868 T 4976: [emsesp] boiler(0x08) B all(0x00), UBAMonitorWWPlus(0xE9), data: 40 (offset 23)
000+00:18:00.041 T 4977: [emsesp] boiler(0x08) B all(0x00), ?(0x0497), data: 02 80 (offset 6)
000+00:18:00.212 T 4978: [emsesp] boiler(0x08) B all(0x00), HPDhwSettings(0x0499), data: 40 (offset 8)
000+00:18:00.446 T 4979: [emsesp] boiler(0x08) B all(0x00), ValueConfig(0xF9), data: EA 00 00 06 07 00 00 00 28 00 00 00 35 00 00 00 3B 00 00 00 28
000+00:18:00.716 T 4980: [emsesp] me(0x0B) R boiler(0x08), UBATotalUptime(0x14), length: 27
000+00:18:00.726 T 4981: [emsesp] boiler(0x08) W me(0x0B), UBATotalUptime(0x14), data: 02 22 DB
000+00:18:00.867 T 4982: [emsesp] me(0x0B) R boiler(0x08), UBAParametersPlus(0xE6), length: 27
000+00:18:00.867 T 4983: [emsesp] boiler(0x08) W me(0x0B), UBAParametersPlus(0xE6), data: 01 1E 00 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00 02 00 00 00 01
000+00:18:01.151 T 4984: [emsesp] me(0x0B) R boiler(0x08), UBAParameterWWPlus(0xEA), length: 27
000+00:18:01.174 T 4985: [emsesp] boiler(0x08) W me(0x0B), UBAParameterWWPlus(0xEA), data: 01 28 00 00 00 00 03 46 00 00 18 32 3C 25 (offset 5)
000+00:18:01.174 T 4986: [emsesp] me(0x0B) R boiler(0x08), UBAParameterWWPlus(0xEA), length: 27 (offset 19)
000+00:18:01.198 T 4987: [emsesp] boiler(0x08) W me(0x0B), UBAParameterWWPlus(0xEA), data: 26 (offset 27)
000+00:18:01.198 T 4988: [emsesp] me(0x0B) R boiler(0x08), UBAParameterWWPlus(0xEA), length: 27 (offset 28)
000+00:18:01.220 T 4989: [emsesp] boiler(0x08) W me(0x0B), UBAParameterWWPlus(0xEA), data: <empty> (offset 28)
000+00:18:01.390 T 4990: [emsesp] me(0x0B) R boiler(0x08), HpPower(0x048D), length: 25
```

Change dhw comfort mode to eco:

``` 000+00:22:48.486 I 6057: [command] Called command boiler/comfort1 (comfort mode) with value eco and id 9 on device 0x08
000+00:22:48.665 T 6058: [emsesp] me(0x0B) W boiler(0x08), ?(0x05), data: AA (offset 70)
000+00:22:48.878 T 6059: [emsesp] me(0x0B) W boiler(0x08), UBAParameterWWPlus(0xEA), data: D8 (offset 13)
000+00:22:49.050 T 6060: [emsesp] me(0x0B) R boiler(0x08), UBAParameterWWPlus(0xEA), length: 1 (offset 13)
000+00:22:49.057 T 6061: [emsesp] boiler(0x08) W me(0x0B), UBAParameterWWPlus(0xEA), data: <empty> (offset 13)
000+00:22:52.565 T 6062: [emsesp] boiler(0x08) R thermostat(0x10), RC300Settings(0x0240), length: 1 (offset 10)
000+00:22:52.588 T 6063: [emsesp] thermostat(0x10) W boiler(0x08), RC300Settings(0x0240), data: F8 (offset 10)
000+00:22:52.599 T 6064: [emsesp] boiler(0x08) R thermostat(0x10), RC300Settings(0x0240), length: 1 (offset 12)
000+00:22:52.622 T 6065: [emsesp] thermostat(0x10) W boiler(0x08), RC300Settings(0x0240), data: FF (offset 12)
000+00:22:52.633 T 6066: [emsesp] boiler(0x08) R thermostat(0x10), HPMode(0x0291), length: 1
[... cut ...]
000+00:22:53.778 T 6080: [emsesp] thermostat(0x10) W boiler(0x08), HPMode(0x0293), data: 00
000+00:22:53.789 T 6081: [emsesp] boiler(0x08) R thermostat(0x10), HPMode(0x0293), length: 1 (offset 5)
000+00:22:53.812 T 6082: [emsesp] thermostat(0x10) W boiler(0x08), HPMode(0x0293), data: 00 (offset 5)
```

### EXPECTED BEHAVIOUR

Values should be changed

### SCREENSHOTS

[image]

### ADDITIONAL CONTEXT

_Add any other context about the issue here._

**(Please remember to close the issue when it has been addressed)**

proddy, 2025-03-29:
Can you do the exact same commands via the physical thermostat and compare the EMS-ESP logs?

lex777777, 2025-03-29:
Can't find the setpoints at the BC400 (integrated in WLW 176).
Maybe it is only possible with the Buderus Logamatic SC300 for FS/2 FS20/2?

MichaelDvP, 2025-03-29:
Emsesp sends the correct data:
000+00:17:58.381 T 4968: [emsesp] me(0x0B) W boiler(0x08), HPDhwSettings(0x0499), data: 3C (offset 8)
Boiler confirm the setting
000+00:17:58.843 T 4971: [emsesp] boiler(0x08) B all(0x00), HPDhwSettings(0x0499), data: 3C (offset 8)
emsesp readsback
000+00:17:59.298 T 4973: [emsesp] me(0x0B) R boiler(0x08), HPDhwSettings(0x0499), length: 1 (offset 8)
boiler answers 60°C
000+00:17:59.312 T 4974: [emsesp] boiler(0x08) W me(0x0B), HPDhwSettings(0x0499), data: 3C (offset 8)
Thermostat overwrites the value with 64°C 
000+00:17:59.802 T 4975: [emsesp] thermostat(0x10) W boiler(0x08), HPDhwSettings(0x0499), data: 40 (offset 8)
Boiler confirm the new value
000+00:18:00.212 T 4978: [emsesp] boiler(0x08) B all(0x00), HPDhwSettings(0x0499), data: 40 (offset 8)

You cannot change boiler values if they are controlled by the thermostat!

lex777777, 2025-03-30:
OK I understood it for the dhw comfort stop temp value, but what about the dhw comfort mode?

MichaelDvP, 2025-03-30:
Comfort mode is for some boiers not changeable. Sometimes there is a dedicated button on the panel to switch it:
This isjust a command without any feedback, Seems not to work here:
000+00:22:48.665 T 6058: [emsesp] me(0x0B) W boiler(0x08), ?(0x05), data: AA (offset 70)

Other boilers have a mode in parameters, setting here
000+00:22:48.878 T 6059: [emsesp] me(0x0B) W boiler(0x08), UBAParameterWWPlus(0xEA), data: D8 (offset 13)
But readback:
000+00:22:49.050 T 6060: [emsesp] me(0x0B) R boiler(0x08), UBAParameterWWPlus(0xEA), length: 1 (offset 13)
000+00:22:49.057 T 6061: [emsesp] boiler(0x08) W me(0x0B), UBAParameterWWPlus(0xEA), data: <empty> (offset 13)
The boiler answers with empty, so mode is not supportet.

If you can change the mode on the boiler, do this and log the telegrams, maybe there is another way. But i think the HMI800 thermostat provides some dhw modes, so any boiler setting will be overwritten.

lex777777, 2025-03-30:
Thank you for the explanations.
I will try to find the settings on the device itself and log them.

Maybe I am approaching the problem the wrong way. My aim is to change the temperature (_dhw set temperature_) of the hot water tank. So far, however, I have not been able to influence the value with anything except increase _dhw set temperature_.

```001+00:52:15.688 I 336134: [command] Called command boiler/comfstop (comfort stop temp) with value 65 and id 9 on device 0x08
001+00:52:15.688 T 336135: [emsesp] me(0x0B) R thermostat(0x10), PVSettings(0x023E), length: 25
001+00:52:15.750 T 336136: [emsesp] thermostat(0x10) W me(0x0B), PVSettings(0x023E), data: 02 02 00 00 00 00 00
001+00:52:15.923 T 336137: [emsesp] me(0x0B) W boiler(0x08), HPDhwSettings(0x0499), data: 41 (offset 8)
001+00:52:16.051 T 336138: [emsesp] boiler(0x08) B all(0x00), UBAMonitorWWPlus(0xE9), data: 41 (offset 23)
001+00:52:16.223 T 336139: [emsesp] boiler(0x08) B all(0x00), ?(0x0497), data: 02 8A (offset 6)
001+00:52:16.394 T 336140: [emsesp] boiler(0x08) B all(0x00), HPDhwSettings(0x0499), data: 41 (offset 8)
001+00:52:16.629 T 336141: [emsesp] boiler(0x08) B all(0x00), ValueConfig(0xF9), data: EA 00 00 06 07 00 00 00 28 00 00 00 35 00 00 00 3C 00 00 00 28
001+00:52:16.815 T 336142: [emsesp] me(0x0B) R boiler(0x08), HPDhwSettings(0x0499), length: 1 (offset 8)
001+00:52:16.826 T 336143: [emsesp] boiler(0x08) W me(0x0B), HPDhwSettings(0x0499), data: 41 (offset 8)
001+00:52:17.042 T 336144: [emsesp] me(0x0B) R thermostat(0x10), Absent(0x016E), length: 25
001+00:52:17.065 T 336145: [emsesp] thermostat(0x10) W me(0x0B), Absent(0x016E), data: 00
001+00:52:17.234 T 336146: [emsesp] me(0x0B) R water(0x2A), SM100wwCirc(0x07A5), length: 25
001+00:52:17.274 T 336147: [emsesp] water(0x2A) W me(0x0B), SM100wwCirc(0x07A5), data: 00 03 0A 04 00 1E 32
001+00:52:17.460 T 336148: [emsesp] me(0x0B) R water(0x2A), SM100wwParam(0x07A6), length: 25
001+00:52:17.547 T 336149: [emsesp] water(0x2A) W me(0x0B), SM100wwParam(0x07A6), data: FF 05 0F 5F 19 01 3C 32 3C 2D 28 12 41 00 3C 1E 03 07 3C 00 0A 00 05
001+00:52:17.757 T 336150: [emsesp] me(0x0B) R water(0x2A), SM100wwKeepWarm(0x07AE), length: 25
001+00:52:17.759 T 336151: [emsesp] water(0x2A) W me(0x0B), SM100wwKeepWarm(0x07AE), data: 00 0F 0F
001+00:52:17.956 T 336152: [emsesp] me(0x0B) R water(0x2A), SM100wwStatus2(0x07E0), length: 25
001+00:52:18.034 T 336153: [emsesp] water(0x2A) W me(0x0B), SM100wwStatus2(0x07E0), data: 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 03 03 03
001+00:52:18.216 T 336154: [emsesp] me(0x0B) R thermostat(0x10), RC300WWtemp(0x031B), length: 1
001+00:52:18.236 T 336155: [emsesp] thermostat(0x10) W me(0x0B), RC300WWtemp(0x031B), data: <empty>
001+00:52:22.475 T 336156: [emsesp] boiler(0x08) B all(0x00), UBAMonitorSlowPlus2(0xE3), data: 01 00 01 00 00 00 00 00 00 00 00 01 00 00 00 00 00 17
001+00:52:22.665 T 336157: [emsesp] boiler(0x08) B all(0x00), UBAMonitorFastPlus(0xE4), data: 17 01 00 00 1A 03 00 00 00 00 00 00 DA 00 00 19 00 01 02 (offset 6)
001+00:52:22.836 T 336158: [emsesp] boiler(0x08) B all(0x00), UBAMonitorFastPlus(0xE4), data: 01 72 (offset 36)
001+00:52:23.073 T 336159: [emsesp] boiler(0x08) B all(0x00), UBAMonitorWWPlus(0xE9), data: 41 02 5A 01 B8 00 00 00 00 46 3C 00 00 00 00 00 00 00 00 00 00 00 00 41
```

[image]

[image]

lex777777, 2025-04-01:
Hello again,

I try to change the hw comfort mode but it seems that it only affected the dhw operating mode:

[log_try_change_dhw_mode.txt](https://github.com/user-attachments/files/19546978/log_try_change_dhw_mode.txt)

Then I change the water system from fresh water station to heat pump ([log_change_water_system.txt](https://github.com/user-attachments/files/19547049/log_change_water_system.txt)).
Now I can change the dhw comfort stop temp to 50°C (Line 113).
When changing back to fresh water station it changed automatically back to 64°C (Line 467):

MichaelDvP, 2025-04-16:
If youre dhw is managed by the water station (SM100), then, i guess, you have to change parameters on this station and not on the boiler.

closed 2025-05-22 by github-actions[bot]