emsesp/EMS-ESP32 #2501: last error code not showing Solar Errors?

Madaraxa, 2025-03-30:
I just noticed, that my heating system has and had errors that were not shown in the boiler -> last error code.
Is this inteded and only "internal" errors are shown?
My Solar Module had several A51/6028 errors (air in system or pump blocked) and has an active A11/6004 (No communication to Solar Module) error.
The Solar Module was not present in EMS-ESP (as there was a communication error). I unplugged the EMS-ESP, restarted the solar Module (took power away and replugged it) and the error was gone. Then i replugged the EMS-ESP and it found the solar module.

Madaraxa, 2025-03-31:
I installed dev 3.7.3 dev4 now (Was dev3 when i posted the issue).
Ems-esp restarted normally and solar values are still present. The error had been resolved (See above) but last error code still shows error from 2021 and Not the 'last' one from 2 says ago

MichaelDvP, 2025-03-31:
`boiler/lascode` shows only the blocking and locking errors stored in boiler, communication errors are stored in thermostat, see `thermostat/lascode`

Madaraxa, 2025-03-31:
hmm.
I checked every Device and i got last error code only in boiler. Under thermostat there is nothing that shows errors :(
I would post a list but dont know how beside posting 3 screenshots :)

[image]

MichaelDvP, 2025-03-31:
Seems the CW400 stores the error log somewhere else, the telegrams 012, 0x13 as known error logs are listed as pending in your system info. Try with `read 10 12` and `read 10 13`, but i think both will respond with `<empty>`

Madaraxa, 2025-03-31:
yeah, sadly they do:
`ems-esp:$ read 10 12
000+01:57:08.089 N 1: [emsesp] thermostat(0x10) W me(0x0B), RCErrorMessage(0x12), data: <empty>
ems-esp:$ read 10 13
000+01:57:12.944 N 2: [emsesp] thermostat(0x10) W me(0x0B), RCErrorMessage2(0x13), data: <empty>
`

VlastiBroucek, 2025-04-01:
@MichaelDvP, seems this applies to other thermostats too. Any chance fixing it? For example my CR50 is using 0x0C0 and 0x0C1. BTW, these are described in the EMS library now. 
@Madaraxa can you try those?

MichaelDvP, 2025-04-01:
I've never seen these telegrams in a log, can someone give me the output from `read 10 c0`, `read 10 c1`, maybe also the following c2 until c9.  Some reported also c7 telegram as error log, but i think only for heatpumps, not for thermostats.

VlastiBroucek, 2025-04-01:
Description: https://docs.emsesp.org/ems-library/ and look for it. 

Example

000+12:44:43.081 N 1: [emsesp] thermostat(0x18) W me(0x0B), ?(0xC0), data: 08 5F 00 12 05 48 30 38 03 FA 99 03 00 01 00 99 03 10 1A 2C 08 5F 00 12 05 48 30
000+12:44:43.185 N 2: [emsesp] thermostat(0x18) W me(0x0B), ?(0xC0), data: 38 03 FA 99 02 12 11 00 99 02 12 11 02 08 5F 00 12 05 48 30 38 03 FA 99 02 10 11 (offset 27)
000+12:44:43.292 N 3: [emsesp] thermostat(0x18) W me(0x0B), ?(0xC0), data: 0B 99 02 10 11 11 08 5F 00 0A 05 20 43 36 00 14 98 0A 00 08 34 98 0A 00 08 3A 08 (offset 54)
000+12:44:43.400 N 4: [emsesp] thermostat(0x18) W me(0x0B), ?(0xC0), data: 5F 00 0A 05 20 42 31 00 1F 95 02 03 01 1F 95 02 03 01 30 08 5F 00 0A 05 20 42 31 (offset 81)
000+12:44:43.416 N 5: [emsesp] thermostat(0x18) W me(0x0B), ?(0xC0), data: 00 1F 95 02 03 01 0F 95 02 03 01 19 08 5F 00 0A 05 20 42 31 00 1F 95 02 03 01 0D (offset 108)
000+12:44:43.416 N 6: [emsesp] thermostat(0x18) W me(0x0B), ?(0xC0), data: 95 02 03 01 0E 08 5F 00 0A 05 20 42 31 00 20 95 02 02 01 37 95 02 03 01 0D 08 5F (offset 135)
ems-esp:$

Madaraxa, 2025-04-01:
`ems-esp:$ read 10 c0
001+00:34:11.922 N 9: [emsesp] thermostat(0x10) W me(0x0B), ?(0xC0), data: 10 9E 00 0E 11 41 31 31 17 74 99 03 15 13 0C 99 03 09 1E 27 30 A4 00 0E 11 41 35
001+00:34:12.049 N 10: [emsesp] thermostat(0x10) W me(0x0B), ?(0xC0), data: 31 17 8C 99 03 0C 13 00 99 03 0C 13 18 30 A4 00 0E 11 41 35 31 17 8C 99 03 0C 12 (offset 27)
001+00:34:12.167 N 11: [emsesp] thermostat(0x10) W me(0x0B), ?(0xC0), data: 16 99 03 0D 12 03 30 A4 00 0E 11 41 35 31 17 8C 99 03 0D 11 23 99 03 0D 11 26 30 (offset 54)
001+00:34:13.338 N 12: [emsesp] thermostat(0x10) W me(0x0B), ?(0xC0), data: A4 00 0E 11 41 35 31 17 8C 99 03 0E 10 22 99 03 0E 10 25 30 A4 00 0E 11 41 35 31 (offset 81)
001+00:34:13.462 N 13: [emsesp] thermostat(0x10) W me(0x0B), ?(0xC0), data: 17 8C 99 03 0C 10 30 99 03 0D 10 02 28 A1 00 10 11 41 34 31 0F DD 99 03 11 0F 20 (offset 108)
001+00:34:13.595 N 14: [emsesp] thermostat(0x10) W me(0x0B), ?(0xC0), data: 99 03 11 0F 21 10 9E 00 0E 11 41 31 31 0F AB 99 03 11 0F 1F 99 03 11 0F 20 30 A4 (offset 135)
001+00:34:13.705 N 15: [emsesp] thermostat(0x10) W me(0x0B), ?(0xC0), data: 00 0E 11 41 35 31 17 8C 99 03 0C 0F 09 99 03 0C 0F 0A 30 A4 00 0E 11 41 35 31 17 (offset 162)
001+00:34:14.860 N 16: [emsesp] thermostat(0x10) W me(0x0B), ?(0xC0), data: 8C 99 03 0D 0E 0D 99 03 0D 0E 0F (offset 189)
ems-esp:$ read 10 c1
001+00:35:12.463 N 17: [emsesp] thermostat(0x10) W me(0x0B), ?(0xC1), data: 30 A4 00 0E 11 41 35 31 17 8C 99 03 0D 0A 17 99 03 0D 0A 1C 30 A4 00 0E 11 41 35
001+00:35:12.577 N 18: [emsesp] thermostat(0x10) W me(0x0B), ?(0xC1), data: 31 17 8C 99 03 0C 09 1B 99 03 0C 09 28 30 A4 00 0E 11 41 35 31 17 8C 99 03 0C 08 (offset 27)
001+00:35:13.734 N 19: [emsesp] thermostat(0x10) W me(0x0B), ?(0xC1), data: 16 99 03 0C 08 30 30 A4 00 0E 11 41 35 31 17 8C 99 03 0C 07 09 99 03 0E 07 0D 30 (offset 54)
001+00:35:13.848 N 20: [emsesp] thermostat(0x10) W me(0x0B), ?(0xC1), data: A4 00 0E 11 41 35 31 17 8C 99 03 0C 06 09 99 03 0C 06 15 30 A4 00 0E 11 41 35 31 (offset 81)
001+00:35:14.854 N 21: [emsesp] thermostat(0x10) W me(0x0B), ?(0xC1), data: 17 8C 99 03 0C 05 02 99 03 0C 05 08 30 A4 00 0E 11 41 35 31 17 8C 99 03 0D 04 2C (offset 108)
001+00:35:14.971 N 22: [emsesp] thermostat(0x10) W me(0x0B), ?(0xC1), data: 99 03 0D 04 2E 30 A4 00 0E 11 41 35 31 17 8C 99 03 0C 03 21 99 03 0C 03 2A 30 A4 (offset 135)
001+00:35:15.794 N 23: [emsesp] thermostat(0x10) W me(0x0B), ?(0xC1), data: 00 0E 11 41 35 31 17 8C 99 03 0C 02 2C 99 03 0D 02 22 30 A4 00 0E 11 41 35 31 17 (offset 162)
001+00:35:15.863 N 24: [emsesp] thermostat(0x10) W me(0x0B), ?(0xC1), data: 8C 99 02 0D 1B 04 99 02 0D 1B 0B (offset 189)
ems-esp:$ read 10 c2
001+00:35:22.142 N 25: [emsesp] thermostat(0x10) W me(0x0B), ?(0xC2), data: 10 9E 00 0E 11 41 31 31 17 74 99 03 15 13 0C 99 03 09 1E 27 10 9E 00 0E 11 41 31
001+00:35:22.261 N 26: [emsesp] thermostat(0x10) W me(0x0B), ?(0xC2), data: 31 0F AB 99 03 11 0F 1F 99 03 11 0F 20 10 9E 00 0E 11 41 31 31 0F AB 99 02 10 0F (offset 27)
001+00:35:24.266 N 27: [emsesp] thermostat(0x10) W me(0x0B), ?(0xC2), data: 30 99 02 10 0F 31 10 9E 00 0E 11 41 31 31 0F AB 99 02 16 08 33 99 02 16 08 34 10 (offset 54)
001+00:35:24.558 W 28: [telegram] Incomplete Rx: 90 0B C2 51 11
ems-esp:$ read 10 c3
001+00:35:30.980 N 29: [emsesp] thermostat(0x10) W me(0x0B), ?(0xC3), data: 00
ems-esp:$ read 10 c4
001+00:35:39.790 N 30: [emsesp] thermostat(0x10) W me(0x0B), ?(0xC4), data: 00
ems-esp:$ read 10 c5
001+00:35:44.122 N 31: [emsesp] thermostat(0x10) W me(0x0B), ?(0xC5), data: <empty>
ems-esp:$ read 10 c6
001+00:35:49.056 N 32: [emsesp] thermostat(0x10) W me(0x0B), ?(0xC6), data: <empty>
ems-esp:$ read 10 c7
001+00:35:51.566 N 33: [emsesp] thermostat(0x10) W me(0x0B), ?(0xC7), data: <empty>
ems-esp:$ read 10 c8
001+00:35:54.006 N 34: [emsesp] thermostat(0x10) W me(0x0B), ?(0xC8), data: <empty>
ems-esp:$ read 10 c9
001+00:35:57.448 N 35: [emsesp] thermostat(0x10) W me(0x0B), ?(0xC9), data: <empty>
`

MichaelDvP, 2025-04-01:
Oh, forgot to do a first read, do a `read 10 C0` and last code should appear in mqtt. I'll fix the build.

Madaraxa, 2025-04-01:
Its there now. Big Thanks. You are the Best. And this read 10 c0 should Not be needed in the future, right? 

{"lastcode":"A11(6004) 19.03.2025 21:12-30.03.2025 09:39","datetime":"01.04.2025 19:05","intoffset":0.0,"floordry":"off","dampedoutdoortemp":11.1,"floordrytemp":0,"building":"medium","minexttemp":-10,"damping":1,"solar":1,"hc2":{"seltemp":22.0,"currtemp":22.4,"mode":"auto","modetype":"comfort","ecotemp":15.0,"manualtemp":21.0,"comforttemp":21.0,"summertemp":17,"designtemp":45,"offsettemp":0,"minflowtemp":26,"maxflowtemp":50,"roominfluence":0,"roominflfactor":4.0,"curroominfl":0.0,"nofrostmode":"outdoor","nofrosttemp":5,"targetflowtemp":35,"heatingtype":"floor","summersetmode":"auto","summermode":"winter","vacationmode":0,"controlmode":"outside basepoint","program":"prog 1","tempautotemp":-1.0,"cooltemp":0.0,"fastheatup":0,"switchonoptimization":1,"reducemode":"outdoor","noreducetemp":-31,"reducetemp":0,"dhwprio":1,"coolingon":1,"hpmode":"heating","control":"RC310","remotetemp":null,"remotehum":null,"switchprogmode":"absolute","redthreshold":21.0,"solarinfl":0,"currsolarinfl":0.0},"dhw":{"mode":"own prog","settemp":60,"settemplow":45,"circmode":"own prog","chargeduration":30,"charge":0,"extra":0,"disinfecting":0,"disinfectday":"tu","disinfecttime":120,"dailyheating":0,"dailyheattime":120}}

VlastiBroucek, 2025-04-01:
> Interesting, first error in 0xC0 and0xC2 is exactly the same A11(6004) from 19.03.2025 (Communication solarmodule), In C1 the error is older A51(6028) from 10.03.2025 (solarpump blocked). 
> Please test with https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest, i've added the C0 telegram.
>  

@MichaelDvP I believe that C0 and C1 in my CR50 form one long list of global errors while C2 is list of errors specific to the thermostat only. @Sbried is of the same opinion 

I cannot test your latest build, I'm currently in hospital awaiting eye surgery.

Sbried, 2025-04-02:
In my setup (heat pump `CUHP`and thermostat `HMC310`) I get these readings:
### thermostat (0x10)
```
read 10 0xC0
08 AC 00 0C 09 41 30 31 14 B2 99 04 08 02 1A 99 04 08 02 1D; (1#) 2025/04/02 08:26 -> 2025/04/02 08:29
08 AC 00 0C 09 41 30 31 14 B2 99 03 07 1F 10 99 03 07 1F 13; (2#) 2025/03/31 07:10 -> 2025/03/31 07:19
08 AC 00 0C 09 41 30 31 14 B2 99 03 07 1B 0C 99 03 07 1B 0F; (3#) 2025/03/27 07:12 -> 2025/03/27 07:15
08 AC 00 0C 09 41 30 31 14 B2 99 03 08 1A 0C 99 03 08 1A 0F; (4#) 
08 AC 00 0C 09 41 30 31 14 B2 99 03 07 1A 10 99 03 07 1A 13; (5#)
08 AC 00 0C 09 41 30 31 14 B2 99 03 09 14 06 99 03 09 14 09; (6#) 
08 AC 00 0C 09 41 30 31 14 B2 99 03 07 14 0E 99 03 07 14 11; (7#) 
10 9E 00 0E 11 41 31 31 0C 13 99 03 0B 13 17 99 03 0B 13 24; (1*) 
10 9E 00 0E 11 41 31 31 0B FF 99 03 0B 13 16 99 03 0B 13 23; (2*)
10 9E 00 0E 11 41 31 31 0C 13 99 03 0A 13 3B 99 03 0B 13 05; (3*) 2025/03/19 10:59 -> 2025/03/19 11:05

read 10 0xC1
10 9E 00 0E 11 41 31 31 0B FF 99 03 0A 13 39 99 03 0B 13 04; (4*) 2025/03/19 10:57 -> 2025/03/19 11:04
08 AC 00 0C 19 41 30 31 14 B2 99 03 08 13 22 99 03 08 13 25; (8#) 2025/03/19 08:34 -> 2025/03/19 08:37
08 AC 00 0C 09 41 30 31 14 B2 99 03 07 12 0F 99 03 07 12 12; (9#)
08 AC 00 0C 09 41 30 31 14 B2 99 03 09 11 02 99 03 09 11 05; (10#)
08 AC 00 0C 09 41 30 31 14 B2 99 03 07 11 19 99 03 07 11 1C; (11#)
08 AC 00 0C 09 41 30 31 14 B2 99 03 0B 0F 16 99 03 0B 0F 19; (12#)
08 AC 00 0C 09 41 30 31 14 B2 99 03 07 0C 14 99 03 07 0C 17;
08 AC 00 0C 09 41 30 31 14 B2 99 03 09 0B 07 99 03 09 0B 0A;
08 AC 00 0C 09 41 30 31 14 B2 99 03 08 05 2C 99 03 08 05 2F;
08 AC 00 0C 09 41 30 31 14 B2 99 03 07 03 12 99 03 07 03 15;

read 10 0xC2
10 9E 00 0E 11 41 31 31 0C 13 99 03 0B 13 17 99 03 0B 13 24; (1*)
10 9E 00 0E 11 41 31 31 0B FF 99 03 0B 13 16 99 03 0B 13 23; (2*)
10 9E 00 0E 11 41 31 31 0C 13 99 03 0A 13 3B 99 03 0B 13 05; (3*)
10 9E 00 0E 11 41 31 31 0B FF 99 03 0A 13 39 99 03 0B 13 04; (4*)
00 9E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00;
00 9E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00;
00 9E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00;
00 9E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00;
00 9E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00;
00 9E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00;
```

### boiler (0x08)
```
read 8 0xC0 empty

read 8 0xC1 empty

read 8 0xC2
08 83 00 0C 09 41 30 31 14 B2 99 04 08 02 1A 99 04 08 02 1D; (1#) 2025/04/02 08:26 -> 2025/04/02 08:29
08 83 00 0C 09 41 30 31 14 B2 99 03 07 1F 10 99 03 07 1F 13; (2#)
08 83 00 0C 09 41 30 31 14 B2 99 03 07 1B 0B 99 03 07 1B 0E; (3#)
08 83 00 0C 09 41 30 31 14 B2 99 03 08 1A 0B 99 03 08 1A 0E; (4#)
08 83 00 0C 09 41 30 31 14 B2 99 03 07 1A 10 99 03 07 1A 13; (5#)
08 83 00 0C 09 41 30 31 14 B2 99 03 09 14 06 99 03 09 14 09; (6#)
08 83 00 0C 09 41 30 31 14 B2 99 03 07 14 0D 99 03 07 14 10; (7#)
08 83 00 0C 09 41 30 31 14 B2 99 03 08 13 21 99 03 08 13 24; (8#)
08 83 00 0C 09 41 30 31 14 B2 99 03 07 12 0E 99 03 07 12 11; (9#)
08 83 00 0C 09 41 30 31 14 B2 99 03 09 11 01 99 03 09 11 04; (10#)
08 83 00 0C 09 41 30 31 14 B2 99 03 07 11 19 99 03 07 11 1C;  (11#)
08 83 00 0C 09 41 30 31 14 B2 99 03 0B 0F 15 99 03 0B 0F 18; (12#)
08 83 00 0C 09 41 30 31 14 B2 99 03 07 0C 13 // incomplete.
```

This shows:
- 0xC1 is the continuation of 0xC0 (compare the thermostat´s last entry in 0xC0 and first entry in 0xC1) 0xC0 hold 10 entries and 0xC1 the following 10 entries
- the thermostat holds the global error list in 0xC0 and 0xC1
- the boiler has no global error list in 0xC0 and 0xC1 => the thermostat is the "master"
- 0xC2 is a local error list
- in the global error list you can find all local errors. I marked the corresponding entries with 1#, 2#, ... or 1*, 2* ... . Sometimes there is a glitch in the minute entry, this may be because of a delay in data transfer

MichaelDvP, 2025-04-02:
Some heatpumps also have 0xC6 and 0xC7, could you check?

Madaraxa, 2025-04-03:
My question is answered but it seems you guys keep fixing stuff. So I wont close the issue but you can if u like. Thanks again for the fast help

closed 2025-04-05 by proddy

Madaraxa, 2025-04-22:
@proddy could it be, that the value of last error code (thermostat) is only updated once the EMS-ESP restarts / boots?
I have errors at the moment and got no notification about it until just a few minutes ago, when i updated from Dev8 to Dev9.
The error is from yesterday 16:59.
If not, this is an issue of my smarthome (iobroker) Adapter.

Madaraxa, 2025-04-22:
Ok. I had the Chance to check:
I opened ems-esp Devices - Thermostat at 17:07 and it showed the following:

[image]
As you can See it shows an error from 21.04. in last error.
The i restarted the EMS-ESP and it showed the following

[image]
There is an error from today.

[image]

So it seems the last error code field is not Updated if EMS-ESP is not restarted. Is this fixable?

[emsesp_system_info.json](https://github.com/user-attachments/files/19852240/emsesp_system_info.json)