emsesp/EMS-ESP32 #2507: 'Download' Button below dev Version Downloads stable

Madaraxa, 2025-04-04:
Hi.
On the Firmware Page there is a download Button below the stable and dev Version (if the System is Not on latest).
I am on the dev Version and when updating from dev 5 to dev 6 it downloaded the stable 3.7.2 and not the latest 3.7.3 dev6.
Same Problem was with dev4 to 5.
Nothing serious :)
I have an ems-esp without the psram (so 16mb Version)

MichaelDvP, 2025-04-04:
If the page does not look like this: 
[image]
clear website data cache for this page in browser.

MichaelDvP, 2025-04-04:
Yes, what's wrong? You are using the latest dev and can only dwnload the stable. If you are on dev.5 you'll get the update available button.

Madaraxa, 2025-04-04:
I maybe missunderstood. I thought there should be a 'switch to stable' Button and Not a 'download' (herunterladen) Button.
When i was on dev5 and i clicked on herunterladen beside my dev Version it downloaded the 3.7.2 and not the 3.7.3dev6. And - if i understand correct - this Was because i had Not cleared Cache?

proddy, 2025-04-04:
@Madaraxa could you post your System Info?

The Version web page will only show the "Download" button for 4MB or 16MB boards based on the older ESP32 chipset and also without PSRAM. This firmware doesn't have https support so EMS-ESP can't automatically fetch and install versions, so the behaviour is different.

proddy, 2025-04-04:
> But shouldnt the download Button download the correct Firmware (so the download Button Next to stable the 3.7.2 and the one Next to development the 3.7.3dev6 at the Moment)?

Yes, it's a bug that I'll fix

commit 3ed5b651 (2025-04-05): fixes #2507
commit baa4f2eb (2025-04-05): Merge pull request #2508 from proddy/dev

fixes #2507 - download based on tasmota builds
closed 2025-04-05 by proddy