emsesp/EMS-ESP32 #2518: Same download link for stable/develop version

misa1515, 2025-04-12:
### DESCRIPTION

_A clear and concise description of what the problem is or the change requested._

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
  Paste System information here....

```

### TO REPRODUCE

Download link for stable and develop versions are same

### SCREENSHOTS

[image]

VlastiBroucek, 2025-04-13:
see #2497

proddy, 2025-04-13:
this has been fixed in a later dev release. you will need to download it manually

closed 2025-04-13 by proddy