emsesp/EMS-ESP32 #2524: IPM mixer Data missing after Update v3.6.5 to v3.7.2

knopfsocke, 2025-04-17:
### DESCRIPTION
Hello,

after updateing from EMS-ESP32 v3.6.5 to v3.7.2 I miss IPM mixer 0x41 data which was added in 2022 by prody/MiachelDvP to ems-esp. Details see https://github.com/emsesp/EMS-ESP32/issues/437.

The IPM is still visible after the update, but it has zero entities. 

Everything works fine before the update for 3 years :-)

Thanks in advance for your support.

### REQUESTED INFORMATION

[system.json](https://github.com/user-attachments/files/19791617/system.json)

MichaelDvP, 2025-04-17:
Please try dev.8 from https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest
We have moved the dhw modules to an extra class, but did not map the IPM at 0x41 correctly.

knopfsocke, 2025-04-17:
Tested sucessfully with EMS-ESP-3_7_3-dev_8-ESP32-4MB.bin. From my side, we can close this issue. Thanks a lot!

proddy, 2025-04-17:
nice!

commit 2854e9cb (2025-04-20): IPM DHW module, fixes #2524
closed 2025-04-20 by proddy