emsesp/EMS-ESP32 #2531: Thermostat - last error code not updating / only by restart

Madaraxa, 2025-04-22:
As i cant reopen my last issue:

I opened ems-esp Devices - Thermostat at 17:07 and it showed the following:

[image]
As you can See it shows an error from 21.04. in last error.
The i restarted the EMS-ESP and it showed the following at 17:08:

[image]
There is an error from today.

[image]

So it seems the last error code field is not Updated if EMS-ESP is not restarted. Is this fixable?

[emsesp_system_info.json](https://github.com/user-attachments/files/19852240/emsesp_system_info.json)

proddy, 2025-04-22:
Hi @Madaraxa , I saw your Discord message on this too. I think @MichaelDvP is the best person to answer this question and he went deep into finding out how codes are set and read

VlastiBroucek, 2025-04-22:
According to @Sbried these are "unicast-on-request" (https://docs.emsesp.org/ems-library/#telegram-0x00c6); hence they will update only on active read (e.g. restart of the system)?

Madaraxa, 2025-04-23:
So can I schedule an Auto restart of the EMS-ESP every day :)?

VlastiBroucek, 2025-04-23:
Maybe follow what @proddy said and wait what @MichaelDvP will have to say.

Madaraxa, 2025-04-23:
Hmm. Am I missing something? As Michael got tagged He will answer as soon as He likes. Or Do i have to write him personally?

VlastiBroucek, 2025-04-23:
@MichaelDvP will respond as soon as he knows good answer and has time to do so - don't forget these guys are doing this in their free time not as a job.

Madaraxa, 2025-04-23:
Man. This is exactly what i wrote :) an I didnt Tag him so He wont get annoyed by notifications

Sbried, 2025-04-23:
All errors are unicast-on-request. What I saw, EMS-ESP issues a read request every 10 minutes to get the last error. Therefore the last error field should be updated.
But, @VlastiBroucek, 0xC6 is the error history of a heat pump. If I read the code correct, the last error message of the thermostat is filled either from 0x12, 0x13 or 0xC0. Whatever is available, I think it will be 0xC0. All 3 telegrams hold _system_ error lists.
@Madaraxa , what you could do: First find out which error list is used by your system with `read 10 12`, `read 10 13` and `read 10 c0` in the console. Whereever you get data back: This is the used telegram. Then watch the telegram in the console with `watch on xx`(xx=the telegram that contains data). Let watch run half hour, then you should see the answers and if the last error is updated in the WebUI. Can you tell what your A51-error is?

VlastiBroucek, 2025-04-23:
Sorry for not being clear, I used 0x0C6 as an example only...

Madaraxa, 2025-04-23:
@Sbried  did that. its 0xC0. read 10 12 and 10 13 got back empty. Before #2501 the Cx didnt get read out by ems-esp at all (if i understood correctly).

I did a `watch on c0` 18 Minutes ago and this was happening just now (09:38):

ems-esp:$ read 10 c0
000+16:21:49.482 N 14: [emsesp] me(0x0B) R thermostat(0x10), RCErrorMessage(0xC0), length: 27
000+16:21:49.596 N 15: [emsesp] thermostat(0x10) W me(0x0B), RCErrorMessage(0xC0), data: 30 A4 00 0E 11 41 35 31 17 8C 99 04 0C 16 2B 99 04 0C 16 2C 30 A4 00 0E 11 41 35
000+16:21:49.596 N 16: [emsesp] me(0x0B) R thermostat(0x10), RCErrorMessage(0xC0), length: 27 (offset 27)
000+16:21:49.725 N 17: [emsesp] thermostat(0x10) W me(0x0B), RCErrorMessage(0xC0), data: 31 17 8C 99 04 10 15 39 99 04 10 15 3B 30 A4 00 0E 11 41 35 31 17 8C 99 04 0F 15 (offset 27)
000+16:21:49.725 N 18: [emsesp] me(0x0B) R thermostat(0x10), RCErrorMessage(0xC0), length: 27 (offset 54)
000+16:21:49.848 N 19: [emsesp] thermostat(0x10) W me(0x0B), RCErrorMessage(0xC0), data: 04 99 04 0F 15 04 30 A4 00 0E 11 41 35 31 17 8C 99 04 0F 14 25 99 04 0F 14 25 30 (offset 54)
000+16:21:50.553 N 20: [emsesp] me(0x0B) R thermostat(0x10), RCErrorMessage(0xC0), length: 27 (offset 81)
000+16:21:50.669 N 21: [emsesp] thermostat(0x10) W me(0x0B), RCErrorMessage(0xC0), data: A4 00 0E 11 41 35 31 17 8C 99 04 0E 14 1B 99 04 0E 14 1C 30 A4 00 0E 11 41 35 31 (offset 81)
000+16:21:50.669 N 22: [emsesp] me(0x0B) R thermostat(0x10), RCErrorMessage(0xC0), length: 27 (offset 108)
000+16:21:50.787 N 23: [emsesp] thermostat(0x10) W me(0x0B), RCErrorMessage(0xC0), data: 17 8C 99 04 0C 14 18 99 04 0C 14 1D 10 9E 00 0E 11 41 31 31 0F AB 99 04 0E 0D 1A (offset 108)
000+16:21:51.475 N 24: [emsesp] me(0x0B) R thermostat(0x10), RCErrorMessage(0xC0), length: 27 (offset 135)
000+16:21:54.133 N 25: [emsesp] me(0x0B) R thermostat(0x10), RCErrorMessage(0xC0), length: 27 (offset 135)
000+16:21:54.240 N 26: [emsesp] thermostat(0x10) W me(0x0B), RCErrorMessage(0xC0), data: 99 04 0E 0D 1B 30 A4 00 0E 11 41 35 31 17 8C 99 04 0C 0C 13 99 04 0C 0C 1C 30 A4 (offset 135)
000+16:21:54.240 N 27: [emsesp] me(0x0B) R thermostat(0x10), RCErrorMessage(0xC0), length: 27 (offset 162)
000+16:21:54.367 N 28: [emsesp] thermostat(0x10) W me(0x0B), RCErrorMessage(0xC0), data: 00 0E 11 41 35 31 17 8C 99 04 0C 0B 19 99 04 0C 0B 1D 30 A4 00 0E 11 41 35 31 17 (offset 162)
000+16:21:54.367 N 29: [emsesp] me(0x0B) R thermostat(0x10), RCErrorMessage(0xC0), length: 27 (offset 189)
000+16:21:54.438 N 30: [emsesp] thermostat(0x10) W me(0x0B), RCErrorMessage(0xC0), data: 8C 99 04 0C 09 19 99 04 0C 09 20 (offset 189)
ems-esp:$ watch on c0
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a telegram type of 0xC0
ems-esp:$
000+16:31:13.659 W 31: [telegram] Incomplete Rx: 90 00 FF 00 02 16
000+16:31:44.779 E 32: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B B0 FF 00 FF 02 5F

It wrote 16:31 but the lines came at 16:39.

Sbried, 2025-04-23:
So far you didn´t fetch a read request on 0xC0. What you see is just an error message for an incomplete Rx..... Let it run.
Yes, when you want to finish the watch, type in `watch off`.

VlastiBroucek, 2025-04-23:
BTW, @Madaraxa , I don't think that "000+16:40:42.850" in telnet console means current time, IMHO it is time since last restart of EMS-ESP. Got caught on that one too... Why it is that way would be question for @proddy and @MichaelDvP

VlastiBroucek, 2025-04-23:
Yes, I am thinking the same - perhaps 0x0C0 IS NOT fetched regularly as @Sbried  thinks. My `watch c0` has been running for 15 minutes and I have not seen fetch either... Maybe it depends on type/model of the thermostat? Mine is:
`Thermostat: CR50/RC200, CW100, CR120, CR50 (DeviceID:0x18, ProductID:157, Version:43.02) (54)`

MichaelDvP, 2025-04-23:
> All errors are unicast-on-request. 

No, the ems (1.0) devices always broadcast on change (also 0x10, 0x11, 0x12). This was a clear logic, don't know why Bosch now changed it in ems+.

@Madaraxa Please try  actual dev.9 from https://github.com/MichaelDvP/EMS-ESP32/releases/tag/lates 
I don't want to fetch 0xC0 every minute, we fetch so much messages and this is critical on no-psram ems-esp hardware. Now the 0xC0 is fetchted once on every 0xBF broadcast. Maybe this work, maybe Bosch changes the BF in upcoming devices.

Sbried, 2025-04-23:
I also started a `watch 10 c0` on my system. Yesterday I did a watch on 0xC2 and 0xC6 and those updated every 10 minutes. Let´s see what happens with 0xC0....

VlastiBroucek, 2025-04-23:
> No, the ems (1.0) devices always broadcast on change (also 0x10, 0x11, 0x12). This was a clear logic, don't know why Bosch now changed it in ems+.

@Sbried , can you change your doco accordingly?

Sbried, 2025-04-23:
> [@Sbried](https://github.com/Sbried) , can you change your doco accordingly?

I´ll do so

Madaraxa, 2025-04-23:
ems-esp:$ watch on c0
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a telegram type of 0xC0
000+00:01:14.295 E 17: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 FF 00 FF 02 1B
000+00:01:29.983 W 18: [telegram] Incomplete Rx: 90 00 FF 00 02 1D 21
000+00:03:19.572 E 19: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 FF 00 FF 03 72
000+00:05:18.856 N 20: [emsesp] me(0x0B) R thermostat(0x10), RCErrorMessage(0xC0), length: 20
000+00:05:18.939 N 21: [emsesp] thermostat(0x10) W me(0x0B), RCErrorMessage(0xC0), data: 30 A4 00 0E 11 41 35 31 17 8C 99 04 0C 16 2B 99 04 0C 16 2C
ems-esp:$
ems-esp:$
ems-esp:$
000+00:15:16.339 N 22: [emsesp] me(0x0B) R thermostat(0x10), RCErrorMessage(0xC0), length: 20
000+00:15:16.427 N 23: [emsesp] thermostat(0x10) W me(0x0B), RCErrorMessage(0xC0), data: 30 A4 00 0E 11 41 35 31 17 8C 99 04 0C 16 2B 99 04 0C 16 2C

Sbried, 2025-04-23:
> I would now `watch off` to not forget it. is this ok? or do you need more? Btw: personally every 10 Minutes is more then enough for me. even once a day would be ok :) I just dont wanna miss an error. Especially in winter its hard to get someone to fix it fast :)

Looks like your problem is solved after using the latest dev release and you can `watch off`.

I will also add the 0xbf telegram to the docu. Seems to be the key player in the newer error logging system 0xC.. .

Madaraxa, 2025-04-23:
thank you guys. 
btw my heating system is from early 2015 or earlier. so the "newer" error logging system isnt that new?!
as i dont understand every step: can you close this issue if its solved :)

MichaelDvP, 2025-04-23:
@Madaraxa Can you please make a full log (log all). The 0xBF normally is send if an error occured and if it is cleared, but not periocally every 10 sec. Maybe i need to check the error in BF.

Sbried, 2025-04-23:
In my system 0xBF is distributed every 10 minutes from each bus member:
```
watch on bf
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a telegram type of 0xBF
000+17:02:47.020 N 14: [emsesp] thermostat(0x10) B all(0x00), SystemError(0xBF), data: 10 9E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+17:05:29.120 N 15: [emsesp] boiler(0x08) B all(0x00), ErrorMessage(0xBF), data: 08 AC 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+17:06:41.462 N 16: [emsesp] thermostat(0x38) B all(0x00), SystemError(0xBF), data: 38 C8 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+17:12:48.207 N 17: [emsesp] thermostat(0x10) B all(0x00), SystemError(0xBF), data: 10 9E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+17:15:29.182 N 18: [emsesp] boiler(0x08) B all(0x00), ErrorMessage(0xBF), data: 08 AC 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+17:16:41.237 N 19: [emsesp] thermostat(0x38) B all(0x00), SystemError(0xBF), data: 38 C8 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+17:22:50.566 N 20: [emsesp] thermostat(0x10) B all(0x00), SystemError(0xBF), data: 10 9E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+17:25:29.232 N 21: [emsesp] boiler(0x08) B all(0x00), ErrorMessage(0xBF), data: 08 AC 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+17:26:41.336 N 22: [emsesp] thermostat(0x38) B all(0x00), SystemError(0xBF), data: 38 C8 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
ems-esp:$ watch off
```
The distribution is periodically, on request (and on change).

Madaraxa, 2025-04-23:
@MichaelDvP how many time / lines do you need? if i put `log all` it kinda spams 2-3 messages every second.

VlastiBroucek, 2025-04-23:
In my case, only CR50 Thermostat (0x18) is broadcasting 0xBF regularly, nothing from the boiler (0x08):
```
000+06:03:23.124 N 1: [emsesp] thermostat(0x18) B all(0x00), SystemError(0xBF), data: 18 9D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+06:13:22.154 N 2: [emsesp] thermostat(0x18) B all(0x00), SystemError(0xBF), data: 18 9D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

Direct reads produce:
```
ems-esp:$ read 8 bf
000+06:38:19.620 N 1: [emsesp] me(0x0B) R boiler(0x08), ErrorMessage(0xBF), length: 27
000+06:38:19.643 N 2: [emsesp] boiler(0x08) W me(0x0B), ErrorMessage(0xBF), data: <empty>
ems-esp:$ read 18 bf
000+06:39:17.343 N 3: [emsesp] me(0x0B) R thermostat(0x18), SystemError(0xBF), length: 27
000+06:39:17.425 N 4: [emsesp] thermostat(0x18) W me(0x0B), SystemError(0xBF), data: 18 9D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

Madaraxa, 2025-04-25:
Fixed with dev10 as i understand :) Thanks again

closed 2025-04-25 by Madaraxa