emsesp/EMS-ESP32 #2540: SECURITY: admin user token does not change with the change of the password

VlastiBroucek, 2025-04-26:
### DESCRIPTION

admin user token does not change with the change of the password and stays the same as the one in 

- test/test_api/api_test.sh
- scripts/memory_test.py
- test/test_api/api_test.http

I have found this by accident while following #2516 

### TO REPRODUCE

On latest dev.10,

tried changing admin user password several times, the token stays the same!!!

### EXPECTED BEHAVIOUR

Token should change with the change of password

### SCREENSHOTS

_If applicable, add screenshots to help explain your issue._

### ADDITIONAL CONTEXT

_Add any other context about the issue here._

**(Please remember to close the issue when it has been addressed)**

proddy, 2025-04-26:
thanks, looking at it now...

closed 2025-04-26 by proddy

VlastiBroucek, 2025-04-26:
Well, let's agree to disagree. It would be more intuitive to use the password as a signature as you actually said yesterday (and promptly changed today 😁):

> The token comes from the admin user's account and derived from the **username and password**. if you haven't changed it from **admin/admin then it'll be the same**. This is clearly described in the documentation at https://docs.emsesp.org/Commands/#http

and I would not agree it is an industry standard to do it the way it is currently done in EMS-ESP. I don't want to boast about anyone's credentials, but I am well published author in the field of forensic computing and security...

The mentioned doco (https://docs.emsesp.org/Commands/#http) doesn't make that clear either.

While I defended you guys previously, arguing you do this in free time etc., and always happy with your work, this particular response is rather disappointing...

VlastiBroucek, 2025-04-26:
Sorry, just forget my initial reaction that I have now deleted - let's just say this should be made clearer in the documentation. Obviously, it was even said incorrectly by yourself yesterday:

> The token comes from the admin user's account and derived from the username and password. if you haven't changed it from admin/admin then it'll be the same. This is clearly described in the documentation at https://docs.emsesp.org/Commands/#http

proddy, 2025-04-26:
I did see your initial reaction and it made me think. To be frank, I'm disappointed too. There are no "guys" it's just me and a lot of help from Michael which I truly honor. The challenge is people purchase a kit and expect a level of entitlement and support from something that is a free hobby project. I'm push out 3.7.2 as the last release that I will support and then close the project and focus on something else that gives me energy.

VlastiBroucek, 2025-04-26:
I definitely did not want to push you out. By guys I meant you and Michael. You are doing fantastic job and if you stop doing it, it will be shame.

IMHO, you should not concentrate on little irrelevant things like the ridiculous request to trigger backlight on thermostat from HA but just on core things. Often you are trying to be too nice to us, often too demanding users... 

I have done my share of development in the past, but never in the IoT area. I am considering starting learning it, but even though I am officially retired it would be on top of my research that I still do on honorary basis and that keeps me reasonably busy.

Sorry, If I made you disappointed - I did not mean it, the response was bit too fast without truly checking JWT... I still think in the area of "super" security even when talking about system that should be not accessible from Internet, hence does not need top notch approach...

Cheers

Vlasti