emsesp/EMS-ESP32 #2542: Access to RC310 temperature while running with virtual RC200 sensor?

airhead1234, 2025-05-01:
I am a happy EMS-ESP user with a Buderus WS170 controlled by a RC310 and use EMS-ESP to create a virtual RC200 temperature sensor. The RC310 is set to regulate based on the remote temperature of the virtual RC200. So far so good.

Now I am wondering whether there is any chance to still access the temperature measured by the RC310 in Home Assistant. To all I can see, all room temperature values reported to Home Assistant via EMS-ESP are the values from the virtual RC200. 

Internally, it seems like the RC310 still does continue to measure temperature becasue there is a setting in which I can opt to regulate based on the lower value of its internal temperature measurement and the temperature measured by the remote sensor.

Is there any chance to have the temperature measured by the RC310 in Home Assistant despite having set it to be controlled by the remote sensor?

Thanks!

MichaelDvP, 2025-05-14:
Some thermostats publish the temperature of the internal sensors, e.g. RC30/35 in 0x3A. But for the other thermostats we don't know if and where the temperatures are published. You can try to log all telegrams and find the temperature values. Should be a 16 bit value with scale 0.1°C, eg. `00 DE` for 22.2°C. If you disable the remote the RC310 publishes the roomtemp, so you'll have the right value to search for.

airhead1234, 2025-05-17:
Thank you. I have tried to do what you have suggest, but did not record the temperature I was looking for.

Have I done it correctly? I entered `watch on` in the console, disabled the remote, noted the room temperature from the RC310 in the console logs, enabled the remote again, and waited for a while to see whether the RC310 temperature turned up in the console logs. Unfortunately, it did not.

Do you know how long I would have to wait to see the value in the logs? Is there a frequency at which all values are published? Or would I only see it if it changed? In the latter case my strategy to wait for a know number would not work.

Thanks!

MichaelDvP, 2025-05-17:
> Have I done it correctly?

I think yes. I would use the web status->log, set it to log all, leave remote disabled and wait 15 min. Then download the log, open in editor. Find the first telegram 0x2A5 (RC300Monitor), the first two bytes are roomtemperature hc1, press ctrl-F and enter the bytes (eg. 00 DE), and check. You should see every 10 sec a 2A5 telegram  (or the temperature changed) and maybe another finding. If the 2A5 offset0 changes and the other also: bingo.

airhead1234, 2025-05-17:
Thank you. Using the web logging is much more convenient. Unfortunately, there was no bingo. It seems like the RC310 does not send the internal temperature when the remote is on. 

As you wrote the first two bytes of 0x2A5 are the room temperature. Unfortunately, the value displayed there does not show up at any other address.

Thank you anyways for your help and the great piece of software you are maintaining.

closed 2025-05-17 by airhead1234