emsesp/EMS-ESP32 #2543: Add entity for storage charging optimization option

SMHRambo, 2025-05-01:
I have found a new option in my RC310 controller that was not available in my RC300 controller.
In the manual, the option is called “Speicherladeoptimierung/Storage charging optimization”.
The option is helpful to reduce energy consumption by switching off the burner early during the charging process, but not the charging pump.

The telegram is:
Option OFF:
thermostat(0x10) W boiler(0x08), UBAParameterWW(0x33), data: 00 (offset 4)
Option ON:
thermostat(0x10) W boiler(0x08), UBAParameterWW(0x33), data: FF (offset 4)

```json
{
  "type": "entities",
  "Entities": {
    "entities": [
      {
        "id": 0,
        "ram": 0,
        "device_id": 8,
        "type_id": 51,
        "offset": 4,
        "factor": 255,
        "name": "chargeoptimization",
        "uom": 0,
        "value_type": 0,
        "writeable": true
      }
    ]
  }
}
```

```json
{
  "system": {
    "version": "3.7.3-dev.10",
    "uptime": "000+00:29:43.441",
    "uptimeSec": 1783,
    "platform": "ESP32",
    "cpuType": "ESP32-D0WDQ6",
    "arduino": "Tasmota Arduino v2.0.17",
    "sdk": "4.4.8.240628",
    "freeMem": 150,
    "maxAlloc": 83,
    "freeCaps": 101,
    "usedApp": 1653,
    "freeApp": 331,
    "partition": "app0",
    "flash_chip_size": 4096,
    "resetReason": "Software reset CPU / Software reset CPU",
    "psram": false,
    "model": ""
  },
  "network": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -65,
    "WIFIReconnects": 0,
    "BSSID": "set",
    "TxPowerSetting": 0,
    "staticIP": false,
    "lowBandwidth": false,
    "disableSleep": true,
    "enableMDNS": true,
    "enableCORS": false,
    "APProvisionMode": "disconnected",
    "APSecurity": "wpa2",
    "APSSID": "ems-esp"
  },
  "ntp": {
    "NTPStatus": "connected",
    "enabled": true,
    "server": "10.13.12.15",
    "tzLabel": "Europe/Berlin",
    "timestamp": 1746131229
  },
  "mqtt": {
    "MQTTStatus": "connected",
    "MQTTPublishes": 1468,
    "MQTTQueued": 0,
    "MQTTPublishFails": 0,
    "MQTTReconnects": 0,
    "enabled": true,
    "clientID": "heizung",
    "keepAlive": 60,
    "cleanSession": false,
    "entityFormat": 3,
    "base": "ems-esp",
    "discoveryPrefix": "homeassistant",
    "discoveryType": 0,
    "nestedFormat": 2,
    "haEnabled": true,
    "mqttQos": 1,
    "mqttRetain": true,
    "publishTimeHeartbeat": 60,
    "publishTimeBoiler": 10,
    "publishTimeThermostat": 10,
    "publishTimeSolar": 10,
    "publishTimeMixer": 10,
    "publishTimeWater": 10,
    "publishTimeOther": 60,
    "publishTimeSensor": 10,
    "publishSingle": false,
    "publish2command": false,
    "sendResponse": false
  },
  "syslog": {
    "enabled": false
  },
  "sensor": {
    "temperatureSensors": 1,
    "temperatureSensorReads": 355,
    "temperatureSensorFails": 1
  },
  "api": {
    "APICalls": 370,
    "APIFails": 1
  },
  "bus": {
    "busStatus": "connected",
    "busProtocol": "Buderus",
    "busTelegramsReceived": 2533,
    "busReads": 710,
    "busWrites": 0,
    "busIncompleteTelegrams": 1,
    "busReadsFailed": 1,
    "busWritesFailed": 0,
    "busRxLineQuality": 100,
    "busTxLineQuality": 100
  },
  "settings": {
    "boardProfile": "CUSTOM",
    "locale": "de",
    "txMode": 1,
    "emsBusID": 11,
    "showerTimer": true,
    "showerMinDuration": 180,
    "showerAlert": false,
    "phyType": 0,
    "rxGPIO": 23,
    "txGPIO": 5,
    "dallasGPIO": 14,
    "pbuttonGPIO": 0,
    "ledGPIO": 25,
    "ledType": 0,
    "hideLed": false,
    "noTokenApi": false,
    "readonlyMode": false,
    "fahrenheit": false,
    "dallasParasite": false,
    "boolFormat": 6,
    "boolDashboard": 2,
    "enumFormat": 1,
    "analogEnabled": false,
    "telnetEnabled": false,
    "maxWebLogBuffer": 25,
    "modbusEnabled": false,
    "forceHeatingOff": false,
    "developerMode": true
  },
  "devices": [
    {
      "type": "boiler",
      "name": "Logano GB1*5, Logamatic MC10",
      "deviceID": "0x08",
      "productID": 72,
      "brand": "",
      "version": "03.05",
      "entities": 73,
      "handlersReceived": "0x10 0x11 0x14 0x15 0x1C 0x18 0x19 0x34 0x04",
      "handlersFetched": "0x16 0x33",
      "handlersPending": "0xBF 0xC2 0xC6 0x1A 0x35 0x2E 0x3B"
    },
    {
      "type": "thermostat",
      "name": "RC310/RC3*0, Moduline 3000/1010H, CW400, Sense II, HPC410",
      "deviceID": "0x10",
      "productID": 158,
      "brand": "",
      "version": "74.05",
      "entities": 63,
      "handlersReceived": "0x06 0xA2 0x02BA 0x02BB 0x031D 0x0267 0xBF 0xC0",
      "handlersFetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0291 0x0292 0x0293 0x0294 0x02F5 0x02F6 0x031B 0x023A 0x0240 0x0269",
      "handlersPending": "0xA3 0x12 0x13 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02BC 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CC 0x0467 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031E 0xBB 0x023E 0x016E"
    },
    {
      "type": "mixer",
      "name": "MM50/MM50",
      "deviceID": "0x20",
      "productID": 159,
      "brand": "",
      "version": "20.03",
      "entities": 7,
      "handlersReceived": "0x02D7",
      "handlersFetched": "0x02CD",
      "handlersIgnored": "0x02E1 0x02EB 0x0255 0x16 0x33 0xBF"
    },
    {
      "type": "controller",
      "name": "BC10, RFM20",
      "deviceID": "0x09",
      "productID": 68,
      "brand": "",
      "version": "02.03",
      "entities": 0
    },
    {
      "type": "temperaturesensor",
      "name": "temperaturesensor",
      "entities": 1
    },
    {
      "type": "custom",
      "name": "custom",
      "entities": 2
    }
  ]
}
```

[image]
[image]

MichaelDvP, 2025-05-04:
Thanks, we had this as `hysteresis off`, but norberts1 also says `charge optimize` and i can only set it to 0 or 0xFF (0,-1). I'll change the name and type to bool.

SMHRambo, 2025-05-06:
Thank you.
Good to know, I also had problems with hysteresis off.
The only settings that were possible are 0 and -1, but I didn't connect that with the charge optimize option in the RC310.
It seems that only the SM200 module has a hysteresis off option, unfortunately I don't know anyone who uses the module to check this.

MichaelDvP, 2025-05-10:
Please try dev11 from https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest

SMHRambo, 2025-05-11:
So I tested the dev11 today.

- dhw/chargepump seems to be working. I won't be able to say for sure until tomorrow, as grafana indicates that the pump didn't start until 3 minutes after the burner started. I have never seen this behavior before, but it could be that this pump is also affected by the pumpontemp.

- dhw/chargeoptimization is unfortunately causing problems. I can see the current status, but when I change the value, nothing happens. 

```
2025-05-11T20:48:52.594 TRACE 54168: [emsesp] boiler(0x08) B all(0x00), UBAMonitorWW(0x34), data: 3F 02 54 7D 00 21 00 00 03 00 00 23 D7 00 01 4D
2025-05-11T20:48:51.997 TRACE 54169: [emsesp] thermostat(0x10) W me(0x0B), RC300WW2mode(0x02F6), data: 00 00 04 03 00 00 08 01 00 08 04 00 00 00 00 00 00 00 00 00 01
2025-05-11T20:48:52.203 TRACE 54170: [emsesp] me(0x0B) R thermostat(0x10), RC300WWtemp(0x031B), length: 25
2025-05-11T20:48:52.227 TRACE 54171: [emsesp] thermostat(0x10) W me(0x0B), RC300WWtemp(0x031B), data: 41 37
2025-05-11T20:48:52.408 TRACE 54172: [emsesp] me(0x0B) R thermostat(0x10), RC300OutdoorTemp(0x023A), length: 25
2025-05-11T20:48:52.434 TRACE 54173: [emsesp] thermostat(0x10) W me(0x0B), RC300OutdoorTemp(0x023A), data: 00 C0
2025-05-11T20:48:53.563 TRACE 54174: [emsesp] me(0x0B) R thermostat(0x10), RC300Settings(0x0240), length: 25
2025-05-11T20:48:53.661 TRACE 54175: [emsesp] thermostat(0x10) W me(0x0B), RC300Settings(0x0240), data: 32 00 03 00 00 00 00 00 FF 03 F7 06 FF 00 00 00 00 00 00 00 00 00 00
2025-05-11T20:48:53.767 INFO 54176: [command] Called command boiler/chargeoptimization (Ladungsoptimierung) with value AUS and id 9 on device 0x08
2025-05-11T20:48:52.849 TRACE 54177: [emsesp] me(0x0B) R thermostat(0x10), RC300Holiday(0x0269), length: 25
2025-05-11T20:48:52.849 TRACE 54178: [emsesp] thermostat(0x10) W me(0x0B), RC300Holiday(0x0269), data: 11 01 01 11 01 01 04 11 02 00 00 00 00 00 00 00 00 00 00
2025-05-11T20:48:53.127 TRACE 54179: [emsesp] me(0x0B) R mixer(0x20), MMPLUSConfigMessage(0x02CD), length: 25
2025-05-11T20:48:53.127 TRACE 54180: [emsesp] mixer(0x20) W me(0x0B), MMPLUSConfigMessage(0x02CD), data: FF 0B 03 00 1E 00
2025-05-11T20:48:53.312 TRACE 54181: [emsesp] me(0x0B) R boiler(0x08), UBAMonitorFast(0x18), length: 1 (offset 23)
2025-05-11T20:48:53.375 TRACE 54182: [emsesp] boiler(0x08) W me(0x0B), UBAMonitorFast(0x18), data: 00 (offset 23)
2025-05-11T20:49:01.796 TRACE 54183: [emsesp] boiler(0x08) B all(0x00), UBAMonitorFast(0x18), data: 07 02 1C 64 00 00 00 00 00 02 54 7D 00 83 00 FF FF FF 30 48 00 00 FF 00 00 80 00
2025-05-11T20:49:01.076 TRACE 54184: [emsesp] boiler(0x08) B all(0x00), UBAMonitorFast(0x18), data: FF FF FF FF 00 00 00 07 00 00 (offset 27)
2025-05-11T20:49:02.483 TRACE 54185: [emsesp] boiler(0x08) B all(0x00), UBAMonitorWW(0x34), data: 3F 02 54 7D 00 21 00 00 03 00 00 23 D7 00 01 4D
2025-05-11T20:49:06.688 TRACE 54186: [emsesp] thermostat(0x10) B all(0x00), RC300Floordry(0x0267), data: 00 00
2025-05-11T20:49:06.931 TRACE 54187: [emsesp] thermostat(0x10) B all(0x00), RC300Monitor(0x02A5), data: 00 BA 31 2A 00 00 2A 22 00 84 03 03 01 00 84 03 00 00 00 11 01 01 07 47 00
2025-05-11T20:49:06.286 TRACE 54188: [emsesp] thermostat(0x10) B all(0x00), RC300Monitor(0x02A5), data: 06 08 00 00 00 00 FF 00 4B 00 3C 01 FF 01 02 (offset 25)
2025-05-11T20:49:06.477 TRACE 54189: [emsesp] thermostat(0x10) W mixer(0x20), ?(0x02E1), data: 01 00 00 00 01
2025-05-11T20:49:07.514 TRACE 54190: [emsesp] thermostat(0x10) W mixer(0x20), ?(0x02EB), data: 00
2025-05-11T20:49:07.552 TRACE 54191: [emsesp] thermostat(0x10) B all(0x00), RC300WWmode2(0x031D), data: 00 00 0A 07
```

MichaelDvP, 2025-05-12:
Should work now with a new dev11 build

SMHRambo, 2025-05-13:
Everything works, thank you.