emsesp/EMS-ESP32 #2544: Add entity for storage charging pump status (boiler)

SMHRambo, 2025-05-01:
With the storage charging optimization option, I noticed that the status of the storage charging pump is not recorded.
Therefore, here is the telegram for the storage tank charging pump on the boiler.

The telegram is:
Pump OFF:
boiler(0x08) B all(0x00), UBAMonitorFast(0x18), data: 07 02 72 64 00 00 80 00 00 02 82 7D 00 83 00 FF FF FF 35 48 00 00 FF 02 00 80 00
Pump ON:
boiler(0x08) B all(0x00), UBAMonitorFast(0x18), data: 07 02 72 64 00 00 80 00 00 02 82 7D 00 83 00 FF FF FF 35 48 00 00 FF 03 00 80 00

```json
{
  "type": "entities",
  "Entities": {
    "entities": [
      {
        "id": 0,
        "ram": 0,
        "device_id": 8,
        "type_id": 24,
        "offset": 23,
        "factor": 1,
        "name": "chargingpump",
        "uom": 0,
        "value_type": 0,
        "writeable": false
      }
    ]
  }
}
```
```json
{
  "system": {
    "version": "3.7.3-dev.10",
    "uptime": "000+00:29:43.441",
    "uptimeSec": 1783,
    "platform": "ESP32",
    "cpuType": "ESP32-D0WDQ6",
    "arduino": "Tasmota Arduino v2.0.17",
    "sdk": "4.4.8.240628",
    "freeMem": 150,
    "maxAlloc": 83,
    "freeCaps": 101,
    "usedApp": 1653,
    "freeApp": 331,
    "partition": "app0",
    "flash_chip_size": 4096,
    "resetReason": "Software reset CPU / Software reset CPU",
    "psram": false,
    "model": ""
  },
  "network": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -65,
    "WIFIReconnects": 0,
    "BSSID": "set",
    "TxPowerSetting": 0,
    "staticIP": false,
    "lowBandwidth": false,
    "disableSleep": true,
    "enableMDNS": true,
    "enableCORS": false,
    "APProvisionMode": "disconnected",
    "APSecurity": "wpa2",
    "APSSID": "ems-esp"
  },
  "ntp": {
    "NTPStatus": "connected",
    "enabled": true,
    "server": "10.13.12.15",
    "tzLabel": "Europe/Berlin",
    "timestamp": 1746131229
  },
  "mqtt": {
    "MQTTStatus": "connected",
    "MQTTPublishes": 1468,
    "MQTTQueued": 0,
    "MQTTPublishFails": 0,
    "MQTTReconnects": 0,
    "enabled": true,
    "clientID": "heizung",
    "keepAlive": 60,
    "cleanSession": false,
    "entityFormat": 3,
    "base": "ems-esp",
    "discoveryPrefix": "homeassistant",
    "discoveryType": 0,
    "nestedFormat": 2,
    "haEnabled": true,
    "mqttQos": 1,
    "mqttRetain": true,
    "publishTimeHeartbeat": 60,
    "publishTimeBoiler": 10,
    "publishTimeThermostat": 10,
    "publishTimeSolar": 10,
    "publishTimeMixer": 10,
    "publishTimeWater": 10,
    "publishTimeOther": 60,
    "publishTimeSensor": 10,
    "publishSingle": false,
    "publish2command": false,
    "sendResponse": false
  },
  "syslog": {
    "enabled": false
  },
  "sensor": {
    "temperatureSensors": 1,
    "temperatureSensorReads": 355,
    "temperatureSensorFails": 1
  },
  "api": {
    "APICalls": 370,
    "APIFails": 1
  },
  "bus": {
    "busStatus": "connected",
    "busProtocol": "Buderus",
    "busTelegramsReceived": 2533,
    "busReads": 710,
    "busWrites": 0,
    "busIncompleteTelegrams": 1,
    "busReadsFailed": 1,
    "busWritesFailed": 0,
    "busRxLineQuality": 100,
    "busTxLineQuality": 100
  },
  "settings": {
    "boardProfile": "CUSTOM",
    "locale": "de",
    "txMode": 1,
    "emsBusID": 11,
    "showerTimer": true,
    "showerMinDuration": 180,
    "showerAlert": false,
    "phyType": 0,
    "rxGPIO": 23,
    "txGPIO": 5,
    "dallasGPIO": 14,
    "pbuttonGPIO": 0,
    "ledGPIO": 25,
    "ledType": 0,
    "hideLed": false,
    "noTokenApi": false,
    "readonlyMode": false,
    "fahrenheit": false,
    "dallasParasite": false,
    "boolFormat": 6,
    "boolDashboard": 2,
    "enumFormat": 1,
    "analogEnabled": false,
    "telnetEnabled": false,
    "maxWebLogBuffer": 25,
    "modbusEnabled": false,
    "forceHeatingOff": false,
    "developerMode": true
  },
  "devices": [
    {
      "type": "boiler",
      "name": "Logano GB1*5, Logamatic MC10",
      "deviceID": "0x08",
      "productID": 72,
      "brand": "",
      "version": "03.05",
      "entities": 73,
      "handlersReceived": "0x10 0x11 0x14 0x15 0x1C 0x18 0x19 0x34 0x04",
      "handlersFetched": "0x16 0x33",
      "handlersPending": "0xBF 0xC2 0xC6 0x1A 0x35 0x2E 0x3B"
    },
    {
      "type": "thermostat",
      "name": "RC310/RC3*0, Moduline 3000/1010H, CW400, Sense II, HPC410",
      "deviceID": "0x10",
      "productID": 158,
      "brand": "",
      "version": "74.05",
      "entities": 63,
      "handlersReceived": "0x06 0xA2 0x02BA 0x02BB 0x031D 0x0267 0xBF 0xC0",
      "handlersFetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0291 0x0292 0x0293 0x0294 0x02F5 0x02F6 0x031B 0x023A 0x0240 0x0269",
      "handlersPending": "0xA3 0x12 0x13 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02BC 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CC 0x0467 0x02CE 0x0468 0x02D0 0x0469 0x02D2 0x046A 0x031E 0xBB 0x023E 0x016E"
    },
    {
      "type": "mixer",
      "name": "MM50/MM50",
      "deviceID": "0x20",
      "productID": 159,
      "brand": "",
      "version": "20.03",
      "entities": 7,
      "handlersReceived": "0x02D7",
      "handlersFetched": "0x02CD",
      "handlersIgnored": "0x02E1 0x02EB 0x0255 0x16 0x33 0xBF"
    },
    {
      "type": "controller",
      "name": "BC10, RFM20",
      "deviceID": "0x09",
      "productID": 68,
      "brand": "",
      "version": "02.03",
      "entities": 0
    },
    {
      "type": "temperaturesensor",
      "name": "temperaturesensor",
      "entities": 1
    },
    {
      "type": "custom",
      "name": "custom",
      "entities": 2
    }
  ]
}
```

MichaelDvP, 2025-05-06:
Is this entity realy needed? It's just the signal of the pump relay and we already have `tapwater_active`, `wwActive` and `wwCharging`, all redundant indicating nearly the same.

SMHRambo, 2025-05-06:
I have created a diagram to show the differences.
It would be very useful for checking exactly whether the pump is running or not. It's also not bad for analyzing how the heating is working, so you can display burner running time versus pump running time.

In my case, the boiler.dhw.active entry does not change at all.

The boiler.dhw.charging and boiler.dhw.recharging entries, on the other hand, show which part of the charging process we are in.

In my case, the pump status changes 1 minute after the charging state entry. Unfortunately, I do not yet know whether this is related to ioBroker or the telegram.

[image]

[image]

SMHRambo, 2025-05-13:
Everything works, thank you.

closed 2025-05-13 by SMHRambo