emsesp/EMS-ESP32 #2548: `auxheateronly` not working with Bosch Compress 6800i AW

MarkusMit, 2025-05-04:
### DESCRIPTION

Tried to change the state of the `auxheateronly` entity for my Bosch Compress 6800i AW heatpump (CS6800i), but it always remains in its original state, or flips back to the original state, when trying to change the value via Home Assistant.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT) -- _No account_
- [x] Provide the System information in the area below, taken from `http://<IP>/api/system`

[System information.json](https://github.com/user-attachments/files/20026845/emsesp-system.json)

### TO REPRODUCE

On EMS-ESP UI, select device _"XCU_THH/CS*800i, Logatherm WLW*"_, find entity _"nur Zusatzheizer"_ (`auxheateronly`), change value from `aus` (off) to `an` (on).
UI says "Befehl schreiben wurde gesendet" (command write was sent) but value remains `aus` (off), both on EMS-ESP and on heatpump.

### EXPECTED BEHAVIOUR

When changing the value of `auxheateronly`, the new value should become effective on the heatpump.

### SCREENSHOTS

_If applicable, add screenshots to help explain your issue._

### ADDITIONAL CONTEXT

To reduce number of starts and mileage of the heatpump's compressor, I want to only use its electrical auxiliary heater for hot water heating during summer time using PV/solar surplus energy.

#### Workaround Custom Entity

When changing the according setting directly on the heatpump's display in the service menu (Anlageneinstellungen / Zuheizer / Einzelbetrieb) the following entries appear in the log ([Full Log](https://github.com/user-attachments/files/20026890/ems-esp-log-additionalheater.txt)):
```
2025-05-04 10:43:21.341 TRACE 31: [emsesp] thermostat(0x10) W boiler(0x08), HPAdditionalHeater(0x0491), data: 01 (offset 3)
2025-05-04 10:43:21.399 TRACE 32: [emsesp] boiler(0x08) B all(0x00), HPAdditionalHeater(0x0491), data: 01 (offset 3)

2025-05-04 10:43:24.988 TRACE 46: [emsesp] thermostat(0x10) W boiler(0x08), HPAdditionalHeater(0x0491), data: 00 (offset 3)
2025-05-04 10:43:26.497 TRACE 55: [emsesp] boiler(0x08) B all(0x00), HPAdditionalHeater(0x0491), data: 00 (offset 3)
```
 [image]

Created a custom entity with this information, and with this custom entity turning on/off the value works:
 [image]

(Helpful discussion in this context: [Bosch/Buderus/Junkers… Heizungs-Integration in HA über KM200 oder EMS-EPS - simon64.com](https://community.simon42.com/t/bosch-buderus-junkers-heizungs-integration-in-ha-ueber-km200-oder-ems-eps/8954/106?page=6))

MarkusMit, 2025-05-04:
on a side note: Changing the language of the EMS-ESP UI seems not to be working with `3.7.3-dev.10`. (Language for entities is German.)
Tried to switch UI language to English, but it just remains on "DE":
[image]

MichaelDvP, 2025-05-04:
We have `auxiliary heater only` in 0x0491 offset 1, see https://github.com/emsesp/EMS-ESP32/issues/802#issuecomment-1348123501
What is diffferent with this setting (offset 3)? 
Please suggest a name.

MarkusMit, 2025-05-04:
Currently the auxiliary heater is heating. Will investigate on differences with offset 1 value once it has finished.

The name in the device's settings is "Stand-alone mode" (DE: "Einzelbetrieb"), but "auxheaterstandalone" is probably a bit too long... 
Other options: "auxheater_alone", "auxheater_only" ?

MarkusMit, 2025-05-04:
To me it looks like the CS6800i does not accept data `01` the value at offset 1 ([Full Log](https://github.com/user-attachments/files/20027892/ems-esp-log-auxheateronly.txt)):
```
2025-05-04 15:35:06.477 INFO 623: [command] Called command boiler/auxheateronly (nur Zusatzheizer) with value an
2025-05-04 15:35:06.546 TRACE 624: [emsesp] me(0x0B) W boiler(0x08), HPAdditionalHeater(0x0491), data: 01 (offset 1)
2025-05-04 15:35:06.820 TRACE 625: [emsesp] me(0x0B) R boiler(0x08), HPAdditionalHeater(0x0491), length: 1 (offset 1)
2025-05-04 15:35:06.830 TRACE 626: [emsesp] boiler(0x08) W me(0x0B), HPAdditionalHeater(0x0491), data: <empty> (offset 1)
...
2025-05-04 15:35:39.431 INFO 662: [command] Called command system/read (send read request) with value 8 0491
2025-05-04 15:35:39.681 NOTICE 663: [emsesp] boiler(0x08) W me(0x0B), HPAdditionalHeater(0x0491), data: 01 00 00 00 00 FB 00 00 00 00 00 00 00 00 14 14 00 3C
```

Changes to offset 3 change the "Stand-alone mode" setting ([Full Log](https://github.com/user-attachments/files/20027909/ems-esp-log-custom_auxheater_only.txt)): 
```
2025-05-04 15:48:05.312 INFO 758: [command] Called command custom/auxheater_only (Sende eigene Entitäten zu EMS) with value an
2025-05-04 15:48:05.374 TRACE 759: [emsesp] me(0x0B) W boiler(0x08), HPAdditionalHeater(0x0491), data: 01 (offset 3)
2025-05-04 15:48:05.395 TRACE 760: [emsesp] boiler(0x08) B all(0x00), HPAdditionalHeater(0x0491), data: 01 (offset 3)
...
2025-05-04 15:48:28.160 INFO 786: [command] Called command system/read (send read request) with value 8 0491
2025-05-04 15:48:28.426 NOTICE 787: [emsesp] boiler(0x08) W me(0x0B), HPAdditionalHeater(0x0491), data: 01 00 00 01 00 FB 00 00 00 00 00 00 00 00 14 14 00 3C
```

MichaelDvP, 2025-05-09:
Seems the Compress x800i series are different to other bosch heatpumps (see #2284 , this, and #2549) , we need a internal flag to separate. Then we can use offset 3 for x800i and offset 1 for other heatpumps. Also change the enums for maxheat if we know the ems-values for the new modes.

MarkusMit, 2025-05-10:
Thanks Michael! I now added some information to both of the other issues, HTH.

MarkusMit, 2025-05-10:
FYI: Created another (potential) CS6800i specific issue:
- #2552

MichaelDvP, 2025-05-10:
Please try dev11 from https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest

MarkusMit, 2025-05-10:
> Please try dev11 from https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest

Unfortunately lots (relevant) of entities became unavailable with that version:
[image]

MichaelDvP, 2025-05-11:
Sorry, typo in the new boiler classes, should be fixed in a updated bin in my repo.

MarkusMit, 2025-05-11:
This version looks good. 
The `auxheateronly` entity seems to working now, and also the `*maxheat*` enums contain the right options.

Thanks a lot!

MarkusMit, 2025-06-15:
@MichaelDvP - did your changes make it to the `EMS-ESP32` repository already?

@proddy - Michael provided a dev-version which fixed this, as well as #2549. When the changes where taken over from Michael's fork, then this two issues would be "closed". 

Regarding the disappearing values described in #2284 - if the are related to the originally described issue by `tomatensuppe` - then yes, that issue should be reopened I guess. Thanks for asking. (can't comment on closed issues.)

proddy, 2025-06-16:
I think this one and 2549 are in the latest official dev release.