emsesp/EMS-ESP32 #2549: Enums different for `*maxheat*` entities with Bosch Compress 6800i AW

MarkusMit, 2025-05-04:
### DESCRIPTION

It seems the _Bosch Compress 6800i AW_ heatpump (CS6800i) has less enum options available for the auxiliary heater (Zusatzheizer/Heizstab) limits:
- `maxheatcomp` - `0 kW | 3 kW | 6 kW | 9 kW`
- `maxheatheat` - `3 kW | 6 kW | 9 kW`
- `dhw/maxheat` - `3 kW | 6 kW | 9 kW`

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT) -- No account
- [x] Provide the System information in the area below, taken from `http://<IP>/api/system`

[System information.json](https://github.com/user-attachments/files/20026845/emsesp-system.json)

### TO REPRODUCE

Selecting a value which is not supported by CS6800i has no effect, previously selected value remains.
(Gives the impression that the entity is not working properly.)

- `maxheatcomp` (Heizstab Limit mit Kompressor)
  - `0 kW | 3 kW | 6 kW | 9 kW` - CS6800i (Begrenzung mit Kompressor)
  - `0 kW | 2 kW | 3 kW | 4 kW | 6 kW | 9 kW` - [ems-esp](https://docs.emsesp.org/Modbus-Entity-Registers/#cs800i-logatherm-wlw)
- `maxheatheat` (Heizstab Limit Leistung)
  - `3 kW | 6 kW | 9 kW` - CS6800i (Leistung Zuheizer begrenzen)
  - `0 kW | 2 kW | 3 kW | 4 kW | 6 kW | 9 kW` - ems-esp
- `dhw/maxheat` (Heizstab Limit für WW)
  - `3 kW | 6 kW | 9 kW` - CS6800i (Leistung Warmwasser-Betrieb begrenzen)
  - `0 kW | 2 kW | 3 kW | 4 kW | 6 kW | 9 kW` - ems-esp

### EXPECTED BEHAVIOUR

ems-esp should only give a choice of supported values.

### SCREENSHOTS

[image]

CS6800i - Limit auxiliary heater with compressor (Begrenzung mit Kompressor), `maxheatcomp`
[image]

CS6800i - Limit auxiliary heater with compressor (Leistung Zuheizer begrenzen), `maxheatheat`
[image]

CS6800i - Limit auxiliary heater hot water mode (Leistung Warmwasser-Betrieb begrenzen), `dhw/maxheat`
[image]

### ADDITIONAL CONTEXT

_Add any other context about the issue here._

MichaelDvP, 2025-05-06:
Are the values in telegram 0x492 also changed (e.g. dhw/maxheat 0=3kW, 1=6kW, 2=9kW) or is it only a limited selection in the UI (e.g. 2= 3kW, 4=6kW, etc.)?

MarkusMit, 2025-05-09:
The values are the same. 
When selecting a supported value in ems-esp, then the selected limit is displayed on the device, and vice versa.

MarkusMit, 2025-05-10:
For confirmation, I now logged iterating through the various settings in the device's settings menue:
```
[command] Called command system/read (send read request) with value 8 492
[emsesp] boiler(0x08) W me(0x0B), HpHeaterConfig(0x0492), data: 04 00 05 05 05
```

CS6800i - Limit auxiliary heater with compressor (Begrenzung mit Kompressor), `maxheatcomp`:
```
[emsesp] thermostat(0x10) W boiler(0x08), HpHeaterConfig(0x0492), data: 00 (offset 2)
[emsesp] boiler(0x08) B all(0x00), HpHeaterConfig(0x0492), data: 00 (offset 2)
[emsesp] thermostat(0x10) W boiler(0x08), HpHeaterConfig(0x0492), data: 02 (offset 2)
[emsesp] boiler(0x08) B all(0x00), HpHeaterConfig(0x0492), data: 02 (offset 2)
[emsesp] thermostat(0x10) W boiler(0x08), HpHeaterConfig(0x0492), data: 04 (offset 2)
[emsesp] boiler(0x08) B all(0x00), HpHeaterConfig(0x0492), data: 04 (offset 2)
[emsesp] thermostat(0x10) W boiler(0x08), HpHeaterConfig(0x0492), data: 05 (offset 2)
[emsesp] boiler(0x08) B all(0x00), HpHeaterConfig(0x0492), data: 05 (offset 2)
```

CS6800i - Limit auxiliary heater with compressor (Leistung Zuheizer begrenzen), `maxheatheat`:
```
[emsesp] thermostat(0x10) W boiler(0x08), HpHeaterConfig(0x0492), data: 02 (offset 3)
[emsesp] boiler(0x08) B all(0x00), HpHeaterConfig(0x0492), data: 02 (offset 3)
[emsesp] thermostat(0x10) W boiler(0x08), HpHeaterConfig(0x0492), data: 04 (offset 3)
[emsesp] boiler(0x08) B all(0x00), HpHeaterConfig(0x0492), data: 04 (offset 3)
[emsesp] thermostat(0x10) W boiler(0x08), HpHeaterConfig(0x0492), data: 05 (offset 3)
[emsesp] boiler(0x08) B all(0x00), HpHeaterConfig(0x0492), data: 05 (offset 3)
```

CS6800i - Limit auxiliary heater hot water mode (Leistung Warmwasser-Betrieb begrenzen), `dhw/maxheat`:
```
[emsesp] thermostat(0x10) W boiler(0x08), HpHeaterConfig(0x0492), data: 02 (offset 4)
[emsesp] boiler(0x08) B all(0x00), HpHeaterConfig(0x0492), data: 02 (offset 4)
[emsesp] thermostat(0x10) W boiler(0x08), HpHeaterConfig(0x0492), data: 04 (offset 4)
[emsesp] boiler(0x08) B all(0x00), HpHeaterConfig(0x0492), data: 04 (offset 4)
[emsesp] thermostat(0x10) W boiler(0x08), HpHeaterConfig(0x0492), data: 05 (offset 4)
[emsesp] boiler(0x08) B all(0x00), HpHeaterConfig(0x0492), data: 05 (offset 4)
```

closed 2025-06-15 by github-actions[bot]

proddy, 2025-06-15:
@MarkusMit let us know if we should re-open and look into this again