emsesp/EMS-ESP32 #2551: Add Option to Configure MQTT Discovery Entities as Optimistic

bsergei, 2025-05-06:
### DESCRIPTION

Please consider adding an option to enable optimistic mode for Home Assistant's MQTT discovery entities generated by EMS-ESP. Especially, for Climate entities.

In Home Assistant, optimistic mode makes the interface immediately reflect changes (like temperature setpoints) without waiting for confirmation from the device. This improves responsiveness, especially for thermostat/climate controls where even a few seconds delay is noticeable.

Please note that the MQTT publishing interval setting in EMS-ESP does not affect this delay. It seems that EMS-ESP waits for the EMS device to respond internally before publishing the updated state to MQTT, which introduces latency.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [x] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
  Paste System information here....

```

### TO REPRODUCE

1. Integrate EMS-ESP with Home Assistant via MQTT discovery.
2. Change a value in the Home Assistant UI, such as the thermostat setpoint.
3. Notice the delay before the value updates in the UI (due to waiting for MQTT state topic to reflect the change).

### EXPECTED BEHAVIOUR

Option to configure MQTT discovery entities (e.g., climate) to be optimistic, so Home Assistant immediately reflects the change after publishing the command.

### SCREENSHOTS

### ADDITIONAL CONTEXT

Some users prefer immediate feedback in Home Assistant when issuing commands. Optimistic mode is already supported by HA for MQTT entities, but currently cannot be enabled when using EMS-ESP’s auto-discovered devices.

In Home Assistant, optimistic mode for MQTT means that HA assumes the command was successful as soon as it publishes the MQTT message, without waiting for a state confirmation from the device. This improves responsiveness (no delay), but can lead to desynchronization if the command fails or is ignored.

See also: [Climate MQTT Optimistic Mode](https://www.home-assistant.io/integrations/climate.mqtt/#optimistic-mode).

**(Please remember to close the issue when it has been addressed)**

proddy, 2025-05-16:
@bsergei could you try it out yourself? Use MQTT Explorer and modify for example the topic `homeassistant/climate/ems-esp/thermostat_hc1/config` adding `optimistic: true` and then testing in HA if it gives you the response you're looking for.

MichaelDvP, 2025-05-16:
In the last PR i've added the optimistic option (checkbox) to mqtt-settings that set optimistic=true to HA-config of all writable entities. Needs to be tested.

bsergei, 2025-05-16:
Thank you! Let me test this.
BTW, Download button for Dev builds is not correct: 

[image]

bsergei, 2025-05-16:
Unfortunately, the fix doesn't seem to be working as expected.
Here's what I observed when changing the temperature in Home Assistant:

- The climate entity appears to be mapped to the ems_esp/thermostat_data/hc2/seltemp attribute.
- When I change the target temperature in Home Assistant (e.g., from 21.0°C to 21.5°C), EMS-ESP initially publishes a message to ems_esp/thermostat_data, but hc2.seltemp still shows the old value (21.0).
- However, I noticed that hc2.manualtemp is updated to 21.5 at this point in about 1-2 seconds.
- Only after about 30 seconds, another message is published where hc2.seltemp is finally updated to 21.5 — and at that moment, Home Assistant GUI reflects the new setpoint.

It looks like Home Assistant still waits for the hc2.seltemp field to change before updating the UI, and this happens with a significant delay.

If need, I can send screen recordings (Github doesn't allow to attach mp4).

[image]

proddy, 2025-05-16:
> Thank you! Let me test this. BTW, Download button for Dev builds is not correct:

We'll look into that. The download button appears when the WebUI is unable to reach the internet to fetch the latest versions, if for example outbound internet traffic is blocked from that wifi network.

proddy, 2025-05-16:
> In mqttExplorer the HA setting has no effect. mqtt will update when ems-esp send a new value, this is when it is read from the bus. If you set seltemp in HA, it should take the value and do not wait for mqtt.
> 
> When i check the settings i get for hc2-seltemp config with HA and optimistic enabled:
> 
> ```
> {
>   "uniq_id": "e32v2_thermostat_hc2_seltemp",
>   "obj_id": "e32v2_thermostat_hc2_seltemp",
>   "cmd_t": "e32v2/thermostat/hc2/seltemp",
>   "mode": "box",
>   "step": 0.5,
>   "min": 0,
>   "max": 30,
>   "name": "hc2 Selected room temperature",
>   "stat_t": "e32v2/thermostat_data",
>   "val_tpl": "{{value_json['hc2']['seltemp'] if value_json.hc2 is defined and value_json['hc2']['seltemp'] is defined else 0}}",
>   "avty": [
>     {
>       "t": "e32v2/thermostat_data",
>       "val_tpl": "{{'online' if value_json.hc2 is defined and value_json['hc2']['seltemp'] is defined else 'offline'}}"
>     }
>   ],
>   "avty_mode": "all",
>   "unit_of_meas": "°C",
>   "stat_cla": "measurement",
>   "dev_cla": "temperature",
>   "dev": {
>     "ids": [
>       "e32v2-thermostat"
>     ],
>     "name": "e32v2 Thermostat"
>   },
>   "optimistic": true
> } 
> ```

From what I read the `optimistic` attribute is only for the HVAC component in HA  (i.e. `climate`) which have *_state_topic attributes. So not for individual element topics like `seltemp`

MichaelDvP, 2025-05-16:
Climate thermostat_hc2 has also optimistic set: 
```
{
"~": "e32v2",
  "uniq_id": "thermostat_hc2",
  "obj_id": "thermostat_hc2",
  "name": "Hc2",
  "mode_stat_t": "~/thermostat_data",
  "mode_stat_tpl": "{%if value_json.hc2 is undefined or value_json.hc2.mode is undefined%}off{%elif value_json.hc2.mode=='manual'%}heat{%elif value_json.hc2.mode=='day'%}heat{%elif value_json.hc2.mode=='night'%}off{%elif value_json.hc2.mode=='off'%}off{%else%}auto{%endif%}",
  "temp_cmd_t": "~/thermostat/hc2/seltemp",
  "temp_stat_t": "~/thermostat_data",
  "temp_stat_tpl": "{{value_json.hc2.seltemp if value_json.hc2 is defined and value_json.hc2.seltemp is defined else 0}}",
  "min_temp": "5",
  "max_temp": "30",
  "temp_step": "0.5",
  "mode_cmd_t": "~/thermostat/hc2/mode",
  "modes": [
    "auto",
    "heat",
    "off"
  ],
  "dev": {
    "name": "e32v2 Thermostat",
    "ids": [
      "e32v2-thermostat"
    ]
  },
  "avty": [
    {
      "t": "~/thermostat_data",
      "val_tpl": "{{'online' if value_json.hc2 is defined and value_json.hc2.seltemp is defined else 'offline'}}"
    },
    {
      "t": "~/thermostat_data",
      "val_tpl": "{{'offline' if value_json.hc2 is undefined or value_json.hc2.mode is undefined else 'online'}}"
    }
  ],
  "avty_mode": "all",
  "optimistic": true
}
```

bsergei, 2025-05-16:
Sorry for confusion. Actually, topic is ems_esp/thermostat_data which holds complex state object. hc2, seltemp are attributes within this complex state object. I think the problem is when this complex object is updating twice: with old seltemp value and with new value. So, optimistic option is not working in this case. I believe, it will work if ems-esp will not send first update with manualtemp. However, I don't think this a good workaround.

MichaelDvP, 2025-05-16:
Maybe, i've changed https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest dev 11 to skip stat_t instead of addding optimistic and only for climate. Please try.

bsergei, 2025-05-16:
> Maybe, i've changed https://github.com/MichaelDvP/EMS-ESP32/releases/tag/latest dev 11 to skip stat_t instead of addding optimistic and only for climate. Please try.

Sorry, I have not noticed any difference.

MichaelDvP, 2025-05-17:
> Sorry, I have not noticed any difference.

Strange, without state topic the climate widget can never update from mqtt, so changes of mode an seltemp made by thermostat or ems-esp webui are not reflected. But this is what the optimistic mode is. You will not see any difference in mqtt, only in climate widget.

proddy, 2025-05-17:
I did some checking. HA's `optimistic` is only supported on MQTT select, switch, number and climate. Not sensor, binary_sensor or text, which makes sense. 

@bsergei could you explain the actual use case, highlighting what you're changing, how and what you're seeing? On my system, without optimistic set I can go into HA and change, for example the `selected room temperature` and that field in HA is instantly updated. The HVAC climate widget, however, does take less than a second to refresh, but that's normal, as it's reading the actual value from the thermostat, and the EMS bus is not super fast. 

[image]

proddy, 2025-05-17:
> > Thank you! Let me test this. BTW, Download button for Dev builds is not correct:
> 
> We'll look into that. The download button appears when the WebUI is unable to reach the internet to fetch the latest versions, if for example outbound internet traffic is blocked from that wifi network.

the download button was removed a long time ago. I think you were using a very old dev version or the browser cache needed refreshed.

proddy, 2025-05-17:
> Do you see a difference when optimistic is set?

no difference at all. We need a detailed use case from @bsergei and even better a video to explain what is happening. I don't see any lag on my setup.

proddy, 2025-05-22:
@MichaelDvP I don't see adding this Optimistic making any difference. I think we should remove it.

closed 2025-06-06 by proddy