emsesp/EMS-ESP32 #2552: "Extra Hot Water" with Bosch Compress 6800i AW

MarkusMit, 2025-05-10:
### DESCRIPTION

Starting the one-time hot water charging with the existing entity `dhw/onetime` does not work with the _Bosch Compress 6800i AW_ heatpump. It seems this is just ignored by the CS6800i.

It seems EMS-ESP needs a new entity/command to allow starting of "Extra Hot Water" period (aka "One Time Charging" ?) for the CS6800i.
"Extra Hot Water" is a feature of this heatpump, where it tries for a pre-configured time (dhw/chargeduration) to charge the water reservoir to a pre-configured temperature (dhw/seltempsingle).

Activating this mode in the Bosch _HomeCom_ app sends the following commands on the bus ([full log](https://github.com/user-attachments/files/20139308/ems-esp-log-extra-hot-water.txt)):
```
TRACE 52: [emsesp] gateway(0x48) W thermostat(0x10), RC300WWmode(0x02F5), data: 04 (offset 10)
TRACE 56: [emsesp] gateway(0x48) W thermostat(0x10), RC300WWmode(0x02F5), data: FF (offset 11)
```
The first row seems to be re-sending the duration (4 quaters of an hour), and the second one enables the hotwater mode.
In another attempt, where I could've changed the time (3 hrs) in the app before activating, I've also seen it like this:
```
TRACE 504: [emsesp] gateway(0x48) W thermostat(0x10), RC300WWmode(0x02F5), data: 0C FF (offset 10)
```

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT) -- No account
- [x] Provide the System information in the area below, taken from `http://<IP>/api/system`

[System information.json](https://github.com/user-attachments/files/20026845/emsesp-system.json)

### TO REPRODUCE

On EMS-ESP UI, select device _"XCU_THH/CS*800i, Logatherm WLW*"_, find entity _"Einmalladung"_ (`dhw/onetime`), change value from `aus` (off) to `an` (on).
The log says _"Called command boiler/onetime (Einmalladung) with value an and id 9 on device 0x08"_ but the _Extra Hot Water_ mode is not acitvated.

### EXPECTED BEHAVIOUR

_Extra Hot Water_ mode should be activated.

#### Workaround

Created a custom entity "extra_hot_water" for `0x10 0x02F5 offset 11` (see below).
Setting `extra_hot_water` to `FF` enables the hot water mode, setting it to `00` disables it again.

### SCREENSHOTS

[image]

MichaelDvP, 2025-05-10:
2F5, offset 10 is `dhw/chargeduration`, offset 11`dhw/charge`

MarkusMit, 2025-05-10:
Strange. I would've sworn that `dhw/charge` was non-writable when I checked it in the morning.
But you're right. `dhw/charge` works. Thanks!

closed 2025-05-10 by MarkusMit