emsesp/EMS-ESP32 #2555: Gateway E32 V2 (Ethernet + WiFi Edition V2) freezing

arieger2, 2025-05-15:
### DESCRIPTION

The device is connected to the ethernet and collecting data from EMS-bus (Bosch). MQTT is used to send data to homeassistant. Homeassistant is running on a vm in a Proxmox environment.
If Proxmox is rebooting (homeassistant gets down), EMS-ESP is freezing after homeassistant is back again.
If only homeassistant is rebooting, EMS-ESP survives.

It seems there is an overflow if MQTT cannot send data for a longer period 

After power cycle of the Gateway E32 V2 everything works again.

bbqkees, 2025-05-15:
Are you on the latest firmware?
Can you add a copy of the settings file/log here.

proddy, 2025-05-16:
See https://docs.emsesp.org/Support/ on how to provide support info.

Which MQTT server/broker are you using? And does it also run as a VM or LXC in Proxmox?

What do you mean by "freeze"? Is the EMS-ESP not showing any LEDs, can you reach the console, the webUI?