emsesp/EMS-ESP32 #2567: Junkers boiler in flow mode - cannot turn off dhw

irsl, 2025-05-25:
### DESCRIPTION

I've an old Junkers boiler (Zwe 28-5 mfa). It has no buffer tank, dhw is made in flow mode. The `dhw turn on/off` function does not work: the write operation succeeds, the change is reflected, but the boiler keeps using the burner to make hot water despite the setting.

Is it possible to turn off the dhw feature?

### REQUESTED INFORMATION

- [x] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [x] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [x] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [x] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
{"system":{"version":"3.7.3-dev.11","uptime":"000+00:00:34.445","uptimeSec":34,"platform":"ESP32S3","cpuType":"ESP32-S3","arduino":"ESP32 Arduino v2.0.17","sdk":"v4.4.7-dirty","freeMem":202,"maxAlloc":179,"freeCaps":8359,"usedApp":2157,"freeApp":2515,"partition":"app1","flash_chip_size":16384,"resetReason":"Software reset CPU / Software reset CPU","psram":true,"psramSize":8189,"freePsram":8158,"model":"","temperature":45.1},"network":{"network":"WiFi","hostname":"ems-esp","RSSI":-60,"WIFIReconnects":0,"TxPowerSetting":0,"staticIP":false,"lowBandwidth":false,"disableSleep":true,"enableMDNS":true,"enableCORS":false,"APProvisionMode":"disconnected","APSecurity":"wpa2","APSSID":"ems-esp"},"ntp":{"NTPStatus":"disconnected","enabled":false,"server":"time.google.com","tzLabel":"Europe/Amsterdam","timestamp":97729},"mqtt":{"MQTTStatus":"connected","MQTTPublishes":90,"MQTTQueued":0,"MQTTPublishFails":0,"MQTTReconnects":0,"enabled":true,"clientID":"esp32-e927843c","keepAlive":60,"cleanSession":false,"entityFormat":0,"base":"ems-esp","discoveryPrefix":"homeassistant","discoveryType":0,"nestedFormat":1,"haEnabled":true,"mqttQos":0,"mqttRetain":false,"publishTimeHeartbeat":60,"publishTimeBoiler":10,"publishTimeThermostat":10,"publishTimeSolar":10,"publishTimeMixer":10,"publishTimeWater":10,"publishTimeOther":60,"publishTimeSensor":10,"publishSingle":false,"publish2command":false,"sendResponse":false},"syslog":{"enabled":false},"sensor":{"temperatureSensors":0,"temperatureSensorReads":0,"temperatureSensorFails":0,"analogSensors":0,"analogSensorReads":0,"analogSensorFails":0},"api":{"APICalls":0,"APIFails":0},"bus":{"busStatus":"connected","busProtocol":"HT3","busTelegramsReceived":47,"busReads":20,"busWrites":0,"busIncompleteTelegrams":0,"busReadsFailed":0,"busWritesFailed":0,"busRxLineQuality":100,"busTxLineQuality":100},"settings":{"boardProfile":"S32S3","locale":"en","txMode":1,"emsBusID":11,"showerTimer":true,"showerMinDuration":180,"showerAlert":false,"hideLed":false,"noTokenApi":false,"readonlyMode":false,"fahrenheit":false,"dallasParasite":false,"boolFormat":1,"boolDashboard":1,"enumFormat":1,"analogEnabled":true,"telnetEnabled":true,"maxWebLogBuffer":50,"modbusEnabled":false,"forceHeatingOff":false,"developerMode":false},"devices":[{"type":"boiler","name":"Condens, Logamax/Logomatic, Cerapur Top, Greenstar, Generic HT3","deviceID":"0x08","productID":95,"brand":"","version":"15.02","entities":61,"handlersReceived":"0x10 0x11 0x15 0x1C 0x18 0x34 0x04","handlersFetched":"0x14 0x16 0x33","handlersPending":"0xBF 0xC2 0xC6 0x19 0x1A 0x35 0x2A 0x2E 0x3B"},{"type":"controller","name":"HT3","deviceID":"0x09","productID":95,"brand":"","version":"15.02","entities":0}]}
```

### TO REPRODUCE

Set `dhw turn on/off` to off and then consume some hot water. Notice the boiler ignores the setting, the burner is started and hot water is made, still.

### EXPECTED BEHAVIOUR

When the `dhw turn on/off`  setting is set to off, the boiler should not make hot water at all.

### ADDITIONAL CONTEXT

This is what I see in the system log when I send the turn off command:

```
000+00:00:15.813 I 16: [emsesp] Detected EMS device: boiler (0x08)
000+00:00:15.938 I 17: [emsesp] Detected EMS device: controller (0x09)
000+00:04:10.144 T 18: [emsesp] boiler(0x08) B all(0x00), UBAMonitorFast(0x18), data: 30 00 D4 64 00 00 00 40 40 00 DB 80 00 80 00 FF FF FF 00 00 00 00 00 00 20
000+00:04:10.382 T 19: [emsesp] boiler(0x08) B all(0x00), UBAMonitorWW(0x34), data: 28 00 DB 80 00 81 00 00 01 00 01 A4 41 01 F5 3A 00
000+00:04:18.616 I 20: [command] Called command boiler/tapactivated (turn on/off) with value off and id 9 on device 0x08
000+00:04:18.760 T 21: [emsesp] me(0x0B) W boiler(0x08), ?(0x1D), data: 5A 00 00 64 FF 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+00:04:20.111 T 22: [emsesp] boiler(0x08) B all(0x00), UBAMonitorFast(0x18), data: 30 00 D6 64 00 00 00 40 40 00 DB 80 00 80 00 FF FF FF 00 00 00 00 00 00 20
000+00:04:20.351 T 23: [emsesp] boiler(0x08) B all(0x00), UBAMonitorWW(0x34), data: 28 00 DB 80 00 81 00 00 01 00 01 A4 41 01 F5 3A 00
```

MichaelDvP, 2025-05-25:
`boiler/dhw/tapactivated` is a very special entity, used for shower coldshot function: It puts the boiler in testmode and switches burner off while tapwater is running.

What you want to change is `boiler/dhw/activated`, but in some systems this is overwritten by the controller and a change via bus is not possible. The easiest way is to set `boiler/dhw/seltemp` to a low value, eg. 10°C.  

Maybe you can switch off with a temporary command to 0x35, offset 1 set to 0x00 annd repeat it every minute. Create a custom entity with boiler telegram0x35, offset 1, UINT8 or BOOL with mask 0x01. This is what my RC35 sends to set dhw to day/night mode.

irsl, 2025-05-25:
I've tried setting the temperature to a low value: the write succeded, but the command was ignored (the entity stayed on the original value 40).

Thanks for the other hint, I'll give it a try. Is this to be sent to the controller (device id 0x9) or the boiler (device id 0x8)?

MichaelDvP, 2025-05-26:
If the boiler have a dial for dhw temperature, try setting this to max and test with different values for boiler/dhw/seltemp. Maybe 10°C is below the allowed range. The seltemp command should work permanent, no need to repeat it.

Telegram 0x35 is a boiler telegram, send it to 0x08.

closed 2025-07-01 by github-actions[bot]