emsesp/EMS-ESP32 #2576: (non)updating emsesp_schedule.json file

SoftVienka, 2025-06-02:
### DESCRIPTION

When I disable some scheduler items from HA using automation, after restarting EMS-ESP these items are enabled again.
I found that on the scheduler board these items are disabled (red), but when I download the scheduler file emsesp_schedule.json these items are ' "active": true '.
When restarting EMS-ESP the scheduler items are probably loaded from this file.
When disabling/enabling on the scheduler board ' active: ' items in emsesp_schedule.json are updated correctly.

### REQUESTED INFORMATION

_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Searched the issue in [issues](https://github.com/emsesp/EMS-ESP32/issues)
- [ ] Searched the issue in [discussions](https://github.com/emsesp/EMS-ESP32/discussions)
- [ ] Searched the issue in the [docs](https://docs.emsesp.org/Troubleshooting/)
- [ ] Searched the issue in the [chat](https://discord.gg/3J3GgnzpyT)
- [ ] Provide the System information in the area below, taken from `http://<IP>/api/system`

```json
{
  "system": {
    "version": "3.7.3-dev.10",
    "uptime": "000+02:25:48.243",
    "uptimeSec": 8748,
    "platform": "ESP32S3",
    "cpuType": "ESP32-S3",
    "arduino": "ESP32 Arduino v2.0.17",
    "sdk": "v4.4.7-dirty",
    "freeMem": 130,
    "maxAlloc": 101,
    "freeCaps": 8268,
    "usedApp": 2151,
    "freeApp": 3545,
    "partition": "app1",
    "flash_chip_size": 16384,
    "resetReason": "Software reset CPU / Software reset CPU",
    "psram": true,
    "psramSize": 8189,
    "freePsram": 8139,
    "model": "",
    "temperature": 58.3
  },
  "network": {
    "network": "WiFi",
    "hostname": "ems-esp",
    "RSSI": -53,
    "WIFIReconnects": 0,
    "TxPowerSetting": 0,
    "staticIP": true,
    "lowBandwidth": false,
    "disableSleep": true,
    "enableMDNS": true,
    "enableCORS": false,
    "APProvisionMode": "disconnected",
    "APSecurity": "wpa2",
    "APSSID": "ems-esp"
  },
  "ntp": {
    "NTPStatus": "connected",
    "enabled": true,
    "server": "time.google.com",
    "tzLabel": "Europe/Bratislava",
    "timestamp": 1748891617
  },
  "mqtt": {
    "MQTTStatus": "connected",
    "MQTTPublishes": 5409,
    "MQTTQueued": 0,
    "MQTTPublishFails": 0,
    "MQTTReconnects": 2,
    "enabled": true,
    "clientID": "ems-esp",
    "keepAlive": 60,
    "cleanSession": true,
    "entityFormat": 1,
    "base": "ems-esp",
    "discoveryPrefix": "homeassistant",
    "discoveryType": 0,
    "nestedFormat": 2,
    "haEnabled": true,
    "mqttQos": 2,
    "mqttRetain": true,
    "publishTimeHeartbeat": 240,
    "publishTimeBoiler": 13,
    "publishTimeThermostat": 17,
    "publishTimeSolar": 180,
    "publishTimeMixer": 180,
    "publishTimeWater": 180,
    "publishTimeOther": 30,
    "publishTimeSensor": 11,
    "publishSingle": false,
    "publish2command": false,
    "sendResponse": false
  },
  "syslog": {
    "enabled": false
  },
  "sensor": {
    "temperatureSensors": 5,
    "temperatureSensorReads": 8745,
    "temperatureSensorFails": 0,
    "analogSensors": 1,
    "analogSensorReads": 0,
    "analogSensorFails": 0
  },
  "api": {
    "APICalls": 2160,
    "APIFails": 6
  },
  "bus": {
    "busStatus": "connected",
    "busProtocol": "HT3",
    "busTelegramsReceived": 12396,
    "busReads": 2608,
    "busWrites": 7,
    "busIncompleteTelegrams": 1,
    "busReadsFailed": 0,
    "busWritesFailed": 0,
    "busRxLineQuality": 100,
    "busTxLineQuality": 100
  },
  "settings": {
    "boardProfile": "S32S3",
    "locale": "en",
    "txMode": 2,
    "emsBusID": 11,
    "showerTimer": false,
    "showerMinDuration": 180,
    "showerAlert": false,
    "hideLed": false,
    "noTokenApi": false,
    "readonlyMode": false,
    "fahrenheit": false,
    "dallasParasite": false,
    "boolFormat": 1,
    "boolDashboard": 1,
    "enumFormat": 1,
    "analogEnabled": true,
    "telnetEnabled": true,
    "maxWebLogBuffer": 100,
    "modbusEnabled": false,
    "forceHeatingOff": false,
    "developerMode": true
  },
  "devices": [
    {
      "type": "boiler",
      "name": "ACU-LOP13/Logamax Plus GB122, Condense 2300, Junkers Cerapur GC2200W",
      "deviceID": "0x08",
      "productID": 234,
      "brand": "Bosch",
      "version": "05.06",
      "entities": 84,
      "handlersReceived": "0xBF 0xC2 0x14 0x15 0xD1 0xE3 0xE4 0xE5 0xE9 0x2E 0x04",
      "handlersFetched": "0xE6 0xEA 0x28",
      "handlersPending": "0x10 0x11 0xC6 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0x3B",
      "handlersIgnored": "0xE7 0x02E0 0x02EA 0xC4 0x0DAC"
    },
    {
      "type": "thermostat",
      "name": "CW100/RC200, CW100, CR120, CR50",
      "deviceID": "0x18",
      "productID": 157,
      "brand": "",
      "version": "41.09",
      "entities": 53,
      "handlersReceived": "0x06 0xA2 0x031D 0xBF 0xC0",
      "handlersFetched": "0x02A5 0x02B9 0x02AF 0x029B 0x0291 0x02F5 0x023A 0x0241 0x043F",
      "handlersPending": "0xA3 0x12 0x13 0x0471 0x02A6 0x02BA 0x02B0 0x029C 0x0472 0x02A7 0x02BB 0x02B1 0x029D 0x0473 0x02A8 0x02BC 0x02B2 0x029E 0x0474 0x02A9 0x02BD 0x02B3 0x029F 0x0475 0x02AA 0x02BE 0x02B4 0x02A0 0x0476 0x02AB 0x02BF 0x02B5 0x02A1 0x0477 0x02AC 0x02C0 0x02B6 0x02A2 0x0478 0x02CC 0x0467 0x02CE 0x0468 0x0292 0x02D0 0x0469 0x0293 0x02D2 0x046A 0x0294 0x02F6 0x031B 0x031E 0x0267 0xBB 0x023E 0x016E",
      "handlersIgnored": "0xEA 0xF7"
    },
    {
      "type": "temperaturesensor",
      "name": "temperaturesensor",
      "entities": 5
    },
    {
      "type": "analogsensor",
      "name": "analogsensor",
      "entities": 1
    },
    {
      "type": "scheduler",
      "name": "scheduler",
      "entities": 12
    },
    {
      "type": "custom",
      "name": "custom",
      "entities": 6
    }
  ]
}

```

### TO REPRODUCE

_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR

_A clear and concise description of what you expected to happen._

### SCREENSHOTS

_If applicable, add screenshots to help explain your issue._

### ADDITIONAL CONTEXT

_Add any other context about the issue here._

**(Please remember to close the issue when it has been addressed)**

proddy, 2025-06-04:
Thanks for reporting. We'll try and reproduce and fix it. It seems changes made to the scheduler via the API are not persisted to the JSON file which indeed is used when EMS-ESP reboots.

MichaelDvP, 2025-06-04:
This is intended. The scheduler setting sets the active state on startup. Changing the state by command (API/MQTT) is only in RAM. Flash erase/write cycles are too limited to store every user initiated change. 
For digitial out sensor we have a setting option: off/on/last state. Last state is written to nvs if selected, nvs has better wear leveling for single values, but still critical, especially on 4MB SOC with only a small nvs partition. I don't think we need this for the scheduler.

proddy, 2025-06-04:
Thanks @MichaelDvP . I'll take a look later to see if we need to make this explicit in the documentation.

SoftVienka, 2025-06-05:
Do I understand correctly that every change (in the planner board), confirmed by the "APPLY CHANGES" button, is written to Flash? 
Maybe I misunderstood.

proddy, 2025-06-05:
> Do I understand correctly that every change (in the planner board), confirmed by the "APPLY CHANGES" button, is written to Flash? Maybe I misunderstood.

Yes, this applies to all WebUI pages. Apply Changes updates the configuration files that are stored in the flash partition using the built-in LittleFS filesystem. It's the way we persist all the settings. Wear levelling is done automatically, and the   ESP32 flash can handle 100.000 cycles at minimum so good for many many years,