emsesp/EMS-ESP32 #2581: Crashes with  EMS-ESP-3_7_3-dev_12-ESP32-16MB+.bin

MichaelDvP, 2025-06-13:
I've observed some crashes with dev12 on a E32V2 gateway., but don't know what causes it.
It happens irregular in average every 3 hours, but sometimes after 20 min, sometimes after 6 h.
Very hard to reproduce and to test, i always have to wait a day.
My E32V2 is for testing and have all enabled, debug, mqtt every 10sec and on change, syslog, and API (on tp1de ioBroker adapter), etc.
I've tested with reverting AsyncTCP, disable debug, disable syslog, enlarge stacks, etc. but does not help.
Dev11 does not have these reboots. 
No API communication (stop ioBroker adapter) seems to fix.
Revert platform to 6.10.0 fixes at last (working 12h now, continuing)

I'll make a PR dev13 with revert to platform 6.10.0, but also set this issue to remember and to get ems-esp stable with new platform.

VlastiBroucek, 2025-06-13:
Just for your info @MichaelDvP, my E32V2 has been running without any issues since dev.12 release... No restart, nothing...

MichaelDvP, 2025-06-13:
Maybe you don't have 2200 api calls per hour. Without the API traffic i also don't see crashes in dev12. 
But i have no crashes in dev11 with api traffic and also in dev13 (with platform 6.10.0).

proddy, 2025-06-13:
Looking at the differences between `platform-espressif32` versions 6.10.0 and 6.11.0 I can't see any changes that could contribute to the crashing (https://github.com/platformio/platform-espressif32/compare/v6.10.0...v6.11.0). Packages are the same:

```
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
```

VlastiBroucek, 2025-06-14:
> Maybe you don't have 2200 api calls per hour. Without the API traffic i also don't see crashes in dev12. But i have no crashes in dev11 with api traffic and also in dev13 (with platform 6.10.0).

God no, of course not.

MichaelDvP, 2025-06-14:
Sad, also with dev13 i've got a crash after 25 hours, and then 20 min later the next one. Needs further investigation, maybe my powersupply gets old or the device is flashed too often. I will test a fewdays with dev11 if it is still stable, if not i will erase chip and usb flash and change powersupply.

MichaelDvP, 2025-06-15:
Seems i am the only one with these crashes. My primary ems-esp (professional, EH-MT, mqtt only) is stable, only the testing gateway with all options enabled shows this. 
@proddy For now i don't think you should spend time with this issue.

MichaelDvP, 2025-06-18:
Ok, i've tested a bit more. The api queries from ioBroker are 9 reads every 15 sec:
```
2025-06-18T09:22:33.405 DEBUG 233: [command] Fetched device entity/attributes for system/values (id=-1)
2025-06-18T09:22:34.624 DEBUG 234: [command] Fetched device entity/attributes for boiler/values (id=-1)
2025-06-18T09:22:34.910 DEBUG 235: [command] Fetched device entity/attributes for thermostat/values (id=-1)
2025-06-18T09:22:34.128 DEBUG 236: [command] Fetched device entity/attributes for mixer/values (id=-1)
2025-06-18T09:22:34.257 DEBUG 237: [command] Fetched device entity/attributes for solar/values (id=-1)
2025-06-18T09:22:35.620 DEBUG 238: [command] Fetched device entity/attributes for temperaturesensor/values (id=-1)
2025-06-18T09:22:35.738 DEBUG 239: [command] Fetched device entity/attributes for analogsensor/values (id=-1)
2025-06-18T09:22:35.858 DEBUG 240: [command] Fetched device entity/attributes for scheduler/values (id=-1)
2025-06-18T09:22:35.978 DEBUG 241: [command] Fetched device entity/attributes for custom/values (id=-1)
```
Disabeling all other IP communication and leave only api active don't have crashes (tested 24h). So there is no fault in api code.
The i've enabled mqtt with every 10 sec:
```
2025-06-18T09:22:35.539 DEBUG 242: [mqtt] Publishing topic 'e32v2/heartbeat', pid 26151
2025-06-18T09:22:36.001 DEBUG 243: [mqtt] Publishing topic 'e32v2/boiler_data', pid 26152
2025-06-18T09:22:36.195 DEBUG 244: [mqtt] Publishing topic 'e32v2/thermostat_data', pid 26153
2025-06-18T09:22:36.321 DEBUG 245: [mqtt] Publishing topic 'e32v2/solar_data', pid 26154
2025-06-18T09:22:36.363 DEBUG 246: [mqtt] Publishing topic 'e32v2/mixer_data', pid 26155
2025-06-18T09:22:36.380 DEBUG 247: [mqtt] Publishing topic 'e32v2/scheduler_data', pid 26156
2025-06-18T09:22:36.380 DEBUG 248: [mqtt] Publishing topic 'e32v2/custom_data', pid 26157
2025-06-18T09:22:36.380 DEBUG 249: [mqtt] Publishing topic 'e32v2/thermostat_switchprog', pid 26158
2025-06-18T09:22:36.380 DEBUG 250: [mqtt] Publishing topic 'e32v2/thermostat_holidays', pid 26159
2025-06-18T09:22:36.410 DEBUG 251: [mqtt] Publishing topic 'e32v2/temperaturesensor_data', pid 26160
2025-06-18T09:22:36.410 DEBUG 252: [mqtt] Publishing topic 'e32v2/analogsensor_data', pid 26161
```
and got a crash after some ten minutes.
I compared to dev 11, which is stable, and rolled AsyncTCP back to 3.4.0. This works now 25h with mqtt/api/syslog(debug)/ntp enabled.  Then tried 3.4.3 and get a crash after afew minutes. 
Continue testing...

MichaelDvP, 2025-06-26:
I've tested with asyncTCP running on core 0 and all other tasks  on core 1. No crashes with 3.4.4. 
I'll open an issue for asyncTCP, but i think it is difficult to reproduce by the maintainers.

mathieucarbou, 2025-06-26:
@MichaelDvP @proddy : 

1. Do you have the stack traces ?
2. Are you able to monitor the free heap ? 
3. Are you able to monitor the high stack watermark for the async_tcp task ?

Move to a discussion here: https://github.com/ESP32Async/AsyncTCP/discussions/75

Also, why core 0 for async_tcp task ? Most callbacks are dealing with user code and states usually handled by core 1.

Core 0 is reserved to Arduino Core system so if you put something on core 0 you need to make sure that it will never block, be fast and does not interact with core 1. 

We advise to run AsyncTCP and EspAsyncWebServer on core 1, especially if used for web server or mqtt which are lower priority services than Arduino ones, and closer to user code.

willmmiles, 2025-06-26:
> > _ZNSt17_Function_handlerIFvPvP11AsyncClientEZN21AsyncWebServerRequestC4EP14AsyncWebServerS2_EUlS0_S2_E1_E9_M_invokeERKSt9_Any_dataOS0_OS2_
> > :?
> > _ZNKSt8functionIFvPvP11AsyncClientEEclES0_S2_
> > :?
> > _ZN11AsyncClient6_closeEv$part$14
> > :?
> > _ZL19_async_service_taskPv
> > :?
> 
> I am having a hard time matching those lines to some actual code in our libs....

[They're "mangled" C++ names.](https://github.com/gcc-mirror/gcc/blob/master/gcc/cp/mangle.cc)

- The bottom stack frame is AsyncTCP.cpp:_async_service_task().  A lot of AsyncTCP's core loop and dispatch functions get inlined here as they're not exposed anywhere else.
- The next one up is AsyncClient::_close() ; probably line 976 (the `std::function` invocation).
- The next one up is `std::function<void<void*,AsyncClient*>>::operator()`.
- The next one up demangles to `std::_Function_handler<void (void*, AsyncClient*), AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*)::{lambda(void*, AsyncClient*)#3}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&)` -- ie. the lambda from `AsyncWebServerRequest::AsyncWebServerRequest()`'s call to `onDisconnect`.
- Above that is a failure to call delete on an object, presumably the AsyncClient itself.

Unfortunately, in this case, the point of failure isn't the point where the mistake was made -- the heap validator has noticed that the memory immediately *prior* to this AsyncClient object has been unexpectedly overwritten.  This indicates a use-after-free somewhere in the code.  AsyncTCP is particularly vulnerable to being the victim for these checks as it does a lot of dynamic allocation.

@mathieucarbou I'll review how we can get to `AsyncClient::_close` from `_async_service_task()`, perhaps there's a clue there.

(Hi @MichaelDvP !  I'm the principal culprit behind the recent AsyncTCP changes.  @mathieucarbou asked me to take a look.)

willmmiles, 2025-06-27:
I'm pretty sure that the call of `AsyncClient::_close()` is getting inlined in to `_async_service_task()` through `AsyncClient::_poll()`, when it detects an rx timeout.  This implies the LwIP event -> AsyncClient pcb match check passed; and it made a call to a real callback, so at least those parts of the AsyncClient object were internally consistent prior to the failure.  The implication is that *this* AsyncClient wasn't (itself) the problem; if there had been a double-free of this object, the pcb pointer wouldn't match as it's set to null during the close operation.

It is curious that a receive timeout was triggered; does that happen regularly in your application, or is that possibly something else that was unexpectedly overwritten?

MichaelDvP, 2025-06-28:
Thanks for your analyze. I've searched the code for rx. I thought it also happend with mqtt-client disabled, but found that mqtt task and mqtt rx checks are still active when mqtt is disabled. So maybe the mqtt-client is not 100% thread safe. But i can't see anything obvious in the code. The crashes are rare, always takes a while to test a change. 
My prvious reported test shows:
-  webserver active, mqtt/syslog/ntp disabled: no crashes: rue, in this case mqtt was really disabled.
- crashes with syslog and/or/ ntp enabled, mqtt disbaled was a  wrong test, there was the mqtt running in background.
Now with really disabled mqtt webserver with syslog and ntp is running since 1 day. 
Another test with mqtt running syncron in the main task runs since 12h without crash.

mathieucarbou, 2025-07-04:
@MichaelDvP : there are too many components in your app that could have impacts other than ESPAsyncWS itself. 
Are you able to disable mqtt completely to test ? If I understand from your tests, the only times it crashes is when you have mqtt activated, using or not AsyncTCP ?

proddy, 2025-07-05:
> @proddy I suggest to use AsyncTCP 3.4.2. so we are safe also on single cores like esp32C3.
Arduino3 takes more flash and more ram, also wifi scan does not work and wifi is not stable with some routers. Newer builds like tasmota 25.04 or 25.05 do not compile or produces bootloops, there are more changes and tests needed.

Thanks @MichaelDvP for the exhaustive effort and time you've put into figuring out why we're seeing crashes. I agree we should rollback to AsyncTCP 3.4.2 until we find a path forward. 

As for Arduino 3/IDF 5, it's disappointing to see that it uses significantly more memory and will require some EMS-ESP code refactoring. I think we should continue exploring the hybrid mode (https://github.com/emsesp/EMS-ESP32/issues/2108#issuecomment-2566377912) otherwise we'll be left behind in the future.

closed 2025-07-08 by proddy

MichaelDvP, 2025-08-01:
I'm not sure that this was the same kind oof reboot as before. My test was extreme all services enabled with lots of data, and memory was low (max_alloc 45k, free mem 109k), Also the crash happend the moment i was accessing the web page. Before the reboots happen random without other interaction. I'll make more tests.

mathieucarbou, 2025-08-01:
> I'm not sure that this was the same kind oof reboot as before. My test was extreme all services enabled with lots of data, and memory was low (max_alloc 45k, free mem 109k), Also the crash happend [...]

I am following this interesting conversation.

Just wanted to drop a note if, in your test, you have the ability to monitor the heap while accessing the web page.
We had a strange user report today about a memory leak but I fail to reproduce it and understand it right now so I am gathering more info. It is possible that this is a user mistake around library versions also. FYI: https://github.com/ESP32Async/ESPAsyncWebServer/discussions/253

mathieucarbou, 2025-08-02:
> But the crash happend when accessing with a fresh started firefox.

This can be caused my the memory leak that was found. FF sends in the background some SSL handshakes on the non-ssl port at the same time of the request so there are actually a lot of requests hitting the ESP, most of them aborted (ssl over http) but the disocnnect callback was not triggered so a lot of memory was leaked.

Ref: https://github.com/ESP32Async/ESPAsyncWebServer/discussions/253#discussioncomment-13963953

A fix has been merged, master should be fixed, I will release soon.

proddy, 2025-08-02:
Also worth pointing out having just seen this yesterday:

[image]

...that EMS-ESP is still on Arduino Core 2.0.17 (ESP-IDF 4.4.7).