emsesp/EMS-ESP32 #2583: Unable to start the hp in cooling mode with 3.7.3-dev.9

masterx1981, 2025-06-14:
Hi!
Seem that something has broken the ha thermostat entity of the heat pump, "CUHP/Enviline, Compress 6000AW" etc model. 
Last year (not know what version i was using) all was working correctly, but now when i set the hp for cooling, the thermostat entity, instead of allowing me to select the "cooling" hvac mode, remain stuck in "heating", the only other option is "off" or "auto".

proddy, 2025-06-14:
You can't remember which version of EMS-ESP you were previously on? Which version are you using now? Also what's the MQTT topic `homeassistant/climate/ems-esp/thermostat_hc1/config`?

masterx1981, 2025-06-14:
My bad i not remember the working version, as i've not activated the cooling in the latest (at least) 2 fw updates.
Now i'm using 3.7.3-dev.9

This is the topic:
`{
  "~": "ems-esp-pdc",
  "uniq_id": "thermostat_hc1",
  "obj_id": "thermostat_hc1",
  "name": "Hc1",
  "mode_stat_t": "~/thermostat_data",
  "mode_stat_tpl": "{%if value_json.hc1 is undefined or value_json.hc1.mode is undefined%}off{%elif value_json.hc1.mode=='manual'%}heat{%elif value_json.hc1.mode=='day'%}heat{%elif value_json.hc1.mode=='night'%}off{%elif value_json.hc1.mode=='off'%}off{%else%}auto{%endif%}",
  "temp_cmd_t": "~/thermostat/hc1/seltemp",
  "temp_stat_t": "~/thermostat_data",
  "temp_stat_tpl": "{{value_json.hc1.seltemp if value_json.hc1 is defined and value_json.hc1.seltemp is defined else 0}}",
  "curr_temp_t": "~/thermostat_data",
  "curr_temp_tpl": "{{value_json.hc1.currtemp if value_json.hc1 is defined and value_json.hc1.currtemp is defined else 0}}",
  "min_temp": "5",
  "max_temp": "30",
  "temp_step": "0.5",
  "mode_cmd_t": "~/thermostat/hc1/mode",
  "action_topic": "~/boiler_data",
  "action_template": "{% if value_json.heatingactive=='on'%}heating{%else%}idle{%endif%}",
  "modes": [
    "auto",
    "heat",
    "off"
  ],
  "dev": {
    "ids": [
      "ems-esp-pdc-thermostat"
    ],
    "name": "ems-esp-pdc Thermostat"
  },
  "avty": [
    {
      "t": "~/thermostat_data",
      "val_tpl": "{{'online' if value_json.hc1 is defined and value_json.hc1.seltemp is defined else 'offline'}}"
    },
    {
      "t": "~/thermostat_data",
      "val_tpl": "{{'online' if value_json.hc1 is defined and value_json.hc1.currtemp is defined else 'offline'}}"
    },
    {
      "t": "~/thermostat_data",
      "val_tpl": "{{'offline' if value_json.hc1 is undefined or value_json.hc1.mode is undefined else 'online'}}"
    }
  ],
  "avty_mode": "all"
}`

proddy, 2025-06-14:
thanks, I'll take a look.

masterx1981, 2025-06-16:
So this problem wasn't causing the hp to stop working in cooling mode, was only a missing "aesthetic" of the ha thermostat entity?
I really not know what to check. Today the hp worked normally, after that has stopped by 2 days in a row.
By now i find only dev13 available. In what dev version i find the fix?
Thanks!

closed 2025-06-16 by masterx1981

proddy, 2025-06-17:
I think it was just EMS-ESP not being configured to show the cooling option for your HP. If that is works now in the latest dev we can close it.

proddy, 2025-06-22:
also the Support Information (see https://docs.emsesp.org/Support/) which will tell is the product ID for your EMS devices

proddy, 2025-06-29:
@T1ppes ?

T1ppes, 2025-06-29:
I edited my above post with hopefully the right information.

masterx1981, 2025-07-19:
Hi! I've installed beta 14 but still i see the thermostat HP in heating mode while in cooling mode....

proddy, 2025-07-31:
Hi @MichaelDvP . I'm catching up after my vacation - I saw you made some changes in HA, does it address this issue or does it still need looking into?

marcothinnes, 2025-08-24:
My 6800i is just ignoring the hc1 HP min. flow temperatur. It keeps the flow temp always at around 18C, which makes me very nervous because of the dew point. Therefore I can’t use it.

masterx1981, 2025-09-20:
Here is the boiler-data of the heat pump:
`{"reset":"","heatingoff":"off","heatingactive":"off","tapwateractive":"off","selflowtemp":14,"heatingpumpmod":5,"outdoortemp":25.9,"curflowtemp":12.9,"rettemp":15.4,"headertemp":13.0,"heatingactivated":"on","heatingtemp":85,"heatingpump":"on","pumpmodmax":100,"pumpmodmin":0,"pumpcharacter":"proportional","curburnpow":100,"burnstarts":0,"burnworkmin":0,"burn2workmin":0,"heatworkmin":0,"heatstarts":0,"ubauptime":1271745,"lastcode":"A01(5265) 05.09.2025 02:59 - 05.09.2025 03:04","servicecode":" 0H","servicecodenumber":203,"emergencyops":"off","emergencytemp":30,"pc1on":"on","uptimetotal":1270665,"uptimecontrol":297706,"uptimecompheating":164393,"uptimecompcooling":116283,"uptimecomppool":71582784,"totalcompstarts":4073,"heatingstarts":1126,"coolingstarts":1814,"poolstarts":4294967295,"nrgconstotal":5742,"nrgconscomptotal":5743,"nrgconscompheating":4184,"nrgconscompcooling":1183,"nrgconscomppool":4294967295,"auxelecheatnrgconstotal":4294967295,"auxelecheatnrgconsheating":4294967295,"auxelecheatnrgconspool":4294967295,"nrgsupptotal":21505,"nrgsuppheating":13668,"nrgsuppcooling":6252,"nrgsupppool":4294967295,"hppower":3.1,"hpmaxpower":100,"hpcompon":"off","hpactivity":"cooling","hpbrinepumpspd":0,"hpswitchvalve":"off","hpcircspd":5,"hpbrinein":0.0,"hpbrineout":0.0,"hptc0":15.4,"hptc1":12.9,"hptc3":12.7,"hptr1":0.0,"hptr3":9.8,"hptr4":31.3,"hptr5":18.0,"hptr6":50.0,"hptr7":12.6,"hptl2":29.0,"hppl1":5.0,"hpph1":29.9,"hptw1":45.2,"poolsettemp":28.0,"hp4way":"cooling & defrost","hpin1":"off","hpin1opt":"100000110000000","hpin2":"off","hpin2opt":"000000000000000","hpin3":"off","hpin3opt":"000000000000000","hpin4":"on","hpin4opt":"001000100000","maxheatcomp":"0 kW","maxheatheat":"0 kW","auxheatersource":"exclusive","auxheateronly":"off","auxheateroff":"off","auxheaterstatus":"off","auxheaterlevel":0,"auxheaterdelay":150,"auxmaxlimit":0.0,"auxlimitstart":0.1,"auxheatrmode":"eco","hphystheat":150,"hphystcool":150,"hphystpool":1125,"silentmode":"off","mintempsilent":-10,"tempparmode":0,"auxheatmix":0,"tempdiffheat":7.0,"tempdiffcool":3.0,"vpcooling":"off","heatcable":"off","vc0valve":"off","primepump":"off","primepumpmod":0,"hp3way":"off","elheatstep1":"off","elheatstep2":"off","elheatstep3":"off","hpea0":"off","hppumpmode":"continuous","shutdown":"","dhw":{"uptimecomp":16470,"startshp":1131,"nrgconscomp":353,"auxelecheatnrgcons":4294967295,"nrgsupp":1639,"maxheat":"0 kW","alternatingop":"on","altopprioheat":35,"altopprio":45,"comfoff":56,"ecooff":52,"hpcircpump":"on","settemp":56,"seltemp":48,"seltemplow":44,"seltempsingle":65,"comfort1":"high comfort","flowtempoffset":25,"maxtemp":56,"circpump":"off","hyston":-2,"hystoff":0,"disinfectiontemp":65,"circmode":"3x3min","circ":"off","curtemp":45.3,"curtemp2":45.2,"curflow":0.0,"activated":"on","onetime":"off","disinfecting":"off","charging":"off","recharging":"off","tempok":"off","3wayvalve":"off","starts":0,"workm":0}}`

Quite funny the ha thermostat entity:

[image]

By now in in "heating mode", but in cooling state 😅

masterx1981, 2025-09-21:
> [@proddy](https://github.com/proddy)
> 
> > I want to see the value of the hpactivity.
> > We would need to adjust the code, in HA the action_template now supports off, heating, cooling, drying, idle, fan and the modes can be auto, off, cool, heat, dry, fan_only.
> 
> The `action_template` already reads `hpactivity` and can show `cooling`. But the RC3xx have only `mode` `auto/manual` switched by the climate-widget. There is no matching mode in the widget (off and heat result in manual mode on the thermostat). For heatpumps the R3xx has also entities `hpmode` with enum: `heating/cooling(heat&cool` and a bool `cooling`-entity, but i don't know how they work and it's difficult to make it work without breaking the something for gas systems. The modes in the climate widged has not changed in ems-esp, there was never be a version that shows cooling mode..

Quite curious as i had an automation that was setting the hp in cooling mode, and stopped working as "cooling" wasn't a valid option anymore. I suppose that something has changed in HA?